ct-Open-Source/tuya-convert #424: Unable to obtain IP address for vtrust-flash for aoycocr mini Wi-Fi switch

dooz127, 2019-11-28:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

Running tuya-convert in a kali docker image on a nuc and using an Android device to try to connect to vtrust-flash and tries to obtain an IP address but never succeeds. The tuya device is an aoycocr mini-Wi-Fi switch ([https://www.amazon.com/gp/product/B07N1JPPXK]).

CLI output:

```

Checking for network interface wlp3s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.............................
```
smarthack-wifi.log:

```
Stopping NetworkManager...
network-manager: unrecognized service
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp3s0...
Configuration file: /dev/stdin
Using interface wlp3s0 with hwaddr a0:c5:89:63:87:ba and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-DISCONNECTED 54:14:fd:01:7c:b9
wlp3s0: AP-STA-CONNECTED 54:14:fd:01:7c:b9

```

kueblc, 2019-11-28:
You don't have anything after `Starting AP in a screen`?

dooz127, 2019-11-28:
No, just a bunch of dots that go on forever until I Ctrl-C.

kueblc, 2019-11-28:
This means it is waiting for a successful ping to 10.42.42.1. Not sure why this wouldn't be successful on your machine.

Did `install_prereq` report any errors? Can you check your `iw` `dnsmasq` and `hostapd` installations? Can you see what versions they are and if there are updates available?

dooz127, 2019-11-29:
install_prereq:

```
Cloning into 'tuya-convert'...
remote: Enumerating objects: 156, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (109/109), done.
remote: Total 1086 (delta 93), reused 93 (delta 47), pack-reused 930
Receiving objects: 100% (1086/1086), 3.43 MiB | 3.94 MiB/s, done.
Resolving deltas: 100% (675/675), done.
root@nuc:/usr/src# cd tuya-convert/
root@nuc:/usr/src/tuya-convert# ls
LICENSE    config.txt  install_prereq.sh  start_flash.sh
README.md  files       scripts		  stop_flash.sh
root@nuc:/usr/src/tuya-convert# ./install_prereq.sh 
Get:1 http://mirrors.ocf.berkeley.edu/kali kali-rolling InRelease [30.5 kB]
Get:2 http://mirrors.ocf.berkeley.edu/kali kali-rolling/main amd64 Packages [16.5 MB]
Get:3 http://mirrors.ocf.berkeley.edu/kali kali-rolling/non-free amd64 Packages [197 kB]
Get:4 http://mirrors.ocf.berkeley.edu/kali kali-rolling/contrib amd64 Packages [105 kB]
Fetched 16.9 MB in 16s (1073 kB/s)                                             
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.8).
curl is already the newest version (7.66.0-1+b1).
dnsmasq is already the newest version (2.80-1).
git is already the newest version (1:2.24.0-1).
haveged is already the newest version (1.9.8-2).
hostapd is already the newest version (2:2.9-3).
iw is already the newest version (5.3-1).
libssl-dev is already the newest version (1.1.1d-2).
mosquitto is already the newest version (1.6.7-1).
net-tools is already the newest version (1.60+git20180626.aebd88e-1).
python-dev is already the newest version (2.7.17-2).
python-pip is already the newest version (18.1-5).
python-setuptools is already the newest version (41.2.0-1).
python-wheel is already the newest version (0.33.6-2).
python3-dev is already the newest version (3.7.5-1).
python3-pip is already the newest version (18.1-5).
python3-setuptools is already the newest version (41.2.0-1).
python3-wheel is already the newest version (0.33.6-2).
screen is already the newest version (4.7.0-1).
The following package was automatically installed and is no longer required:
  libboost-timer1.67.0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 175 not upgraded.
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-req-build-7rfxw1eg
Requirement already satisfied (use --upgrade to upgrade): sslpsk==1.0.0 from git+https://github.com/M4dmartig4n/sslpsk.git in /usr/local/lib/python3.7/dist-packages
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.7/dist-packages (1.5.0)
Requirement already satisfied: pyaes in /usr/local/lib/python3.7/dist-packages (1.6.1)
Requirement already satisfied: tornado in /usr/local/lib/python3.7/dist-packages (6.0.3)
Requirement already satisfied: pycrypto in /usr/lib/python3/dist-packages (2.6.1)
Building wheels for collected packages: sslpsk
  Running setup.py bdist_wheel for sslpsk ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-qjgusy3j/wheels/c8/51/1e/5094f34953a89ff37e7a2b3a6327fc9364e532ccb8be3bf39c
Successfully built sslpsk
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-req-build-DG9c6c
Requirement already satisfied (use --upgrade to upgrade): sslpsk==1.0.0 from git+https://github.com/M4dmartig4n/sslpsk.git in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: pycrypto in /usr/lib/python2.7/dist-packages (2.6.1)
Building wheels for collected packages: sslpsk
  Running setup.py bdist_wheel for sslpsk ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-UgcS6G/wheels/c8/51/1e/5094f34953a89ff37e7a2b3a6327fc9364e532ccb8be3bf39c
Successfully built sslpsk
Ready to start upgrade
```
running apt-cache policy on these packages:

```
iw:
  Installed: 5.3-1
  Candidate: 5.3-1
  Version table:
 *** 5.3-1 500
        500 http://http.kali.org/kali kali-rolling/main amd64 Packages
        100 /var/lib/dpkg/status
```
```
dnsmasq:
  Installed: 2.80-1
  Candidate: 2.80-1
  Version table:
 *** 2.80-1 500
        500 http://http.kali.org/kali kali-rolling/main amd64 Packages
        100 /var/lib/dpkg/status
```
```
hostapd:
  Installed: 2:2.9-3
  Candidate: 2:2.9-3
  Version table:
 *** 2:2.9-3 500
        500 http://http.kali.org/kali kali-rolling/main amd64 Packages
        100 /var/lib/dpkg/status
```

dooz127, 2019-12-01:
I installed tuya-convert directly onto the nuc (running ubuntu 18.04) but still not flashing. However, I do get more output from the command line:

./startflash.sh:

```
Checking for network interface wlp3s0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 1489.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service

Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] y
```
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3908853/smarthack-wifi.log)

kueblc, 2019-12-02:
Unsure why `dnsmasq` is not assigning IP addresses, or why the gateway is not being set up.

Can you confirm `dnsmasq` is running?

Could you try `dig a3.tuyaus.com` on the machine running `tuya-convert`?

dooz127, 2019-12-03:
The output for `dnsmasq`:

```
nobody   15823  0.0  0.0  59980   384 ?        S    19:19   0:00 dnsmasq --no-resolv --interface=wlp3s0 --bind-interfaces --listen-address=10.42.42.1 --dhcp-range=10.42.42.10,10.42.42.40,12h --address=/#/10.42.42.1
```

When I run `./start_flash.sh`, `dig` just fails. Before running .`/start_flash.sh` and after resetting my machine, running `dig` gives me:

```
; <<>> DiG 9.11.3-1ubuntu1.11-Ubuntu <<>> a3.tuyaus.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 118
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;a3.tuyaus.com.			IN	A

;; ANSWER SECTION:
a3.tuyaus.com.		11	IN	A	34.218.147.23
a3.tuyaus.com.		11	IN	A	54.69.37.156
a3.tuyaus.com.		11	IN	A	54.187.112.38

;; Query time: 128 msec
;; SERVER: 127.0.0.53#53(127.0.0.53)
;; WHEN: Mon Dec 02 19:29:23 PST 2019
;; MSG SIZE  rcvd: 90
```

kueblc, 2019-12-12:
> When I run `./start_flash.sh`, `dig` just fails

This is not good, this means neither the DNS nor the DHCP duties of `dnsmasq` are being fulfilled.

Have you tried removing `--bind-interfaces --listen-address=10.42.42.1`?

kristianwiklund, 2019-12-26:
Try disabling the systemd built-in resolver, if running.

sudo systemctl disable systemd-resolved.service
sudo systemctl stop systemd-resolved

ametad, 2020-01-12:
Try to run docker with: 

`docker run -it --net="host" --privileged kalilinux/kali-linux-docker /bin/bash`

flufftronix, 2020-01-17:
I had some similar issues with Ubuntu 18.04 (specifically ZorinOS); I tried a bunch of stuff so I'm not sure exactly what fixed it, but: if `systemctl -t service` shows dnsmasq as failed, you're not going to be able to assign IPs via DHCP and your dummy device + smart device can't start the conversion process. (Oddly enough though dnsmasq doesn't show up via `systemctl -t service` when it's running via tuya-convert but if your dummy device is getting an IP assigned it's working.)

I did a lot of `sudo lsof -i :53` and killing any process that's not supposed to be there; though the systemd ones will keep coming back unless you run `sudo service systemd-resolved stop`. 

After this I had to make a couple changes to smarthack-psk.log. First, I was seeing the [str/bytes issue](https://github.com/ct-Open-Source/tuya-convert/issues/500) which was fixed by changing `hint=self.hint)` to `hint=self.hint.decode())`. At this point my smart device (BNC60) finally wasn't being dropped from the access point every 6 seconds or so.

However, smarthack-psk.log was showing:

`    self._sslobj.do_handshake()
SystemError: <class 'ssl.SSLError'> returned a result with an error set`

..which I was able to work around by [switching psk-frontend.py's shebang to Python 2](https://github.com/ct-Open-Source/tuya-convert/issues/365#issuecomment-552129467).

Almost there; but as reported by many other users ./start-flash.sh had to cycle through a second time before it noticed the intermediate firmware. Once this worked on one plug, I was able to repeat the process on two more plugs.

Kinda crammed a lot of stuff into one post, but I'd just hope some of this might save someone else some time!

kueblc, 2020-01-25:
It's come up in some other issues that a firewall might interfere with `dnsmasq`. Try checking if `ufw` or `firewalld` is running.

compactcode, 2020-07-03:
Has the same issue with docker running on NixOS host machine. Disabling the default host firewall sorted this issue for me, thank you!