ct-Open-Source/tuya-convert #426: Suggestion to add a warning if trying to flash over SSH

marcelm, 2019-11-28:
I had the same problem as reported in issue #409: The process appeared to stop after the message "Starting AP in a screen" (because setting up the AP dropped the SSH connection I had to the Raspberry Pi).

Could you perhaps add a warning that is printed if `./start_flash.sh` is called from within an SSH session? You could check whether the `SSH_CONNECTION` variable is set.

kueblc, 2019-11-28:
Nice suggestion, thanks for the tip!

meingraham, 2019-11-29:
I run Tuya-Convert on a headless RPi 3B. So I always run Tuya-Convert over SSH. What does this and #409 imply for my setup?

marcelm, 2019-11-29:
You don’t need to worry if it works for you. A problem occurs only when the SSH connection runs via the same WiFi interface that Tuya-Convert uses to open an access point (AP). If it works for you, I would guess that you SSH into your Raspberry Pi through an ethernet cable. Is that correct?

Perhaps the message should not be phrased as a warning, more as a "hint". Something like this (perhaps shown only when `SSH_CONNECTION` is set):
```
Hint: The following step drops any SSH connections that use the WiFi interface. If you see no further messages, use SSH over a cable connection or a local terminal.
Starting AP in a screen
```

meingraham, 2019-11-30:
OK, yes, I always use the eth0 connection for the SSH session leaving the wlan0 interface for Tuya-Convert. In any case, yes one interface devoted to SSH, while a Wi-Fi interface is left free for Tuya-Convert.

markfinn, 2019-12-18:
Seconded this request.  The main documentation suggests using a pi, so I assumed that it would be running in AP and client mode.  Was rudely surprised.

I was able to work around it by putting myself in a screen before I ran ./start_flash.  that way when I lost connection I could attach my phone the the VTRUST ssid (as is required anyway) I could ssh back in and take over the screen with `screen -rd`

PR #466 (2020-01-01): Print a warning if running over SSH
This adds a check to the `setup_checks.sh` script that will test whether the script is run over an SSH connection using the WiFi interface that is about to be reconfigured into AP mode (dropping the SSH connection).

*Note*: I have not fully tested this since I don’t have my Raspberry Pi at hand at the moment. I’m opening this PR anyway, hoping that perhaps someone else may have a chance to test it before me.

Closes #426
commit b30aece0 (2020-01-01): Print a warning if running over SSH

Close #426
closed 2020-01-12 by kueblc