ct-Open-Source/tuya-convert #427: Erreur 50500 on wifi network confirmation

PPCM, 2019-11-29:
After a successfull installation, I am trying the flash procedure.
`pi@raspberrypi:~/tuya-convert-2.3.0 $ ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 1889.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to terminate dnsmasq
kill: (1889): No such process
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
`

My mobile is connected to vtrust-flash network, and when I try to add a device in the app, firstly, it ask to select a Wifi network.
Once again, vtrust-flash is choosen, but when I confirm the choice, an erreur occurs with the code 505000

Any idea?

kueblc, 2019-11-29:
You should not be using any app as part of the procedure.

PPCM, 2019-11-29:
SO, I don't understand something, when `./start_flash.sh` is started, we have the following instruction

```
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
```

How can we connect the plug to the Wifi vtrust-flash without Smart Home?

kueblc, 2019-11-29:
You connect your phone, put the device in pairing mode, and press enter. What step isn't clear?

PPCM, 2019-11-29:
I connect my phone to vtrust-flash, when I pairing with the app Smart Life.

According to your message I tryed just connecting the phone the vtrust-flash and press ENTER, here are logs

```
$ ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```

kueblc, 2019-11-29:
Do not use the app, just put the device in pairing mode

PPCM, 2019-11-30:
I did it, the previous log is the result of the try

One question: I didn't try with Raspbian Stretch but with Raspbian Buster?
Does it matter?

PPCM, 2019-11-30:
That's it!!!
With Raspbian Buster it doesn't work, I made the same procedure with Raspbian Stretch and it works fine...

So the issue is the compatibility with Raspbian Buster

kueblc, 2019-12-01:
Great, glad you got it working @PPCM

closed 2019-12-01 by kueblc

VivantSenior, 2020-02-15:
I can confirm that as of today Tuya-Convert isn't working on Raspian Buster.

kueblc, 2020-02-15:
You'll need to be more specific, can you open your own issue including your logs?

VivantSenior, 2020-02-15:
There is nothing in the Tuya-Convert logs, it's just retrying:

```(text)
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets

etc...
```

I suppose that this is the same issue as in this topic. I managed to flash OTA via Raspberry 3B+ and Raspbian Stretch.