ct-Open-Source/tuya-convert #428: Gosund SP112 / Blitzwolf BW-SHP5 not connecting to AP

dskw, 2019-11-29:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
no

**Do you have a question about configuring your newly flashed device?**
no

**Did you have issues getting a device to flash?**
I'm trying to flash a Gosund SP112, same device as Blitzwolf BW-SHP5, but fail with the initial connection.
The devices is supported by Tasmota and more infos on it are in this issue: https://github.com/arendst/Tasmota/issues/5756

I've applied the following steps:
1. successfully run install_prereq.sh 
2. run start_flash.sh
3. connect with phone to vtrust-flash
4. power on socket, goes into flash mode
5. hit enter on the script
6. notice a click and blue led going through some cycles, then only red with manual on/off mode

This is only happening when the phone is connected as well. Without the phone the socket stays in the wait-for-flash mode. Once I connect the phone .6 happens right away.
```
$ ./start_flash.sh 
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 7009.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
..................................
SmartConfig complete.
Resending SmartConfig Packets
..........................................
SmartConfig complete.
Resending SmartConfig Packets
................................................................
SmartConfig complete.
Auto retry in 3s. ..
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Stopping smart config
Do you want to try flashing another device? [y/N] n
./start_flash.sh: Zeile 107:  9295 Beendet                 ./smartconfig/main.py
Exiting...
```

```
$ cat scripts/*.log
1575052372: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1575052372: Using default config.
1575052372: Opening ipv4 listen socket on port 1883.
1575052372: Opening ipv6 listen socket on port 1883.
^C1575052560: mosquitto version 1.4.15 terminating
new client on port 443 from 10.42.42.24:31829
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening on 10.42.42.1:80
[I 191129 19:33:04 web:2246] 302 GET /generate_204 (10.42.42.22) 1.49ms
[I 191129 19:33:05 web:2246] 302 GET /generate_204 (10.42.42.22) 0.88ms
[I 191129 19:33:05 web:2246] 302 GET /generate_204 (10.42.42.22) 1.18ms
[I 191129 19:33:07 web:2246] 302 GET /generate_204 (10.42.42.22) 0.62ms
[I 191129 19:33:07 web:2246] 200 GET / (10.42.42.22) 1.22ms
[I 191129 19:33:07 web:2246] 302 GET /favicon.ico (10.42.42.22) 1.01ms
[I 191129 19:33:07 web:2246] 304 GET / (10.42.42.22) 1.03ms
[I 191129 19:33:08 web:2246] 302 GET /generate_204 (10.42.42.22) 0.99ms
[I 191129 19:33:08 web:2246] 302 GET /generate_204 (10.42.42.22) 0.92ms
[I 191129 19:34:05 web:2246] 302 GET /generate_204 (10.42.42.22) 0.99ms
^CReceived SIGINT, exiting...
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 5c:51:4f:54:9b:42 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 2c:f4:32:b7:76:e0
wlan0: AP-STA-CONNECTED ac:0d:1b:f9:82:ab
wlan0: AP-STA-DISCONNECTED 2c:f4:32:b7:76:e0
wlan0: AP-STA-CONNECTED 2c:f4:32:b7:76:e0
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 2c:f4:32:b7:76:e0
wlan0: AP-STA-DISCONNECTED ac:0d:1b:f9:82:ab
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
```

kueblc, 2019-11-29:
Check for logs in the backup folder as well

dskw, 2019-11-29:
There is no backup folder, these are the only logs I've got with the content I had posted:
smarthack-mqtt.log
smarthack-psk.log
smarthack-udp.log
smarthack-web.log
smarthack-wifi.log

kueblc, 2019-11-30:
According to your logs, the device did connect. This is the device, right?
```
wlan0: AP-STA-CONNECTED ac:0d:1b:f9:82:ab
```

Something else must be preventing it from making a call to the web server. Perhaps the psk frontend. Did you have any problems with `install_prereq`? Can you provide your installed versions of `python` and `openssl`?

dskw, 2019-11-30:
Thank you for helping out kueblc!

Yep, AC:.. is the device.
openssl version
OpenSSL 1.1.1  11 Sep 2018
python --version
Python 2.7.15+

```
Ign:1 http://ftp.halifax.rwth-aachen.de/linuxmint tara InRelease
OK:2 http://ftp-stud.hs-esslingen.de/ubuntu bionic InRelease
OK:3 http://ppa.launchpad.net/nextcloud-devs/client/ubuntu xenial InRelease
OK:4 http://archive.canonical.com/ubuntu bionic InRelease
OK:5 http://ftp.halifax.rwth-aachen.de/linuxmint tara Release
OK:6 http://ftp-stud.hs-esslingen.de/ubuntu bionic-updates InRelease
OK:7 http://security.ubuntu.com/ubuntu bionic-security InRelease
OK:8 http://packages.microsoft.com/repos/vscode stable InRelease
OK:9 http://ftp-stud.hs-esslingen.de/ubuntu bionic-backports InRelease
Ign:10 http://dl.google.com/linux/chrome/deb stable InRelease
OK:11 http://dl.google.com/linux/chrome/deb stable Release
OK:12 http://repo.steampowered.com/steam precise InRelease
Paketlisten werden gelesen...
Paketlisten werden gelesen...
Abhängigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
build-essential ist schon die neueste Version (12.4ubuntu1).
iw ist schon die neueste Version (4.14-0.1).
net-tools ist schon die neueste Version (1.60+git20161116.90da8a0-1ubuntu1).
python-dev ist schon die neueste Version (2.7.15~rc1-1).
python-setuptools ist schon die neueste Version (39.0.1-2).
python3-setuptools ist schon die neueste Version (39.0.1-2).
dnsmasq ist schon die neueste Version (2.79-1).
haveged ist schon die neueste Version (1.9.1-6).
python-wheel ist schon die neueste Version (0.30.0-0.2).
python3-wheel ist schon die neueste Version (0.30.0-0.2).
curl ist schon die neueste Version (7.58.0-2ubuntu3.8).
git ist schon die neueste Version (1:2.17.1-1ubuntu0.4).
libssl-dev ist schon die neueste Version (1.1.1-1ubuntu2.1~18.04.5).
python3-dev ist schon die neueste Version (3.6.7-1~18.04).
screen ist schon die neueste Version (4.6.2-1ubuntu1).
hostapd ist schon die neueste Version (2:2.6-15ubuntu2.5).
mosquitto ist schon die neueste Version (1.4.15-2ubuntu0.18.04.3).
python-pip ist schon die neueste Version (9.0.1-2.3~ubuntu1.18.04.1).
python3-pip ist schon die neueste Version (9.0.1-2.3~ubuntu1.18.04.1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-lp7tnfir-build
  Requirement already satisfied (use --upgrade to upgrade): sslpsk==1.0.0 from git+https://github.com/M4dmartig4n/sslpsk.git in /usr/local/lib/python3.6/dist-packages
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.6/dist-packages
Requirement already satisfied: pyaes in /usr/lib/python3/dist-packages
Requirement already satisfied: tornado in /usr/local/lib/python3.6/dist-packages
Requirement already satisfied: pycrypto in /usr/lib/python3/dist-packages
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-req-build-Zv7QBS
Requirement already satisfied (use --upgrade to upgrade): sslpsk==1.0.0 from git+https://github.com/M4dmartig4n/sslpsk.git in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: pycrypto in /usr/lib/python2.7/dist-packages (2.6.1)
Building wheels for collected packages: sslpsk
  Building wheel for sslpsk (setup.py): started
  Building wheel for sslpsk (setup.py): finished with status 'done'
  Stored in directory: /tmp/pip-ephem-wheel-cache-ewdvqb/wheels/c8/51/1e/5094f34953a89ff37e7a2b3a6327fc9364e532ccb8be3bf39c
Successfully built sslpsk
You are using pip version 19.0.3, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Ready to start upgrade
```

dskw, 2019-12-01:
Tried it again, same procedure as asked for in the script: Start script, connect phone, set device in pairing mode and hit ENTER on the script.
After some seconds the devices relay clicks some times and the led goes off and shows red.
I've attached the scripts from that run.
It looks like the device somehow turns off after connecting to the wifi. I haven't used the Go Smart app yet. Out of fear that the device might do an update to a newer firmware. Do I have to connect it once to the app before using tuya convert?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3908559/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3908560/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3908561/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3908562/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3908563/smarthack-wifi.log)

kueblc, 2019-12-01:
The relay might click when the smart config succeeds and the device connects to the AP.

It does look like it connected again from your logs, same as before. But what's strange is that it apparently doesn't make any requests, nothing in psk nor web logs.

Can you perform a network recording of the process? `tcpdump -i wlan0 -w capture.pcap`

kueblc, 2019-12-01:
> I haven't used the Go Smart app yet. Out of fear that the device might do an update to a newer firmware. Do I have to connect it once to the app before using tuya convert?

You are correct, do not use the app to avoid accidental firmware upgrades.

dskw, 2019-12-01:
Here are the files. Same behavior. I wonder if something is changed in this stock firmware and something is missing in the communication
[capture.pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/3908688/capture.pcap.zip)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3908689/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3908690/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3908691/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3908692/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3908693/smarthack-wifi.log)

dskw, 2019-12-01:
I'll keep my hands off the app then. I have 4 devices here and opened one up already for potential hard flashing. So if need be, I can try to do a dump of the stock firmware too

kueblc, 2019-12-02:
This is very helpful, thank you. It looks like HTTPS requests are not being answered. This indicates an issue with the `psk-frontend` but the log doesn't contain any additional clues. I'll keep digging.

dskw, 2019-12-05:
Ok I'm going to flash one of the devices using the pins. Do you know how to dump the stock image using esptool.py or such?

dskw, 2019-12-05:
Never mind, got it flashed using esptool now and also did a backup of the stock firmware:
[sp112_stock_image1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/3926600/sp112_stock_image1M.zip)

I still have 3 more of them and would still like to flash them using tuya-convert :D Let me know if i can test anything to help out

prodigy7, 2020-01-02:
Bought four SP112 too and run into the (i think) same issues like dskw.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4015680/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4015682/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4015683/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4015684/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4015685/smarthack-wifi.log)

Any news? Anything I can support make it working?

prodigy7, 2020-01-02:
Some trys more... Okay, tried it again and now works everything fine :-)

prodigy7, 2020-01-02:
Update: It's just a question of timing! Flashed now 4 SP112 sockets without any problem at end. Simply keep hold button until it flash for a very short time, then start the flash process with enter. In some situations the socket go back to normal light flash and I pressed again the button and the "Resending SmartConfig Packets" dot line would be longer and longer and found finally the socket for flashing. In other cases the line grows but socket was not found. Answere the question for try flashing another device runs directly into finding the socket and flashing it.

kueblc, 2020-01-25:
@dskw are you still interested in converting your devices? It has come up in some other issues that OpenSSL might be out of date. Also might want to try again with the `development` branch which should hopefully fix a number of things.

kueblc, 2020-01-25:
Closing due to inactivity, feel free to comment if this still needs attention

closed 2020-01-25 by kueblc

joebusch67, 2020-04-24:
Hi can somebody help me. 
I have the same problem, described above. I can't flash the gosund SP112 with the tasmota image. Until the flash process i get again and again the message "SmartConfig complete.
Resending SmartConfig Packets"

Nothing of the above workarounds help me.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4528589/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4528590/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4528591/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4528592/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4528593/smarthack-wifi.log)

kueblc, 2020-04-24:
@joebusch67 your device firmware is incompatible #483

ChristianWeyer, 2020-08-08:
We are having the same issue as described here. You mentioned an outdated OpenSSL @kueblc - do you have any suggestions on how to proceed from here?
Thanks.