ct-Open-Source/tuya-convert #430: Can't flash Teckin SP21 (Invalid x-www-form-urlencoded body)

szeiger, 2019-11-30:
I'm trying to flash some Teckin SP1 devices on a Samsung NC10 running Ubuntu Server 18.04 LTS. Everything looks normal (as far as I can tell) in the console but SmartConfig never succeeds:

```
Checking for network interface wlp2s0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 1206.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 1194.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.............
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]

Exiting...
```

I don't have any experience with tuya-convert so all my information about what to expect in the logs is from other tickets but this line in smarthack-web.log is the only thing that looks suspicious to me:

```
[33m[W 191130 13:58:53 httputil:789][m Invalid x-www-form-urlencoded body: 'utf-8' codec can't decode byte 0xe7 in position 2: invalid continuation byte
``` 

Here are the complete log files:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3907206/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3907207/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3907208/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3907209/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3907210/smarthack-wifi.log)

kueblc, 2019-11-30:
`Invalid x-www-form-urlencoded body` is coming from your phone. All of the HTTP requests in the web log come from the same IP address, so the IoT device never established a connection.

Unfortunately this device has been known to be incompatible due to changing microcontrollers. Can't be sure whether this particular unit is running an ESP8266 unless we establish a connection. Have you been able to flash any other devices with `tuya-convert` recently? What is your OpenSSL version?

kueblc, 2019-12-01:
Since the original question has been answered I'm closing the issue, but please feel free to comment and reopen if you need help with anything else

szeiger, 2019-12-03:
This is the first time I'm trying to use tuya-convert and I don't have any other devices here that I could try. OpenSSL version is 1.1.1. I opened up one of the devices and can confirm that it has an ESP8266.

(just commenting here because I can't reopen)

kueblc, 2019-12-03:
Could you share some pictures of your device breakdown? So far our test for non-ESP devices hasn't been wrong, so it would be great to get confirmation.

szeiger, 2019-12-03:
Here's the main board:
[image]

I didn't want to destroy the case unnecessarily so I improvised a mirror from a piece of reflective film to get a look at the chip on the daughter board. It's good enough to read ESP8266EX:

[image]

Let me know if there's anything else of interest on that board which would justify destroying the case for science :-)

szeiger, 2019-12-03:
I took another attempt at flashing one of the devices to observe the indicator LED more closely. If I put into pairing mode by keeping the button pressed for a few seconds, the LED starts blinking and continues to do so for quite a while. When tuya-convert sends the SmartConfig Packets, this blinking stops, there's a short burst of flashes and the LED goes dark. It certainly looks like the device reacts to the packets.

kueblc, 2019-12-03:
Ah I was mistaken, I must be confusing this with another issue thread. There is no indication in your logs that this is not an ESP. Thanks for confirming.

It looks like perhaps an issue with the `psk-frontend`. Perhaps related to @Boldie's issue in #418 or #428. To confirm this, could you do a network capture while attempting the process? You can do this with `tcpdump -i wlp2s0 -w capture.pcap`.

shiaky, 2019-12-04:
Hi I got the Same problem flashing the SP21 brought one week ago. I captured the required pcap for you.

[capture_flashing_SP21.pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/3923182/capture_flashing_SP21.pcap.zip)

kueblc, 2019-12-04:
Hi @shiaky, this does not look related to this issue at all. Thank you for trying to help anyway.

There is a variety of things that can cause a device not to flash, just because you have the same device doesn't imply it is the same problem. In this case, @szeiger's device never made any requests, yours clearly did.

You will need to open a new issue and post your logs for assistance.

shiaky, 2019-12-04:
@kueblc Thank you. I thought it could be the same issue. I will open a new one.

larseggert, 2019-12-18:
I have the same issue as @szeiger, and I see these log messages:
```
new client on port 443 from 10.42.42.26:49162
ID: 0142416f3f3f3f3f3f3f3f3f3f3f3f3f3f504e9a9f592aa25ecc61c37e957b9b345fbba0082966c6b71b82e0ea090dd4e4b0
Prefix: BAo?????????????
PSK: 9d9400b02cf31264eb9b5d0afea3f0d4cd93380df9fe1e62093f0c0089e4681a
('could not establish sslpsk socket:', SSLError(1, u'[SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:727)'))
new client on port 443 from 10.42.42.28:56092
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.28:56093
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.28:56094
('could not establish sslpsk socket:', SSLError(1, u'[SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:727)'))
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 104, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```
and
```
POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=59&uuid=1e629600fd25c298&v=4.1&sign=a678ef5e22f1dc41f464c584cc067d43
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload 827B224B091F72EE227E87CBD5B0168B
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJb7mbvSzunYMRTYao7293Qf9F2VBmqNqTqJcWuiX4YubPLpx8vFNLs0L08MQ/LhFEe7mevrGF2WUpAllBDW/Q5l","t":1576665244,"sign":"3731bd07eedfe271"}
ESC[32m[I 191218 10:34:04 web:2246]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=59&uuid=1e629600fd25c298&v=4.1&sign=a678ef5e22f1dc41f464c584cc067d43 (10.42.42.1) 11.67ms
```

kueblc, 2019-12-18:
@larseggert your issue is not the same as @szeiger, see this line
> WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

Unfortunately you have one with a different microcontroller.

Duplicate of #269 #310 #321 #327 #341 #364 #370 #380 #391 #425 #438 #447

kueblc, 2020-01-25:
@szeiger let me know if you are still working on this. Closing for now to keep the issue queue clean.

closed 2020-01-25 by kueblc