ct-Open-Source/tuya-convert #431: Device did not appear with the intermediate firmware

Nerospeed, 2019-12-01:
Hello, 

I try to flash my SP22 with the new Firmware. 

My System is a Lenovo T530 Notebook with a Ubunut 18.04.3 Desktop. 
My WLAN-Adapter is wlp3s0.

I got the follwing error:
_Device did not appear with the intermediate firmware_

`Checking for network interface wlp3s0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 14027.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.............
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...`

When i try to run psk-frontend.py I got a failure in

`:~/tuya-convert/scripts$ ./psk-frontend.py 
Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 97, in main
    proxies = [PskFrontend(gateway, 443, gateway, 80), PskFrontend(gateway, 8886, gateway, 1883)]
  File "./psk-frontend.py", line 48, in __init__
    self.server_sock = listener(listening_host, listening_port)
  File "./psk-frontend.py", line 17, in listener
    sock.bind((host, port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address `

mybe I can not do the script like this, i don´t know.

Here my logfiles. Help :-D

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3908135/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3908136/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3908137/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3908138/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3908139/smarthack-web.log)

kueblc, 2019-12-01:
From your web log
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```
Duplicate of #269 #310 #321 #327 #341 #364 #370 #380 #391 #425

closed 2019-12-01 by kueblc

dave-castle, 2020-02-24:
which devices on german market do you recommend for tuya convert? I used Teckin SP22 last year without any problems, tried two packages from amazon this month with the error above