ct-Open-Source/tuya-convert #432: Slow Connection and Constant WiFi Drop

moccajava, 2019-12-01:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_
Yes. Managed to flash: Aseer 1-gang & 2-gang Wall Socket, Shawader Dimmer Switch, Bomcosy E7 Bulbs. However, have not managed to figure out the GPIO settings for Aseer and Shawader. Bomcosy followed the template in Blackadder + new colour UI in tasmota v7.1.1. 

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_
Will spend time reading up on configuring the GPIO.

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
The main issue was that after the flash, I noticed the CORE/SDK was showing STAGE2/2.2.2 (Dev). Any subsequent changes to the device from here on was excruciatingly slow. It appears that the WiFi details got disconnected and has no way of reconnecting again. It took about 20-30 minutes before the tasmota AP appeared, and had to re-enter the password, the SSID details is stored though.

Even if doing a power cycle, same symptoms. I took the slow approach of upgrading the firmware to the latest tasmota.minimal first and then tasmota v7.1.1. Now it appears to be stable.

From what I read in this release, there is no need to follow-up with a curl command as the start flash process is now menu driven all the way.

Any clues why? :S I'm putting off flashing the others until I find out why.

**Please remember to close your issue when the problem has been addressed!**

moccajava, 2019-12-01:
Also, is a "reset 5" command necessary after the initial flash from tuya-convert? On the first login instance...

kueblc, 2019-12-02:
Sounds like you successfully installed third party software. We do not develop Tasmota, you will have to go to them for post install support. Seems like you have a solution there by flashing minimal between upgrades.

As for `reset 5`, tuya-convert now resets the system param area during the flashing process so it shouldn't be necessary, but some are still recommending it in case some other areas of flash cause problems when dirty.

closed 2019-12-02 by kueblc

moccajava, 2019-12-02:
Thanks for the response. I tried to replace the tasmota.bin with the latest available v7.1.1 so that during the flash process it could be upgraded straight one time but it's not within the minimum size.

Is the upgrade.bin a 3rd party as well or its internal use for tuya-convert?

kueblc, 2019-12-02:
Yes, `upgrade.bin` is our internally used intermediate firmware loader. After it is loaded by the Tuya firmware, it enables you to make a backup and install your thirdparty firmware.

meingraham, 2019-12-03:
To flash the latest Tasmota use http://thehackbox.org/tasmota/tasmota-wifiman.bin. This is always the latest codebase (within 1-2 hours). This binary is built to have the major Tasmota features while still retaining a small size that can be flashed with Tuya-Convert.

moccajava, 2019-12-03:
@meingraham : thanks will try that out. is the binary on thehackbox = full tasmota.bin on Git?

meingraham, 2019-12-03:
thehackbox is the official Tasmota repository for both the release versions and the development versions. There are several firmware variants that get pre-compiled which include different features (since the ESP82xx has limited memory resources). tasmota-wifiman.bin is "quite" full featured to try to address most "Tuya" type device needs. wifiman is the version that is included with Tuya-Convert by default. The link I posted is just the current version of that variant.