ct-Open-Source/tuya-convert #433: ubuntu 18.04.3 package dnsmasq is not available but is referred to by another package

hfcheng66, 2019-12-02:
I'm trying in on ubuntu 18.04.3 usb live on my laptop. 
running ./install_prereq.sh gave me some errors, and the first error is:
`package dnsmasq is not available but is referred to by another package
`
Searched around and found this one may related: https://github.com/ct-Open-Source/tuya-convert/issues/187
However it is marked as closed.
Can anyone help me on how to resolve this?

Thanks.

kueblc, 2019-12-02:
#187 is unrelated, it is about avoiding run time conflicts with `dnsmasq`, your error says it couldn't be installed in the first place.

Did you try running `sudo apt-get update` first?

hfcheng66, 2019-12-02:
I did update and also upgrade. looking back history seems there is error showing out of space? I have the ubuntu 18 desktop live on a 8GB usb, space supposes to be enough.

dpkg: error processing archive /tmp/apt-dpkg-install-19SJfR/157-linux-firmware_1.173.12_all.deb (--unpack):
 unable to make backup link of './lib/firmware/radeon/BONAIRE_sdma.bin' before installing new version: No space left on device
No apport report written because the error message indicates a disk full error
                                                                              dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)

kueblc, 2019-12-02:
The error is clear then, you do not have enough room to install `dnsmasq`. You will need to install Ubuntu or recreate the live disk with more persistent storage space.

closed 2019-12-02 by kueblc