ct-Open-Source/tuya-convert #435: AP vtrust-flash never come up again after ubuntu sleep/reboot

hfcheng66, 2019-12-02:
I have a fresh installation of ubuntu 18.04.3 desktop on my laptop. I am able to flash one of smart plug to tasmota. 

However after ubuntu went to sleep mode and when I tried to flash more, the AP vtrust-flash never come up. Tried reboot ubuntu, reboot cellphone, tried different cellphone, still no AP.

searched up https://github.com/ct-Open-Source/tuya-convert/issues/172, tried to kill wpa_supplicant and start again, still no AP.

log attached. Thanks!

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3913603/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3913604/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3913605/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3913606/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3913607/smarthack-wifi.log)

kueblc, 2019-12-12:
> However after ubuntu went to sleep mode and when I tried to flash more, the AP vtrust-flash never come up

Did you run `stop_flash` after you were done? Was the AP running when you put the computer to sleep?

> searched up #172, tried to kill wpa_supplicant and start again, still no AP.

That's an old, closed, resolved issue. Do not kill wpa_supplicant as we take care of that automatically now.

hfcheng66, 2019-12-13:
The first time happened, I think the AP was still running when the computer went to sleep since I was trying to configure the Tasmota UI from another computer.

kueblc, 2020-01-26:
Can you create an AP using `hostapd` directly? Have you tried updating?

hfcheng66, 2020-01-27:
> Can you create an AP using `hostapd` directly? Have you tried updating?

Yes I can able to create the AP manually, however it is not visible and can't connect to it. Only a fresh install of ubuntu works. I think this is not really related to the tuya convert. I tried on two separate laptops with ubuntu 18.04 LTS, all behavior same. could be related to either ubuntu itself or the laptops I am using.

NickVeld, 2021-04-28:
I have the problems with AP after sleep on Ubuntu with the connected WiFi adapter. But a restart and device reconnection solved it.