ct-Open-Source/tuya-convert #436: looping with "Resending SmartConfig Packets" when attempting to flash LSC smart LED

onaicul, 2019-12-03:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

I've repeatedly failed in flashing a RGB Light bulb.

Model: LSC Smart Connect Smart LED 806lm 970714 9W 70mA E27, bought in Action, virgin (never connected).

I start the flash with start_flash.sh and I connect android phone and/or Win10 HP laptop to the "vtrust-flash". 
Once I start device in a connect mode (flashing fast) I'd press enter. Device keeps flashing for several seconds, and than light stops flashing and stays lit and calm (like in this video around here: https://youtu.be/dyUyewiKpRA?t=443) and I am expecting it to proceed with flashing, but it never completes the job. Instead, I repeatedly see "Resending SmartConfig Packets" until tuya-convert quits and asks me if I want to flash another device.

I've seen incompletes with arp. Below is output of arp -a and uname -a, and smarthack-wifi.log is attached.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3918433/smarthack-wifi.log)

Raspberry Pi 4 (4GB RAM) w fresh Raspbian install:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

pi@raspberrypi:~ $ arp -a
? (10.42.42.23) at 9c:5c:f9:34:38:df [ether] on wlan0
? (10.42.42.33) at 68:57:2d:19:63:a8 [ether] on wlan0
? (10.42.42.42) at <incomplete> on wlan0
? (10.42.42.31) at d8:f2:ca:7b:2d:70 [ether] on wlan0
? (10.42.42.31) at d8:f2:ca:7b:2d:70 [ether] on wlan0
pi@raspberrypi:~ $

onaicul, 2019-12-03:
I didn't realize other logs are useful besides smarthack-wifi.log; in the smarthack-web.log I found following:

payload 26C04AC77A5132CD5276CBB7A11ACD0F
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJZXaOb5J/d17A7pmSD7kTkI9F2VBmqNqTqJcWuiX4YubA/dq3YgfE3LJFWA8PXMXWO7mevrGF2WUpAllBDW/Q5l","t":1575401614,"sign":"3254083603db8d03"}
ESC[32m[I 191203 19:33:34 web:2162]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=29&uuid=9d6d21c3a3dddec7&v=4.1&sign=a9db9ad1d7aa71baa5eefcb12a0788b8 (10.42.42.1) 62.70ms

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=31&uuid=9d6d21c3a3dddec7&v=4.1&sign=ebf8902a50e7f4171098de35a7e57068
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload DC6C853FE6A7E728B776B3105CE77977
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJaFZDTurV0oEl53A8KqzQmV9F2VBmqNqTqJcWuiX4YubD3dTpZ4/wAIqUsu+JxfHVi7mevrGF2WUpAllBDW/Q5l","t":1575401616,"sign":"d19dd041eaec98e3"}
ESC[32m[I 191203 19:33:36 web:2162]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=31&uuid=9d6d21c3a3dddec7&v=4.1&sign=ebf8902a50e7f4171098de35a7e57068 (10.42.42.1) 264.27ms

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=33&uuid=9d6d21c3a3dddec7&v=4.1&sign=baa81b7e7e43c5125022575cc5d7c77c
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload EFF7FA450EB0EA27A97C470BC0B014C5
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJb9V3S2A8ZU505rzbUp8jye9F2VBmqNqTqJcWuiX4YubMhJp2QAEHj1u9y+2hL4xUa7mevrGF2WUpAllBDW/Q5l","t":1575401618,"sign":"812dc48d2201fc50"}
ESC[32m[I 191203 19:33:38 web:2162]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=33&uuid=9d6d21c3a3dddec7&v=4.1&sign=baa81b7e7e43c5125022575cc5d7c77c (10.42.42.1) 95.64ms
ESC[32m[I 191203 19:35:24 web:2162]ESC[m^O 302 GET /gslb/?ver=3.0&type=wifi&uuid=0&list=app.chat.xiaomi.net%2Cresolver.msg.xiaomi.net&sdkver=26&osver=26&os=F8131%3A1629766330&mi=0&key=2ca2041de5c6e7b18516f2e97e35fc6e (10.42.42.23) 3.11ms
ESC[32m[I 191203 19:35:24 web:2162]ESC[m^O 200 GET /?ver=3.0&type=wifi&uuid=0&list=app.chat.xiaomi.net%2Cresolver.msg.xiaomi.net&sdkver=26&osver=26&os=F8131%3A1629766330&mi=0&key=2ca2041de5c6e7b18516f2e97e35fc6e (10.42.42.23) 2.33ms
ESC[32m[I 191203 19:38:28 web:2162]ESC[m^O 302 GET /gslb/?ver=3.0&type=wifi&uuid=0&list=app.chat.xiaomi.net%2Cresolver.msg.xiaomi.net&sdkver=26&osver=26&os=F8131%3A1629766330&mi=0&key=2ca2041de5c6e7b18516f2e97e35fc6e (10.42.42.23) 2.97ms
ESC[32m[I 191203 19:38:28 web:2162]ESC[m^O 200 GET /?ver=3.0&type=wifi&uuid=0&list=app.chat.xiaomi.net%2Cresolver.msg.xiaomi.net&sdkver=26&osver=26&os=F8131%3A1629766330&mi=0&key=2ca2041de5c6e7b18516f2e97e35fc6e (10.42.42.23) 2.35ms
^CReceived SIGINT, exiting...

This is contrary to the reports I've seen here: https://blakadder.github.io/templates/lsc_smart_connect_filament.html but I am not sure as I haven't seen this exact model of bulbs being flashed or mentioned somewhere.

kueblc, 2019-12-03:
Looks like you figured it out yourself, this device unfortunately can't be flashed. Hope you can exchange these for something with an ESP, good luck!

closed 2019-12-03 by kueblc

onaicul, 2019-12-03:
Can you please check my logs as well? Is it possible that it was attempt to flash my phone or laptop instead?
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3918512/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3918513/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3918514/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3918515/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3918516/smarthack-mqtt.log)

onaicul, 2019-12-03:
ok, confirmed, those devices don't come with ESP but with Tyua Wi-Fi Module — 
WR3L (RTL8710BX)

https://www.mikrocontroller.net/topic/479796#5985325

So this ain't gonna work. Please close the issue.