ct-Open-Source/tuya-convert #437: hang backing up firmware, flash works

chriscook8, 2019-12-04:
Teckin SP20 smartplug

The stage in start_flash that backs up the existing firmware would hang after about 7000 bytes transferred. Ctrl-C out to a prompt and then running the next step, firmware-picker, would continue as normal providing a re-flashed module. 

Reproduced over 4 separate plugs. Other model plugs ran same install of start_flash all the way through as expected.

kueblc, 2019-12-04:
Hi @chriscook8, thanks for your report.

It is a known issue with some devices, perhaps due to dirty power supplies.

What is your expected resolution?

Did you manage to complete a backup for any of the SP20s?

chriscook8, 2019-12-04:
No, none of them backed up. I don't need a resolution, just reporting an issue and offering to help troubleshoot if need be. I suppose a "do you want to skip the backup step?" prompt in the script would have sped things up for me after I discovered the issue. Thanks

kueblc, 2019-12-04:
Do you have any other stock SP20s for testing?

> I suppose a "do you want to skip the backup step?" prompt in the script would have sped things up for me after I discovered the issue

Not a bad idea

chriscook8, 2019-12-04:
Sorry, they are all happily Tasmota now.... thanks to tuya-convert!

kueblc, 2019-12-04:
I'll look into this but it will be difficult to verify without one of these devices. If anyone willing to help can reproduce this, please comment.

tobit6795, 2019-12-06:
hi, i have the same issue atm with this guys
https://www.action.com/de-de/p/lsc-smart-connect-smarter-multicolor-spot-2/
i have succesfully flashed 2/4 with tasmota, the other 2 are stuck in firmware backup

kueblc, 2019-12-06:
Could you share your firmware backups please?

tobit6795, 2019-12-06:
[20191206_215833.zip](https://github.com/ct-Open-Source/tuya-convert/files/3934326/20191206_215833.zip)

kueblc, 2019-12-06:
Those are both empty files. Check in the backups directory for the 2/4 devices that succeeded in backing up.

tobit6795, 2019-12-06:
[20191206_224152.zip](https://github.com/ct-Open-Source/tuya-convert/files/3934384/20191206_224152.zip)

kueblc, 2019-12-06:
Thanks but still doesn't look like a complete backup. A full backup should be exactly 1MB.

When you say success, you mean backup success or flashing success? They should all be flashable, as discussed above you can manually continue the process if the backup fails.

tobit6795, 2019-12-06:
[20191206_233259.zip](https://github.com/ct-Open-Source/tuya-convert/files/3934413/20191206_233259.zip)

Now this one should be what you were looking for

tobit6795, 2019-12-06:
interesting, after i shortened the distance of the devices, the backup seems to work (or is it just luck?). Now im stuck in flashing. everytime i enter 2 for flash tasmota.bin i get "Not found: /flash" as error

kueblc, 2019-12-07:
Thanks @tobit6795 I will take a look at your backup and see if I can reproduce the issue.

The error you got when trying to flash was due to running an older version of the intermediate firmware with a more recent version of tuya-convert. This probably means you attempted to flash this device with an older version of tuya-convert but never proceeded to `/flash3`.

Glad you were able to sort it out!

mkono87, 2019-12-20:
Im having the same issue with 1 of 2 sp20s and cant get past the backup. Just says speed 0 and eta of 20+ hours.

kueblc, 2019-12-20:
@mkono87 see #459

commit a07f3d7e (2020-01-29): Add timeout on backup and allow user to continue anyway, fix #437
PR #524 (2020-01-29): Release v2.4.3
- Update `sslpsk` installation
- Add timeout on backup and allow user to continue anyway, fix #437 
- Attempt to stop firewalls, fix #518
closed 2020-01-29 by kueblc