ct-Open-Source/tuya-convert #439: Can't flash Beca BHT-6000GCLW

radokristof, 2019-12-04:
Hi!

Others reported that they were able to flash a custom firmware to this thermostat via tuya-convert.

However mine can't upload the intermediate firmware. My fear is that it has a newer firmware which doesn't accept this firmware?
Looking at the logs I think it connects to the AP and also on the device, after I start smartconfig part, the Wi-Fi logo stops blinking.
Can this be resolved somehow or am I doing something wrong?

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3924346/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3924347/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3924348/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3924349/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3924350/smarthack-mqtt.log)

kueblc, 2019-12-05:
I do see that the device connected, but oddly at 10.0.0.88. It should have picked up an address in our DHCP range 10.42.42.*. Did you change `GATEWAY` in config or alter any of the scripts?

Would you mind sharing your `dnsmasq` and `openssl` versions?

radokristof, 2019-12-05:
Yes for sure:
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80

The problem can be two things (haven't noticed that the range is not correct):
- This is not a fresh install and maybe something conflicts.
- When installing the prerequsities, apt showed that there were a dnsmasq config conflict and I had to choose a config. Maybe I choosed the wrong one. I will try to look up that config and maybe change it what it is used here.

radokristof, 2019-12-05:
I have found the dnsmasq config, everything is commented out now.
Does something needs to be enabled here? (I suppose that dnsmasq is providing the DHCP server right?)

kueblc, 2019-12-05:
Hmm sounds like #424

kueblc, 2019-12-05:
I wonder if it's related to a change in dnsmasq 2.80, you and @dooz127 both have this version, I'm on 2.79. Checking [the changelog](https://github.com/ldir-EDB0/dnsmasq/blob/f73f7397d7e99f8412b7dc465061d20f642cdc85/CHANGELOG#L89) it looks like some behavior changed with dhcp-range.

Why don't you try removing [this line](https://github.com/ct-Open-Source/tuya-convert/blob/dbeab1ba9d8730b2fbf97b48b7115c2b418a592e/scripts/setup_ap.sh#L30) or the next and see if you get offered an IP.

dooz127, 2019-12-05:
Tried removing one line, then the other, and both lines but still can't get an IP for the Android.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3925171/smarthack-wifi.log)

radokristof, 2019-12-05:
I have tried running dnsmasq with the same parameters individually, it worked... So I don't know what can be wrong.
I will try to downgrade now...

radokristof, 2019-12-05:
Nothing changed... also if I reconnect it to my network and even deleting the reserved address by DHCP it gets the same IP address... I don't know why.

kueblc, 2019-12-05:
You are resetting the device into pairing mode correct?

radokristof, 2019-12-05:
Yes (I hope that this is the only way). This is the way how I can connect to using the original app. When I try to flash the firmware, it would not reconnect to my 'real' network so it paired to the vtrust-flash AP.

dooz127, 2019-12-05:
Yes, I unplug the device and put into pairing mode right before the last prompt from `./start_flash.sh`.

radokristof, 2019-12-05:
Same for me. The only difference from @dooz127 for me I think, that every device I try to connect it gets a right IP address, only the ESP device is which doesn't get an IP address from that range

radokristof, 2019-12-05:
I was able to flash it with Ubuntu in VMWare and using a D-Link Plug&Play Wifi Adapter.
Thanks!

closed 2019-12-05 by radokristof