ct-Open-Source/tuya-convert #441: no AP created

gue0411, 2019-12-07:
I wanted to run tuya-convert on a Kali Live.
I didn't get an AP, but an error message like this:

`IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
..Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
Traceback (most recent call last):
  File "./smartconfig/main.py", line 26, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address
......................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...`

Here is the smarthack-web.log:
`Could not start server on port 80
[Errno 99] Cannot assign requested address`

Here is the smarthack-wifi.log:
`Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...`

Here is the smarthack-psk.log:
`Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 97, in main
    proxies = [PskFrontend(gateway, 443, gateway, 80), PskFrontend(gateway, 8886, gateway, 1883)]
  File "./psk-frontend.py", line 48, in __init__
    self.server_sock = listener(listening_host, listening_port)
  File "./psk-frontend.py", line 17, in listener
    sock.bind((host, port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address`

I hope you can help me please?

henfri, 2019-12-07:
Please post the Output of iwconfig and config.txt.

gue0411, 2019-12-07:
**config.txt**:
`# Please input the wlan device to be used (most of the time it is wlan0 or wlan1)
WLAN=wlan0

# Here you could change the WIFI-name but most likely most scripts won't work after
# Because the WIFI-credentials are hardcoded in the esp8266-ota-flash-convert
AP=vtrust-flash
GATEWAY=10.42.42.1`

**iwconfig**:
`root@kali:~# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"daham"  
          Mode:Managed  Frequency:5.22 GHz  Access Point: E8:DF:70:7D:22:A3   
          Bit Rate=27 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=54/70  Signal level=-56 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:24   Missed beacon:0`

I'm running as root.

kueblc, 2019-12-09:
Use triple backticks (` ``` `) instead of single backticks to enclose multiline blocks.

The other services are not starting because the interface never became available.

Did you try doing a web search for the error in your wifi log?

gue0411, 2019-12-17:
I tried a lot of thing by now.
Unfortunately all without a solution of my problem.

I found that kali has no hostapd.conf file, so this could be the problem...
But no more ideas...  :(

gue0411, 2019-12-20:
I tried with a laptop and a new kali image so far.
Now I tried with a new raspberry and it worked!
So it's ok...

closed 2019-12-20 by gue0411