ct-Open-Source/tuya-convert #442: Tuya Convert, could not connect to kernel, getting stuck

wpbezemer, 2019-12-08:
Tuya Convert is not starting the AP and other stuff.

When starting Tuya convert, type in "yes" and enter.
Two times i type "Y" to continue.

i got the message "starting AP ......."  and then it got stuck.
I dont get any logs to check.

When i start the script "setup_ap.sh" i get error "Could not connect to Kernel".

I have a raspberry pi 3B, with build-in. Clean instal of Raspbian strech lite "26 sep 2019", i updated the OS before installing git and getting tuya convert.

Anybody advise here?

kueblc, 2019-12-09:
When opening a new issue please follow the template provided. You will need to post your logs as recommended in the template that you overwrote when opening this issue. In this case the `smarthack-wifi` log is the most relevant.

wpbezemer, 2019-12-09:
Hi @kueblc 
thanks for replying, however the smarthack-wifi.log is empty.
i used sudo nano to open this file, but 2 empty rows are there.

any other suggestion how i can get logs?

ps, one observation is that somehow i lose connection after it is stuck with "Starting AP".

kueblc, 2019-12-09:
> ps, one observation is that somehow i lose connection after it is stuck with "Starting AP".

That there is your problem. See #426. It creates the AP using your wifi adapter, so you cannot use the same interface for `ssh`. You'll need to connect via ethernet or USB, or control the rPi directly.

meingraham, 2019-12-09:
@wpbezemer 

More specifically ensure that your terminal session is not using the Wi-Fi connection.

https://github.com/ct-Open-Source/tuya-convert/issues/426#issuecomment-560015469

kueblc, 2019-12-12:
Closing as resolved. Let us know @wpbezemer if you have any more issues.

closed 2019-12-12 by kueblc

wpbezemer, 2019-12-13:
Hi, 
Im sorry, im traveling right now and was not able to answer the remarks / questions.
The terminal sessions was not used with WIFI, the raspberry pi is connected with ethernet cable. However, i saw a remark that maybe using this raspberry pi directly instead of using putty might help. So im going to try that.

I'm not able to do this soon, so i will update hopefully next week. At least thanks for throwing suggestions and idea's!
regards,

wpbezemer, 2019-12-17:
Hi All,
it seems like that in my case the raspberry pi is rebooting due a issue on raspberry or voltage drop.
i still need to analyze further. Sorry for the trouble.
regards,