ct-Open-Source/tuya-convert #446: BSD33 can't be flashed, not an ESP

netpok, 2019-12-09:
I received some smart sockets which I identified as BSD33 ([Tasmota template](https://blakadder.github.io/templates/bsd33.html), [AliExpress listing](https://www.aliexpress.com/item/4000060646437.html))

I tried to flash it but it stays in
```
SmartConfig complete.
Resending SmartConfig Packets
.................
```
loop. I found the following line in the web log:
`WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware`.

As far as I seen it's glued together so I don't want to break it, but if it helps I could set up one with the tuya cloud.

The logs:
[mqtt](https://pastebin.com/P2FvsF3P)
[psk](https://pastebin.com/6pKtb1G7)
[udp](https://pastebin.com/2KWDapNV)
[web](https://pastebin.com/4nthGnAW)
[wifi](https://pastebin.com/7J8GG6zy)

kueblc, 2019-12-09:
Hi @netpok, what is your question or issue? It seems you have answered it yourself.

netpok, 2019-12-10:
I mean it has a Tasmota template so I think there is 99.9% chance that it's ESP based.

Is there a way to  say "I don't care just flash it", I don't see a way to get it apart without breaking it so I would like to leave that as a last try after I bricked it.

tonysprenk, 2019-12-10:
It could be that the manufacturer has changed components. Until now the warning message has always been proven right.

netpok, 2019-12-10:
I'm sorry. You're right, I got it apart and it turned out it's based on the WinnerMicro W600 (I missed that there are contenders to esp).

Here are some pictures for anyone interested: https://imgur.com/a/UvVetqD

closed 2019-12-10 by netpok

kueblc, 2019-12-10:
For future readers,

> I mean it has a Tasmota template so I think there is 99.9% chance that it's ESP based.

Unfortunately, as you discovered, manufacturers are switching.

> Is there a way to say "I don't care just flash it", I don't see a way to get it apart without breaking it so I would like to leave that as a last try after I bricked it.

tuya-convert isn't blocking you, the device is. tuya-convert is just telling you why it won't work.

Wim3d, 2019-12-31:
> I'm sorry. You're right, I got it apart and it turned out it's based on the WinnerMicro W600 (I missed that there are contenders to esp).
> 
> Here are some pictures for anyone interested: https://imgur.com/a/UvVetqD

Can you explain how you dismantled it?

netpok, 2020-01-02:
You need to pry it open.

Just put a screwdriver in the cutout at the earth and apply some force by using the other side of the socket as a pivot point you should be able to crowbar it out without damaging the socket.

Wim3d, 2020-01-02:
> You need to pry it open.
> 
> Just put a screwdriver in the cutout at the earth and apply some force by using the other side of the socket as a pivot point you should be able to crowbar it out without damaging the socket.

Thanks, that works perfectly. My device is an ESP8266 version.

kueblc, 2020-01-02:
Did you try to flash it with `tuya-convert`?

Wim3d, 2020-01-02:
I will try to flash it with my own firmware via a hardware connected serial connection. I just figured out the pinout of the device. I will post the code on Github when I succeed.

kueblc, 2020-01-02:
OK just wanted to make sure you did not receive a false positive from `tuya-convert`.

dcboy87, 2020-05-02:
I also ordered this socket and had problems flashing it with tuya-convert. First i thought, it may also be a WinnerMicro chip inside, but then i did a MAC address search and found it was an Espressif Inc. address. I updated tuya-convert to the latest version, but still no luck. My last option was to open the socket and try to flash it by serial connection. Unfortunately i damaged the relais, so it's useles now.

My question is now, if i can do anything to help to flash this socket with tuya-convert, as there is still the original firmware on the **ESP8285** chip.
Is this ESP even supported from tuya-convert?

Thanks to @Wim3d and his tutorial on https://www.instructables.com/id/Hardware-and-Software-Hack-Smart-Devices-Tuya-and-/ how to open this socket!

Chuckame, 2020-08-26:
I'm interested for this update, I want my device with tasmota (local) instead of chinese mens 😄 But same error: 
```
 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=20&uuid=61018072286dcd431d74&v=4.1&sign=67dced75c2326f292aa5867a635af4fb (10.42.42.1) 65.81ms

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=21&uuid=61018072286dcd431d74&v=4.1&sign=79d7455d48afa0552205ace6c0c884e9
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload 10DD7D39E3A5710D8E6D87D285214BE9
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJZmwwRPzoJwOu8gFAidZ2yT9F2VBmqNqTqJcWuiX4YubOF2yg37JRyuu6tJU5i7kSS7mevrGF2WUpAllBDW/Q5l","t":1598477559,"sign":"bbb089a694774684"}

```

kueblc, 2020-08-27:
@Chuckame you cannot use Tasmota on a non-ESP device.

netpok, 2020-08-27:
Sadly you first need to port Tasmota to the Allwinner W600 platform (or any other one that your device uses), last time I checked the W600 only had a binary arduino platform for windows so I let it go.

To be honest that was a year ago at least so maybe it changed, but in my opinion it's cheaper to buy an esp based device (I don't know the market shares these days).

Chuckame, 2020-08-27:
> Sadly you first need to port Tasmota to the Allwinner W600 platform (or any other one that your device uses), last time I checked the W600 only had a binary arduino platform for windows so I let it go.

@netpok doable by OTA ? Or you don't know ?

> To be honest that was a year ago at least so maybe it changed, but in my opinion it's cheaper to buy an esp based device (I don't know the market shares these days).

@netpok Difficult to find on aliexpress, vendors are not saying "yes, it is an esp device, you can flash it !" 😄 Soooooo the only thing to do is to read all comments and search for someone that have already dissociated the device to know if it is an esp82xx 😢 My smart plugs are generic "Tuya" devices with power monitoring, cost only 7€