ct-Open-Source/tuya-convert #448: Can't connect to device

rebel7580, 2019-12-10:
I have Gosund WP3 devices I'm trying to convert to Tasmota.
Running tuya-convert on a RPI zero with Wifi Dongle, and serial console. I got my phone to connect to vtrust-flash at 10.42.42.26. Blue light on device is blinking. Here what I get when trying start_flash:

```
hvxl@rpia:~/tuya-convert$ ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... ss: invalid option -- 'H'
Available.
Checking UDP port 67... ss: invalid option -- 'H'
Available.
Checking TCP port 80... ss: invalid option -- 'H'
Available.
Checking TCP port 443... ss: invalid option -- 'H'
Available.
Checking UDP port 6666... ss: invalid option -- 'H'
Available.
Checking UDP port 6667... ss: invalid option -- 'H'
Available.
Checking TCP port 1883... ss: invalid option -- 'H'
Available.
Checking TCP port 8886... ss: invalid option -- 'H'
Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
[... 110 lines cut ...]
new client on port 443 from 10.42.42.40:7658
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.40:32448
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.40:62994
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
^C1575998548: mosquitto version 1.4.14 terminating
new client on port 443 from 10.42.42.40:5723
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.40:27153
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
new client on port 443 from 10.42.42.40:41079
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 104, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
^Cwlan0: AP-STA-DISCONNECTED d8:f1:5b:aa:bb:bb
wlan0: AP-STA-DISCONNECTED a4:08:ea:27:80:37
AP closed
Stopping DNSMASQ server...
```
I'm a Linux NOOB, so don't know what to make of any of the log info, although it look like a connection couldn't be made.

tonysprenk, 2019-12-10:
Have a look at the weblog in the Tuya-Convert folder. It looks like the device might not have a esp chip.

rebel7580, 2019-12-10:
That would be a surprise, since others in the Tasmota community have converted this device, but I'll look.
 My RPI is down right now to copy the SD image. I'll take a look as soon as it's back on line. Thanks

meingraham, 2019-12-10:
If new WP3 has switched to RealTek 😢

tonysprenk, 2019-12-10:
Could be the manufacturer has started using a different chip. @kueblc might be good to post the message about incompatible chip in the terminal to avoid unnecessary issues being opened.

meingraham, 2019-12-10:
Tuya has started making their Wi-Fi modules with RealTek chips instead of the Espressif ESP82xx. Manufacturers buy the modules from Tuya "off the shelf" with the Tuya firmware on board. Gosund doesn't really care or know the difference because the modules are pin and function compatible. They plug it onto their PCB and off they go.

rebel7580, 2019-12-10:
Great! :(   I guess I'm a little late...   I'll take one apart to confirm. They're useless to me anyway if I can't convert them.

tonysprenk, 2019-12-10:
Have a look at your weblog first. It will tell you if there’s a different chip in there.

rebel7580, 2019-12-10:
Will do. Thanks! If it's the RealTek, I'll post as such and close this. If it's still the ESP82xx, then I might still need some help (wishful thinking I guess!)

rebel7580, 2019-12-10:
I don't see a weblog file anywhere under ~/tua-convert. Just screenlog.0

kueblc, 2019-12-10:
It looks like you have a number of issues.

The reason you can't find the web log is because the web server never started
```
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
  File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1655
    async def _execute(
            ^
SyntaxError: invalid syntax
```
This is because your Python 3 is outdated

```
new client on port 443 from 10.42.42.26:50698
('could not establish sslpsk socket:', SSLError('No cipher can be selected.',))
```
This could be due to an older version of openssl

```
Checking UDP port 53... ss: invalid option -- 'H'
```
`-H` on `ss` is `--no-headers`, again could be due to outdated `ss`

kueblc, 2019-12-10:
`ss` added `-H` with 7a4559f67c8c4dcd6d8d361018ac239090713b69, in July of 2016. Make sure you're running a more recent version.
```
$ ss -v
ss utility, iproute2-ss180129
```

Minimum Python version is 3.5
```
$ python3 --version
Python 3.6.8
```

I'm not positive on the minimum OpenSSL version, but I'm running
```
$ openssl version
OpenSSL 1.1.1  11 Sep 2018
```

rebel7580, 2019-12-10:
OK, I'll need to check. I know python and openssl. Not sure about ss. Can I just apt-get install ss or is it provided by another package?

kueblc, 2019-12-10:
I believe `ss` is provided by `iproute2`

rebel7580, 2019-12-10:
I'm doing a (lenthy) update/upgrade now (which I should have done first). Then I'l check the versions.

rebel7580, 2019-12-11:
OpenSSL 1.0.1t  3 May 2016
Python 3.4.2
ss utility, iproute2-ss140804
Unfortunately, apt-get says they are the "newest" version.
I'm thinking I need to move up to an entirely new RPI distro to get the latest.

rebel7580, 2019-12-11:
While waiting for a 3.7.2 Python build and an SD card to be imaged to Buster, I cracked one open.
Doesn't look like the usual ESP card. The card that's there is a board with a shield, no markings on it that would support that it was an ESP.  So I'm guessing I'm out of luck.

meingraham, 2019-12-11:
I shield over the ESP chip is not unusual.

Can you post a photo if the Wi-Fi module?

rebel7580, 2019-12-11:
On second look, it MAY be a ESP, or whatever it is uses the same IO numbering.
I can see I00, I02, I04, I05, I013, RST, GND and 3v3
[image]

meingraham, 2019-12-11:
What's silk screened on the other side (TYxxxx?).

rebel7580, 2019-12-11:
T201-V1.0

meingraham, 2019-12-11:
Apparently T201 has an ESP8285 - https://electronics.stackexchange.com/a/404895

Or, at least it used to be

rebel7580, 2019-12-11:
Yeah, that's it all right. IO postions match.   Don't recall if Tasmota works with an 8285. Of course I haven't gotten very far on tuya-convert in any case.

meingraham, 2019-12-11:
Yes, the ESP8285 is just an ESP8266 with the flash memory integrated into the same IC. ESP8266 uses an external flash memory chip (and utilizes GPIO9/10 to interface). The ESP8285 thus frees up IO9/10.

Anyway, Tasmota supports it.

Now you need to find out if after updating Python you can get Tuya-Convert to flash it.

rebel7580, 2019-12-11:
I built and installed Python 3.7.2, it got farther, but still had python  issues (no tornado).  And I still haven't figured out how to update openssl and ss.   I think next I'll try the "Clean" distro I now have on an SD card.  Tomorrow I hope.

rebel7580, 2019-12-11:
Yay!   Success!
Worked with a new "Buster" distro; Python3, openssl, ss were at or newer than needed.

So my contrite apologies to @kueblc @meingraham @tonysprenk for wasting your time because I tried to do this on my existing RPI platform, even after reading that a clean build was recommended.

Thanks for your support to  dummies like me. :)

closed 2019-12-11 by rebel7580

kueblc, 2019-12-11:
Glad to hear you got it sorted out. I may want to integrate version checks as part of the install script to mitigate this situation in the future. Thanks for reporting your experience.

rebel7580, 2019-12-11:
Actually, only the first one has worked so far, the other ones I've tried all get into the mode of auto retrying until it fails 10 times. Not sure what my problem is this time.

kueblc, 2019-12-11:
Try the troubleshooting page in the wiki or post your latest log files and I may be able to help.

rebel7580, 2019-12-11:
@kueblc  Didn't see anything in troubleshooting that applied.  I rebooted,  cleared the logs before starting. Everything looked good (although the first time through, it asked to shut down dnsmasq and mqtt), so pressed enter to take next steps. Kept getting auto retry until it gave up. The psk log looks interesting, the rest didn't tell me anything. I'm puzzled since it worked once today. Not sure what I'm doing wrong.

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3952667/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3952668/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3952669/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3952671/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3952672/smarthack-mqtt.log)

kueblc, 2019-12-11:
Hmm unfortunately the logs aren't giving us any leads. Looks like the device never connected.

Can you confirm that you have connected your phone to `vtrust-flash` and that the device is in pairing/ez config mode (fast blink) when you press enter?

rebel7580, 2019-12-11:
Yep. Phone was connected, and device is blinking ~2times/second.  Just to be sure I had it in the right mode, I also tried the "slow mode" where device blinks once every few secs, but as expected that didn't work either.  Tried with two othe rdevices from the same batch - I bought a 4-pack - with same results. The 4th one I haven't tried yet since that's the one I opened.   My setup has the device about  18 inches from the RPI Zero with wifi adaptor, and the phone about the same distance in the opposite direction. Should be getting a good signal.

rebel7580, 2020-02-19:
FYI: Two months later, I finally got back to trying this again with the rest of the devices (3 of 4) that didn't seem to take the first time. All three upgraded perfectly this time! Not sure why I had problems before, but everything worked flawlessly this time. 

Thanks for the great upgrade process!