ct-Open-Source/tuya-convert #449: Stalling at Starting AP in a screen

bbalz, 2019-12-10:
I've tried flashing a few times and used the lite and full Raspian builds.  Each time I've run into this same exact issue. 

It clones, installs and when I try to start flash, it stalls at the 'Starting AP in a screen' and never shows anything further. 

[image]

It does broadcast the vtrust-flash SSID but the command prompt never progresses any further. 

I'm at a loss with what to do.  I've never had an issue with this process but it was months ago when it was working without a hiccup.  Since then I reflashed a new pi as noted above and I've even tried 2 different cards, 2 different versions of the Pi and no luck. 

Any help would be greatly appreciated.

kueblc, 2019-12-10:
You might be connected to the device via the same wifi card that's used for the AP. You need to use another connection (ethernet, USB, etc) since the wifi is needed to create the AP.

bbalz, 2019-12-10:
Oh geez, you're right.  I totally forgot about that when I did it the first time.  I'm sorry.  

Thank you a bajillion for the fast response.

closed 2019-12-10 by bbalz