ct-Open-Source/tuya-convert #450: Cant obtain IP from AP with any devices on Fedora WS

JonHeron, 2019-12-11:
Fedora30_64
Modified the install_prereq.sh for fedora
`#!/bin/bash

set -e

sudo dnf update
sudo dnf install -y git iw dnsmasq hostapd screen curl python-pip python3-pip python-setuptools python3-setuptools python-wheel python3-wheel python-devel python3-devel mosquitto haveged net-tools openssl-devel
sudo dnf groupinstall "Development Tools" "Development Libraries"

PY3_DEPENDENCIES="paho-mqtt pyaes tornado git+https://github.com/M4dmartig4n/sslpsk.git pycrypto"
PY2_DEPENDENCIES="git+https://github.com/M4dmartig4n/sslpsk.git pycrypto"

if python3 -c 'import sys; exit(0) if sys.version_info.major == 3 and sys.version_info.minor < 7 else exit(1)' ;
then
	sudo -H pip3 install $PY3_DEPENDENCIES
	sudo -H pip2 install $PY2_DEPENDENCIES
else
	sudo -H python3 -m pip install $PY3_DEPENDENCIES
	sudo -H python2 -m pip install $PY2_DEPENDENCIES
fi

echo "Ready to start upgrade"`
Modified the setup-ap.sh to properly start and stop NetworManager in fedora:
`#!/bin/bash

# Source config
. ../config.txt

setup () {
	wpa_supplicant_pid=$(pidof wpa_supplicant)
	if [ -n "$wpa_supplicant_pid" ]; then
		echo "Attempting to stop wpa_supplicant"
		sudo kill $wpa_supplicant_pid
	fi

	if test -d /etc/NetworkManager; then
		echo "Stopping NetworkManager..."
		sudo systemctl stop NetworkManager

	fi

	echo "Configuring AP interface..."
	sudo ip link set $WLAN down
	sudo ip addr add $GATEWAY/24 dev $WLAN
	sudo ip link set $WLAN up
	sudo ip route add 10.42.42.0/24 dev $WLAN src $GATEWAY
	sudo ip route add 255.255.255.255 dev $WLAN

	echo "Starting DNSMASQ server..."
	sudo dnsmasq \
		--no-resolv \
		--interface=$WLAN \
		--bind-interfaces \
		--listen-address=$GATEWAY \
		--dhcp-range=10.42.42.10,10.42.42.40,12h \
		--address=/#/$GATEWAY

	echo "Starting AP on $WLAN..."

	# Read hostapd.conf with interface from stdin for
	# backward compatibility (hostapd < v2.6). See #398
	printf "$(cat hostapd.conf)\ninterface=$WLAN" | sudo hostapd /dev/stdin
}

cleanup () {
	sudo pkill hostapd
	echo "AP closed"

	echo "Stopping DNSMASQ server..."
[... 779 lines cut ...]
wlp2s0: AP-STA-DISCONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-CONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-DISCONNECTED a4:e4:b8:96:39:eb
wlp2s0: AP-STA-CONNECTED a4:e4:b8:96:39:eb
wlp2s0: AP-STA-DISCONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-CONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-DISCONNECTED a4:e4:b8:96:39:eb
wlp2s0: AP-STA-DISCONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-CONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-STA-CONNECTED a4:e4:b8:96:39:eb
wlp2s0: interface state ENABLED->DISABLED
wlp2s0: AP-STA-DISCONNECTED a4:e4:b8:96:39:eb
wlp2s0: AP-STA-DISCONNECTED ec:fa:bc:44:ce:95
wlp2s0: AP-DISABLED 
wlp2s0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...`

dnsmasq version dnsmasq-2.80-10.fc30.x86_64

Not sure what to try next?
Cheers,
Jon

JonHeron, 2019-12-11:
Whoa, I screwed up the formatting there somehow. 
Too bad the code tags dont truncate the code, or am I doing it wrong?
I will put the log files in a zip file and upload them instead...
Cheers,
Jon

JonHeron, 2019-12-11:
[smarthack-logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3948001/smarthack-logs.zip)

JonHeron, 2019-12-11:
[modifiedFiles.zip](https://github.com/ct-Open-Source/tuya-convert/files/3948007/modifiedFiles.zip)

JonHeron, 2019-12-11:
lol I cant fix the big fonts, not sure what is happening there, please bear with me...
Cheers,
Jon

kueblc, 2019-12-11:
Use triple backtick (` ``` `) to enclose code blocks. Also note the preview feature.

JonHeron, 2019-12-11:
Thanks for that!
Jon

JonHeron, 2019-12-11:
I did some more playing with this today and checked some of the versions I am running for reference.
Python 2.7.17
Python3 3.7.5
OpenSSL 1.1.1d FIPS  10 Sep 2019
From pip:
Name: sslpsk Version: 1.0.0

I am thinking I will see what is happening with the AP using wireshark.
Do you think this is looking in the right direction?
Cheers,
Jon

kueblc, 2019-12-11:
How about your `dnsmasq` version? This acts as our DHCP server so if you're unable to get an IP, it is likely related to `dnsmasq`.

Possibly related to #424

JonHeron, 2019-12-11:
$ dnsmasq -version
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus no-i18n IDN2 DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect inotify dumpfile

JonHeron, 2019-12-11:
Thanks I think you may be right, it appears it is not even starting up.
sudo systemctl status dnsmasq
[sudo] password for jonnyslaptop: 
● dnsmasq.service - DNS caching server.
   Loaded: loaded (/usr/lib/systemd/system/dnsmasq.service; disabled; vendor pr>
   Active: failed (Result: exit-code) since Wed 2019-12-11 15:11:59 EST; 5min a>
  Process: 10314 ExecStart=/usr/sbin/dnsmasq -k (code=exited, status=2)
 Main PID: 10314 (code=exited, status=2)

Dec 11 15:11:59 jonnyslaptop-localdomain systemd[1]: Started DNS caching server>
Dec 11 15:11:59 jonnyslaptop-localdomain dnsmasq[10314]: dnsmasq: failed to cre>
Dec 11 15:11:59 jonnyslaptop-localdomain systemd[1]: dnsmasq.service: Main proc>
Dec 11 15:11:59 jonnyslaptop-localdomain dnsmasq[10314]: failed to create liste>
Dec 11 15:11:59 jonnyslaptop-localdomain systemd[1]: dnsmasq.service: Failed wi>
Dec 11 15:11:59 jonnyslaptop-localdomain dnsmasq[10314]: FAILED to start up
lines 1-12/12 (END)
I will dig into this deeper...
PS I dont seem to have a back tick on my Asus laptop keyboard...
Cheers,
Jon

JonHeron, 2019-12-11:
Strange, even though it says it cant start cause port 53 is in use, it appears there are 2 instances of dnsmasq running...
$sudo lsof -i:53
COMMAND  PID    USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
dnsmasq 1242 dnsmasq    5u  IPv4  35524      0t0  UDP jonnyslaptop-localdomain:domain 
dnsmasq 1242 dnsmasq    6u  IPv4  35525      0t0  TCP jonnyslaptop-localdomain:domain (LISTEN)

Lol wtf.
On Fedora I think NetworkManger uses dnsmasq so perhaps that is the conflict, however stopping Network manager should resolve the conflict but it doesnt...
Cheers,
Jon

JonHeron, 2019-12-14:
I got dnsmasq logging to a file now, it appears it is starting up normally though eh?
``` Dec 13 20:42:04 dnsmasq[23176]: started, version 2.80 cachesize 150
Dec 13 20:42:04 dnsmasq[23176]: compile time options: IPv6 GNU-getopt DBus no-i18n IDN2 DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Dec 13 20:42:04 dnsmasq-dhcp[23176]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Dec 13 20:42:04 dnsmasq-dhcp[23176]: DHCP, sockets bound exclusively to interface wlp2s0
Dec 13 20:42:04 dnsmasq[23176]: read /etc/hosts - 3 addresses
Dec 13 20:43:47 dnsmasq[23176]: exiting on receipt of SIGTERM```
I tried removing the lines you suggested in #424 with no joy either.
Cheers,
Jon

billtown, 2020-01-09:
if dnsmasq isn't logging any attempts, but your ap is seen, likely it's your firewall.  as a stop gap

Check dmesg and look for REJECT and SPT=68 DPT=67 from 0.0.0.0  aka your missing dhcp requests.....

on fedora 30

```sudo systemctl stop firewalld.service```

Do you're thing upgrade of stuff or retest dnsmaq

Then turn your firewall back on when you are finished.

```sudo systemctl start firewalld.service when you are finished.```

JonHeron, 2020-01-12:
That was it!
Thank you sir!
I feel dopey for not trying this sooner...
Cheers,
Jon

closed 2020-01-12 by JonHeron