ct-Open-Source/tuya-convert #451: 3A Smart Wall outlet with 2 USB...instead of connecting to AP it openes own...help

Darkyputz, 2019-12-12:
Hello...
Since i got my initial tuya plug converted to tasmota, i keep adding those devices to my home for my home automation server...
But now i ran into a Wall mount single outlet with 2 USB from 3A, that behaves pretty strange.
My Ubuntu is doing its magc with AP and such, but besides the vrtrust AP it shows a second one up, Called SmartHome-B73D...which seems to be to the Outlet.
I googled around and found on the Tasmota wiki, that people were able to get a 3A Smarthome Outlet flashed...but i am not quite sure how i can do that and i am afraid that i accidentally grabbed a Zigbee...
How can i determine quick and easy if i have a tuya convert compatible one...
Thx for any ideas...
The main Chip that is present on the internals states E469716 and the boards "name" printed on it is: XN-MG1WCZ+2USB-SB-T1.1 2019.05.10

kueblc, 2019-12-12:
There are two pairing modes, EZ config (fast blink) and AP config (slow blink). If you see an AP you are in AP config mode. You need to use EZ config for `tuya-convert`.

Darkyputz, 2019-12-12:
Hello and thx for the reply...so do you mean that this device can both?
Or that i am out of luck?
Are there AP config mode flash options that you are aware of?

kueblc, 2019-12-12:
As far as I know all Tuya ESP devices support both. You are just in the wrong mode. Try putting the device in pairing mode again.

Darkyputz, 2019-12-12:
Is there a way to determine if this bugger IS a tuya device?
I couldnt find anything about the chip on the board...

kueblc, 2019-12-12:
What app does it use? That could tell us.

Darkyputz, 2019-12-12:
App is called "Hui Home"...and the manufacturer is: https://3asmarthome.com/
As soon i find my usb cable i will post a pic of the board

Darkyputz, 2019-12-12:
It is this outlet/switch https://www.amazon.com/Smart-Power-Outlet-Control-Switch/dp/B07SGMB4BM/ref=sr_1_3?crid=3H2EQOHMNTDZE&keywords=tuya+smart+plug&qid=1576178720&sprefix=tuya+%2Cspecialty-aps%2C151&sr=8-3

Darkyputz, 2019-12-12:
In the add it says: control lights or other appliances through Smart Life or Tuya APP from anywhere
But who knows if they just copied that from somewhere
It is just strange that is can flash every darn plug i have, but this is therapy resistant

kueblc, 2019-12-12:
Right, looks like a Tuya device. Have you tried it with Smart Life?

kueblc, 2019-12-12:
Are you running the latest `tuya-convert`? Have you checked your log files?

Darkyputz, 2019-12-12:
I read i am not supposed to use the tuya app since i run in chances that it updates the firmware and then is locked for update again...is that not true anymore?
Will check out which version i use...

kueblc, 2019-12-12:
Yes it is safer that you do not use the app, I am only asking if you have.

Darkyputz, 2019-12-12:
Yup, not touched that crap...and was successful right straight with the other devices so far...so i was afraid that i might have ended with a ZigBee (what ever that is) or so...
So will try again today, with fresh updated tuya convert...

Darkyputz, 2019-12-13:
Soooooo...here is the magic from the log files...
Hope it tells you something
`payload 89470C75CE99EC3E706A2E5279C4A469
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJbsZH1sCGrEXdhgKEZKSgvp9F2VBmqNqTqJcWuiX4YubFEuE1wjdn0f/cutR8N9K1O7mevrGF2WUpAllBDW/Q5l","t":1576153722,"sign":"9fcabd7d38bf3005"}
[32m[I 191212 07:28:42 web:2246][m 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=20&uuid=9588ded6a62ae7e5&v=4.1&sign=46604829ad99c04dc2a4310770bfc29d (10.42.42.1) 45.14ms

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=21&uuid=9588ded6a62ae7e5&v=4.1&sign=6117692cb14c4acaebe5d67880f6583c
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37
`

kueblc, 2019-12-13:
> WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

You have your answer, that's too bad. Maybe you can find a similar model that still uses the ESP.

closed 2019-12-13 by kueblc