ct-Open-Source/tuya-convert #452: Unable to flash Arlec PB190HA smart plug

hell1337hound, 2019-12-12:
Hi i have two arlec PB190HA smart plugs from bunnings and i'm unable to flash them with Tuya-Convert. I've managed to flash an arlec powerboard.

`Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
RTNETLINK answers: File exists
Starting AP on wlp2s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp2s0 with hwaddr 18:cf:5e:ed:7a:a3 and ssid "vtrust-flash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
wlp2s0: STA 34:29:12:38:d4:00 IEEE 802.11: authenticated
wlp2s0: STA 34:29:12:38:d4:00 IEEE 802.11: associated (aid 1)
wlp2s0: AP-STA-CONNECTED 34:29:12:38:d4:00
wlp2s0: STA 34:29:12:38:d4:00 RADIUS: starting accounting session C21CC582432580B9
wlp2s0: STA 34:29:12:38:d4:00 WPA: pairwise key handshake completed (RSN)
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: authenticated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: associated (aid 2)
wlp2s0: AP-STA-CONNECTED d8:f1:5b:92:aa:d7
wlp2s0: STA d8:f1:5b:92:aa:d7 RADIUS: starting accounting session 1098184EA398748C
wlp2s0: STA d8:f1:5b:92:aa:d7 WPA: pairwise key handshake completed (RSN)
wlp2s0: AP-STA-DISCONNECTED d8:f1:5b:92:aa:d7
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: disassociated
wlp2s0: STA d8:f1:5b:92:aa:d7 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlp2s0: interface state ENABLED->DISABLED
wlp2s0: AP-STA-DISCONNECTED 34:29:12:38:d4:00
wlp2s0: AP-DISABLED 
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
`

`Listening on port 80
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 160, in <module>
    main()
  File "./fake-registration-server.py", line 156, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.6/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 160, in <module>
    main()
  File "./fake-registration-server.py", line 156, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.6/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)`

Seems whatever Tuya-Convert is doing its wiping the config to the plug and the app can’t find it so thats quite interesting

I tried flashing the devices before connecting one to the app, then i've connected it to the app and i still can't get it to flash

kueblc, 2019-12-12:
Use triple backtick (` ``` `) to enclose multiline code blocks and preserve formatting for logs and such.

It looks like you are using a very old version of `tuya-convert`. If you installed using the recommended method via `git`, you can update by doing `git pull`.

kueblc, 2020-01-25:
Closing due to lack of response. Let me know if you have issues after updating to the current version.

closed 2020-01-25 by kueblc