ct-Open-Source/tuya-convert #453: Flashing failed but not before killing my device

tam481, 2019-12-13:
Hello
I've managed to download the software and follow the instructions which were fairly straightforward. However, upon attempting to flash my 2NICE power socket, I got this

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.............../start_flash.sh: line 106:   722 Terminated              ./smartconfig/main.py
...........................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...

Now the power button on the socket turns blue briefly when I plug it in but then nothing happens and has disappeared from the original app.

Please help.

kueblc, 2019-12-13:
See [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting) in the wiki, note "Device appears bricked" and "vtrust-recovery SSID is visible"

tam481, 2019-12-13:
You're a star. Apologies for not checking the troubleshooting part before posting.
Manged to get Tasmota installed now :-)

closed 2019-12-13 by tam481