ct-Open-Source/tuya-convert #455: no log files in the 'scripts' folder.

lustyffh, 2019-12-13:
Hi,

I can't flash my smart socket (SS1002). And I don't even have any logs in the 'scripts' folder that you normally ask for. The only generated file is 'screenlog.0' which I [attach here](https://gist.github.com/lustyffh/035c075af43943b2cf4fd00de618db5e). The process hangs at:

```
root@ubuntu:~/tuya-convert# ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.............................
Device did not appear with the intermediate firmware

```

kueblc, 2019-12-13:
Is this your socket? `28:6d:cd:2d:48:77`

kueblc, 2019-12-13:
If so, it is a [Winner SoC](https://aruljohn.com/mac/286DCD) which is unfortunately not supported. We can only install ESP82xx firmware.

lustyffh, 2019-12-14:
> 
> 
> If so, it is a [Winner SoC](https://aruljohn.com/mac/286DCD) which is unfortunately not supported. We can only install ESP82xx firmware.

Thank you! Indeed, it's my socket =(. What a pity!

closed 2019-12-14 by lustyffh

carmolim, 2020-01-13:
Just got one of those, it arrived today. There's something to make it work locally only?

kueblc, 2020-01-13:
You could try [tuyapi](https://github.com/codetheweb/tuyapi/)

carmolim, 2020-01-13:
But using the API the plug will still be connected to the cloud. Is there any other way that doesn't depend on the cloud?

kueblc, 2020-01-13:
`tuyapi` controls your device over LAN, not the cloud. You can take it a step further and block the device from contacting the cloud via network isolation.