ct-Open-Source/tuya-convert #456: failed to find water boiler switch

syntithenai, 2019-12-15:
Hi,
  I'm trying to flash an unbranded 20A glass water boiler switch. 

https://www.ebay.com.au/itm/143345057272
[image]

I did flash one successfully and then ordered six more from the same supplier but the new ones don't seem to work.

In trying to flash them I git pulled the latest tuya-convert (and stupidly didn't note which version worked for the first switch)

Neither the older version that worked with the first switch nor the latest version work with the new switches.

The script retries all 120 times until it times out and quits.
Log files attached/included below.

any help much appreciated

cheers
 
Steve

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3964773/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3964774/smarthack-web.log)

## smarthack-wifi.log

wlp2s0: AP-STA-CONNECTED 68:57:2d:56:67:4e
wlp2s0: AP-STA-CONNECTED 68:57:2d:56:67:4e
wlp2s0: AP-STA-DISCONNECTED d0:b1:28:14:1b:91
wlp2s0: AP-STA-CONNECTED d0:b1:28:14:1b:91
wlp2s0: AP-STA-CONNECTED 68:57:2d:56:67:4e
wlp2s0: interface state ENABLED->DISABLED
wlp2s0: AP-STA-DISCONNECTED d0:b1:28:14:1b:91
wlp2s0: AP-STA-DISCONNECTED 68:57:2d:56:67:4e
wlp2s0: AP-DISABLED 
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp2s0...
Configuration file: /dev/stdin
Using interface wlp2s0 with hwaddr 9c:b6:d0:60:b2:09 and ssid "vtrust-flash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
wlp2s0: AP-STA-CONNECTED d0:b1:28:14:1b:91
wlp2s0: AP-STA-CONNECTED 68:57:2d:56:67:4e
wlp2s0: interface state ENABLED->DISABLED
wlp2s0: AP-STA-DISCONNECTED 68:57:2d:56:67:4e
wlp2s0: AP-STA-DISCONNECTED d0:b1:28:14:1b:91
wlp2s0: AP-DISABLED 
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...

## smarthack-mqtt.log

1576401211: mosquitto version 1.4.15 (build date Wed, 19 Jun 2019 12:01:14 -0300) starting
1576401211: Using default config.
1576401211: Opening ipv4 listen socket on port 1883.
1576401211: Opening ipv6 listen socket on port 1883.
^C1576401335: mosquitto version 1.4.15 terminating

## smarthack-udp.log

Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
10.42.42.12 {'ip': '10.42.42.12', 'gwId': '91bfeacf5b5159cf', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyqwddxxv8ukxkj', 'version': '3.3'}
^C

davidgurr, 2019-12-18:
I have an identical boiler switch and have the very same problem. Keen to hear any potential solutions or suggestions!

pashok2398, 2019-12-22:
Same here... i guess its after the last firmwware update...

kueblc, 2019-12-22:
It looks like this device may not be using an ESP82xx. Can anyone confirm?

davidgurr, 2019-12-22:
To get access to the main circuit board (with the control chip l you have to desolder a secondary board that shields it. I don't trust my soldering technique enough to attempt that. 

Anyone else?

davidgurr, 2020-01-15:
Just had it confirmed by the vendor that it's using the RTL8710BN rather than ESP82xx. So no Tasmota option :-(

kueblc, 2020-01-15:
Too bad! Hope you can find a suitable replacement.

closed 2020-01-15 by kueblc

radinsky, 2021-02-01:
Can maybe somebody recommend supported (esp) touch glass boiler switch (wide)? Thanks!