ct-Open-Source/tuya-convert #457: Issue Flashing ZooZee SA102 Power Monitoring Plug

war1000, 2019-12-16:
Hi Everyone, hope you can help me with this. I successfully flashed 3 of these plug and got stuck on the 4th one. Here are my log files.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3969396/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3969397/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3969398/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3969399/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3969400/smarthack-web.log)

[image]

kueblc, 2019-12-16:
Try [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting), looks like your device is running the intermediate firmware so all you have to do is plug the device in while `start_flash` is still running. You do not need to pair the device, just press enter to continue.

war1000, 2019-12-16:
> Try [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting), looks like your device is running the intermediate firmware so all you have to do is plug the device in while `start_flash` is still running. You do not need to pair the device, just press enter to continue.

thank you for replying. I did try what you said, but nothing changes, it keeps resending the packets. I unplugged and plugged back the device while start_flash is running and i press enter. It still keep resending.
 

I also checked the troubleshooting link. But don't know what else to try.

kueblc, 2019-12-16:
Like I said, you don't need to pair the device again, it is already paired and running the intermediate, so resending smart config is not important.

Can you check to see if there is a `vtrust-recovery` SSID that appears a minute or two after plugging the device in? Can you verify that the device buttons and lights do not function?

war1000, 2019-12-16:
> Like I said, you don't need to pair the device again, it is already paired and running the intermediate, so resending smart config is not important.
>
> Can you check to see if there is a [...]

This is the process I am following:

1. Turn on the raspberrypi and ssh to it.
2. go to the tuya-convert folder and run ./start_flash.sh
3. connect phone to the vtrust-flash and hit enter on the keyboard
4. this is where the smartconfig packet loop starts

Where do I check for the vtrust-recovery SSID? before hitting enter on the keyboard?

Also yes the device buttons and lights do not work when plugged in right now.

kueblc, 2019-12-16:
You should be able to see `vtrust-recovery` when the device is plugged in and `start_flash` is not running. This will confirm that the device is running the intermediate firmware.

kueblc, 2019-12-16:
> Also yes the device buttons and lights do not work when plugged in right now.

This is consistent with the device running the intermediate firmware. The device will attempt to connect to `vtrust-flash` and create a `vtrust-recovery` AP if that fails.

war1000, 2019-12-16:
> > Also yes the device buttons and lights do not work when plugged in right now.
> 
> This is consistent with the device running the intermediate firmware. The device will attempt to connect to `vtrust-flash` and create a `vtrust-recovery` AP if that fails.

No I don't see the vtrust-flash. i rebooted the pi and did not start the convert script. unplugged the device and plugged it back in. waited for a few mins. no vtrust-recovery. is the device bricked?

Tenebruspt, 2019-12-17:
I had the same exact issue with 2 out of 4 plugs:
 - Followed all troubleshooting steps, but did not work
 - plug does not connect to vtrust-flash
 - curl commands fail
 - vtrust-recovery does not show up.

What I did notice is that instead of vtrust-recovery, a secured hidden network shows up. I am unable to connect to it as I don't have a password.

I am sure this hidden network is from the device. I never see it unless I power up the plugs and can consistently make it show up a couple of minutes after plugging it in. Network strength is also consistent with it being right next to my machine.

war1000, 2019-12-17:
I do not even get any other network. Seems like bricked to me. Unless anyone has any other recommendation...

war1000, 2019-12-31:
Guys, just as an update. I bought two new plugs, reformatted the sd card and tried redoing the entire steps. I am still stuck at the Smartconfig complete. Resending SmartConfig Packets step. Anything else I can do?

kueblc, 2019-12-31:
@war1000 you'll need to provide logs to figure out what's not working.

war1000, 2019-12-31:
> @war1000 you'll need to provide logs to figure out what's not working.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4013037/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4013038/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4013040/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4013041/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4013042/smarthack-wifi.log)

kueblc, 2019-12-31:
> ('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))

This is from your phone, no problem here

> WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

This is your problem, found in your web log. Unfortunately you won't be able to install custom firmware like Tasmota on this device. Better luck next time!

closed 2019-12-31 by kueblc

war1000, 2019-12-31:
> This is from your phone, no problem here
>
>
> This is your problem, found in your web log. Unfortunately you won't be able to install custom firmware like Tasmota on this device. Better luck next time!

Thank you for the reply. I was afraid of this. The new switches I got are the SA101. I didn't realize till last night that the chips are different than the SA102. Ah well....more cloud apps...its like no way out....