ct-Open-Source/tuya-convert #458: tuya-convert terminates on Gosund SP1

martinlang83, 2019-12-18:
Issue: Cant convert device. tuya-covnert terminates

Device: Gosund SP1
New device, never connected to the APP
Device Firmware unknown
tuya-convert, master branch, latest.

Logfiles attached.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3980528/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3980529/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3980531/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3980532/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3980533/smarthack-wifi.log)

Output of script:

sudo ./start_flash.sh
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 1s. ../start_flash.sh: line 106: 10133 Terminated              ./smartconfig/main.py
..............................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]

Exiting...

martinlang83, 2019-12-18:
Device: 
Raspberry Pi 3b+
Raspbian stretch

OpenSSL: OpenSSL 1.1.0l  10 Sep 2019

kueblc, 2020-01-22:
It is not entirely clear why the process is failing unfortunately. The device seems to have connected, the PSK was generated successfully, and the device made the proper registration calls.

It did not respond to the upgrade command, which suggests only that the command has changed or some other validation was implemented in a newer firmware.

If you have the ability to flash by wire, please make a backup of the stock firmware first, that we can use to understand what is happening under the hood.

martinlang83, 2020-01-23:
thanks for feedback. do you have some instructions for the wire backup? I can try to do it.
regards

kueblc, 2020-01-23:
The Tasmota docs include a tutorial for the [hardware prep](https://tasmota.github.io/docs/#/installation/Hardware-Preparation) and subsequent [backup and flashing](https://tasmota.github.io/docs/#/installation/Flashing?id=backup-firmware-optional-step) using `esptool.py` and an FTDI or similar USB to serial device.

baracoder, 2020-06-15:
I had the same problem and flashed mine manually.
Here is a backup of the firmware https://zebar.de/owncloud/index.php/s/Ky9D2GwKAcjAmwH