ct-Open-Source/tuya-convert #459: Stuck after failed firmware backup

TaakoMagnusen, 2019-12-20:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

*Did you have issues getting a device to flash?**
Yes

Teckin SP22 (Old Model, bought in March 2019).

Logs

smarthack-mqtt.log
```
pi@raspberrypi:~/tuya-convert/scripts $ cat smarthack-mqtt.log 
1576865605: mosquitto version 1.5.7 starting
1576865605: Using default config.
1576865605: Opening ipv4 listen socket on port 1883.
1576865605: Opening ipv6 listen socket on port 1883.
1576865777: mosquitto version 1.5.7 starting
1576865777: Using default config.
1576865777: Opening ipv4 listen socket on port 1883.
1576865777: Opening ipv6 listen socket on port 1883.
^C1576866033: mosquitto version 1.5.7 terminating
1576866047: mosquitto version 1.5.7 starting
1576866047: Using default config.
1576866047: Opening ipv4 listen socket on port 1883.
1576866047: Opening ipv6 listen socket on port 1883.
1576866155: mosquitto version 1.5.7 starting
1576866155: Using default config.
1576866155: Opening ipv4 listen socket on port 1883.
1576866155: Opening ipv6 listen socket on port 1883.
1576866204: New connection from 10.42.42.28 on port 1883.
1576866204: New client connected from 10.42.42.28 as 35004736b4e62d4c5d1d (c1, k30, u'35004736b4e62d4c5d1d').
1576866204: Will message specified (107 bytes) (r0, q1).
1576866204:     tuya/smart/will
1576866204: Sending CONNACK to 35004736b4e62d4c5d1d (0, 0)
1576866204: Received SUBSCRIBE from 35004736b4e62d4c5d1d
1576866204:     smart/device/in/35004736b4e62d4c5d1d (QoS 0)
1576866204: 35004736b4e62d4c5d1d 0 smart/device/in/35004736b4e62d4c5d1d
1576866204: Sending SUBACK to 35004736b4e62d4c5d1d
1576866209: Received PUBLISH from 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/out/35004736b4e62d4c5d1d', ... (80 bytes))
1576866209: Received PUBLISH from 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/out/35004736b4e62d4c5d1d', ... (81 bytes))
1576866212: New connection from 127.0.0.1 on port 1883.
1576866212: New client connected from 127.0.0.1 as fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (c1, k60).
1576866212: No will message specified.
1576866212: Sending CONNACK to fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (0, 0)
1576866212: Received PUBLISH from fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (d0, q0, r0, m0, 'smart/device/in/35004736b4e62d4c5d1d', ... (147 bytes))
1576866212: Sending PUBLISH to 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/in/35004736b4e62d4c5d1d', ... (147 bytes))
1576866212: Received DISCONNECT from fb78bbf1-1d35-445d-8f5a-4a1f31f12150
1576866212: Client fb78bbf1-1d35-445d-8f5a-4a1f31f12150 disconnected.
1576866254: Client 35004736b4e62d4c5d1d has exceeded timeout, disconnecting.
1576866254: Socket error on client 35004736b4e62d4c5d1d, disconnecting.
1576866456: mosquitto version 1.5.7 starting
1576866456: Using default config.
1576866456: Opening ipv4 listen socket on port 1883.
1576866456: Opening ipv6 listen socket on port 1883.
1576871173: mosquitto version 1.5.7 starting
1576871173: Using default config.
1576871173: Opening ipv4 listen socket on port 1883.
1576871173: Opening ipv6 listen socket on port 1883.
1576871940: mosquitto version 1.5.7 starting
1576871940: Using default config.
1576871940: Opening ipv4 listen socket on port 1883.
1576871940: Opening ipv6 listen socket on port 1883.
1576872138: mosquitto version 1.5.7 starting
1576872138: Using default config.
[... 311 lines cut ...]
[I 191220 19:51:11 web:2246] 302 GET /generate_204 (10.42.42.20) 2.50ms
Listening on 10.42.42.1:80
Listening on 10.42.42.1:80
[I 191220 20:02:21 web:2246] 302 GET /generate_204 (10.42.42.20) 4.28ms
[I 191220 20:02:24 web:2246] 302 GET /generate_204 (10.42.42.20) 2.48ms
[I 191220 20:02:24 web:2246] 200 GET / (10.42.42.20) 4.20ms
[I 191220 20:02:25 web:2246] 302 GET /favicon.ico (10.42.42.20) 2.48ms
[I 191220 20:02:25 web:2246] 304 GET / (10.42.42.20) 2.78ms
[I 191220 20:02:25 web:2246] 302 GET /generate_204 (10.42.42.20) 3.03ms
[I 191220 20:02:26 web:2246] 302 GET /generate_204 (10.42.42.20) 2.66ms
```

So i purchased these Teckin SP22 (USA) smart plugs with energy monitoring early in 2019 but discovered I was too late and they had the updated firmware that couldnt do OTA. I tried cracking one open and soldering to it to flash over FTDI but there was some flubbing and it didnt work right, not sure what went wrong.

Anyway I still have one good one and heard that the new Tuya Convert can handle them now. (these are not the new Teckins without an ESP chip).

I followed the instructions in DigiBlur's new tuya convert video and I got to the point where it said "Successfuly found device on 10.42.42.42 and started a backup but it took a long time and i noticed it wasnt showing any transferred bytes so i CTRL-C'd out and rebooted. But now when I plug in my Teckin device I cant get it int othe flashing mode. It looks like the firmware has been modified as when i plug it in the LED blinks (somewhat sporadically, not necessarily an even blinking pattern) and eventually it stops blinking entirely!

I tried multiple times to get the ./start_flash to work again but it seems like it cant get the device (smartconfig retires infinitely).

I don't know when the logs are generated, or overwritten, but i attached the logs i have on hand. They may not be the logs with the attempt that failed to do firmware backup

the firmware backup is 0 bytes by the way

Is the device bricked?

kueblc, 2019-12-20:
> [I 191220 18:23:30 web:2246] 206 GET /files/upgrade.bin (10.42.42.28) 910.55ms

It looks like it successfully flashed the intermediate firmware. It's most likely still running the intermediate firmware, since you said you can't put it into pairing mode. The intermediate firmware serves no function other than backing up and flashing a new image.

kueblc, 2019-12-20:
I would try restarting `start_flash` and then plugging the device back in. Open a new terminal and run the `firmware_picker` directly.

TaakoMagnusen, 2019-12-20:
I was able to grab the only device that i cracked open. It seems to be one stock tuya. so I tried the procedure and here is an image of it stuck at the backup: https://imgur.com/a/bcpNqMx

So same exact deal. It can't transfer.

I logged in with another shell while its in this process and here are the logs from this one

smarthack-mqtt.log
```
pi@raspberrypi:~/tuya-convert-2/scripts $ cat smarthack-mqtt.log 
1576865605: mosquitto version 1.5.7 starting
1576865605: Using default config.
1576865605: Opening ipv4 listen socket on port 1883.
1576865605: Opening ipv6 listen socket on port 1883.
1576865777: mosquitto version 1.5.7 starting
1576865777: Using default config.
1576865777: Opening ipv4 listen socket on port 1883.
1576865777: Opening ipv6 listen socket on port 1883.
^C1576866033: mosquitto version 1.5.7 terminating
1576866047: mosquitto version 1.5.7 starting
1576866047: Using default config.
1576866047: Opening ipv4 listen socket on port 1883.
1576866047: Opening ipv6 listen socket on port 1883.
1576866155: mosquitto version 1.5.7 starting
1576866155: Using default config.
1576866155: Opening ipv4 listen socket on port 1883.
1576866155: Opening ipv6 listen socket on port 1883.
1576866204: New connection from 10.42.42.28 on port 1883.
1576866204: New client connected from 10.42.42.28 as 35004736b4e62d4c5d1d (c1, k30, u'35004736b4e62d4c5d1d').
1576866204: Will message specified (107 bytes) (r0, q1).
1576866204:     tuya/smart/will
1576866204: Sending CONNACK to 35004736b4e62d4c5d1d (0, 0)
1576866204: Received SUBSCRIBE from 35004736b4e62d4c5d1d
1576866204:     smart/device/in/35004736b4e62d4c5d1d (QoS 0)
1576866204: 35004736b4e62d4c5d1d 0 smart/device/in/35004736b4e62d4c5d1d
1576866204: Sending SUBACK to 35004736b4e62d4c5d1d
1576866209: Received PUBLISH from 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/out/35004736b4e62d4c5d1d', ... (80 bytes))
1576866209: Received PUBLISH from 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/out/35004736b4e62d4c5d1d', ... (81 bytes))
1576866212: New connection from 127.0.0.1 on port 1883.
1576866212: New client connected from 127.0.0.1 as fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (c1, k60).
1576866212: No will message specified.
1576866212: Sending CONNACK to fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (0, 0)
1576866212: Received PUBLISH from fb78bbf1-1d35-445d-8f5a-4a1f31f12150 (d0, q0, r0, m0, 'smart/device/in/35004736b4e62d4c5d1d', ... (147 bytes))
1576866212: Sending PUBLISH to 35004736b4e62d4c5d1d (d0, q0, r0, m0, 'smart/device/in/35004736b4e62d4c5d1d', ... (147 bytes))
1576866212: Received DISCONNECT from fb78bbf1-1d35-445d-8f5a-4a1f31f12150
1576866212: Client fb78bbf1-1d35-445d-8f5a-4a1f31f12150 disconnected.
1576866254: Client 35004736b4e62d4c5d1d has exceeded timeout, disconnecting.
1576866254: Socket error on client 35004736b4e62d4c5d1d, disconnecting.
1576866456: mosquitto version 1.5.7 starting
1576866456: Using default config.
1576866456: Opening ipv4 listen socket on port 1883.
1576866456: Opening ipv6 listen socket on port 1883.
1576871173: mosquitto version 1.5.7 starting
1576871173: Using default config.
1576871173: Opening ipv4 listen socket on port 1883.
1576871173: Opening ipv6 listen socket on port 1883.
1576871940: mosquitto version 1.5.7 starting
1576871940: Using default config.
1576871940: Opening ipv4 listen socket on port 1883.
1576871940: Opening ipv6 listen socket on port 1883.
1576872138: mosquitto version 1.5.7 starting
1576872138: Using default config.
1576872138: Opening ipv4 listen socket on port 1883.
1576872138: Opening ipv6 listen socket on port 1883.
1576874586: mosquitto version 1.5.7 starting
1576874586: Using default config.
1576874586: Opening ipv4 listen socket on port 1883.
1576874586: Opening ipv6 listen socket on port 1883.
1576874810: New connection from 10.42.42.22 on port 1883.
1576874810: New client connected from 10.42.42.22 as 35808047b4e62d1d56f8 (c1, k30, u'35808047b4e62d1d56f8').
1576874810: Will message specified (107 bytes) (r0, q1).
[... 424 lines cut ...]
Content-Length: 37

payload {"subId":null}
Answer tuya.device.upgrade.get
reply {"t":1576874811,"e":false,"success":true,"result":{"auto":true,"type":0,"size":"505867","version":"9.0.0","url":"http://10.42.42.1/files/upgrade.bin","md5":"9aeeb1f7b6dacb6f251445aac49181a9"}}
[I 191220 20:46:51 web:2246] 200 POST /gw.json?a=tuya.device.upgrade.silent.get&gwId=35808047b4e62d1d56f8&t=1576874811&v=4.1&sign=370a11396806ef337454a1344cf741f9 (10.42.42.22) 5.66ms

POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=35808047b4e62d1d56f8&t=1576874811&sign=a19bbca26951454bed59f388a9354e8b
Host: 10.42.42.1
Ty-Ua: HW
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 69

payload {"upgradeStatus":2}
Answer s.gw.upgrade.updatestatus
reply {"t":1576874811,"e":false,"success":true}
[I 191220 20:46:51 web:2246] 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=35808047b4e62d1d56f8&t=1576874811&sign=a19bbca26951454bed59f388a9354e8b (10.42.42.22) 5.40ms
[I 191220 20:46:55 web:2246] 206 GET /files/upgrade.bin (10.42.42.22) 46.96ms
[I 191220 20:46:55 web:2246] 206 GET /files/upgrade.bin (10.42.42.22) 702.72ms
encoding {"data":{"gwId":"35808047b4e62d1d56f8"},"protocol":15,"s":1523715,"t":1576874819} using protocol 2.1
b'2.1ea816da8c883335bt5inCMDHpcJKQpxfOe2RbXrJJwHixICx7wGFAYqe+3EqfPP+VAvS0l6F221IjwSNJsm5uMafEvnzLLz/1MQ4v/NF37LbHM/2JpeWXHkTZ6jdg4SXZL9kLEC9aqKXAOwI'
```

I made sure to copy of the repo so the logs are separated from the originals i posted.

TaakoMagnusen, 2019-12-20:
> I would try restarting `start_flash` and then plugging the device back in. Open a new terminal and run the `firmware_picker` directly.

Should i still connect to the vtrust-flash with my phone even though it's in the intermediate firmware stage?

TaakoMagnusen, 2019-12-20:
Well that worked! I was able to flash ESPURNA on one device. Although I meant to hit flash tasmota....

Now i need to find out how to get it onto tamsota

kueblc, 2019-12-20:
Awesome! Glad it worked. You can flash Tasmota from Espurna.

> Should i still connect to the vtrust-flash with my phone even though it's in the intermediate firmware stage?

For future readers, no need to reconnect phone if in intermediate.

closed 2019-12-20 by kueblc

TaakoMagnusen, 2019-12-20:
So, maybe this was misleading. I actually haven't flashed the original device i posted about (the first one to get stuck). I only flashed the second device (the one i orignally cracked open). I am still attempting to get it to flash but it looks like it can't find the device.

TaakoMagnusen, 2019-12-20:
False Alarm, after banging it around and plugging and unplugging a ton it i got the device to be discovered!!!

Now i have two perfectly usable, secure, private devices! Thank you Tuya-Convert maintainers!