ct-Open-Source/tuya-convert #460: Can't flash BSD29 smart socket

chriswales1024, 2019-12-21:
I am having no success flashing unbranded UK smart socket with model number BSD29. I see that the device is in the list of compatible devices (HTTP firmware). I have tried Tuya-Convert v2.2.0 and v2.3.0. The device is not identified. I can see using Fing that the devise has an IP of 10.42.42.38 and has port 6668 open.

I have attached the logs. Thanks in advance (-:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3991525/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3991526/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/3991527/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3991530/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3991531/smarthack-wifi.log)

kueblc, 2019-12-21:
Looks like the device connected but did not succeed in making HTTP(S) requests. It may be a DNS issue. Check that `dnsmasq` is running.

If it is, try `dig a.tuyaus.com` and `dig a.tuyaus.com @10.42.42.1`

Also please let us know what system you are using and the `dnsmasq --version`.

chriswales1024, 2019-12-22:
Hi Colin, thanks for such a speedy response (-:

Running on Raspbian Buster Lite kernel version 4.19 (26-Sep-2019). Fresh install of raspbian and tuya-convert so no other detritus in the way. You are correct that this is DNS related as on running start_flash.sh it said
`./start_flash.sh`
`Checking for network interface wlan0... Found.`
`Checking UDP port 53... Occupied by dnsmasq with PID 503.`
`Port 53 is needed to resolve DNS queries`
`Do you wish to terminate dnsmasq? [y/N]`
...and blindly I agreed.
On testing again, if I respond with _No_ then start_flash aborts.
Feels like I am snookered, or missing something?

`dig a.tuyaus.com` returns
;; QUESTION SECTION:
;a.tuyaus.com.                  IN      A
;; ANSWER SECTION:
a.tuyaus.com.           60      IN      A       52.24.126.12
a.tuyaus.com.           60      IN      A       52.24.198.180
a.tuyaus.com.           60      IN      A       34.214.213.182
a.tuyaus.com.           60      IN      A       34.215.43.212
a.tuyaus.com.           60      IN      A       35.162.71.88
a.tuyaus.com.           60      IN      A       35.162.121.9
a.tuyaus.com.           60      IN      A       52.10.172.122
a.tuyaus.com.           60      IN      A       52.24.43.219

`dig a.tuyaus.com @10.42.42.1` returns
;; QUESTION SECTION:
;a.tuyaus.com.                  IN      A
;; ANSWER SECTION:
a.tuyaus.com.           0       IN      A       10.42.42.1

`dnsmasq --version` returns
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley

kueblc, 2019-12-22:
> `./start_flash.sh`
> `Checking for network interface wlan0... Found.`
> `Checking UDP port 53... Occupied by dnsmasq with PID 503.`
> `Port 53 is needed to resolve DNS queries`
> `Do you wish to terminate [...]

We check the ports we need to use and attempt to clear them before starting our own services. It looks like you already have another instance of `dnsmasq` running which we need to close in order to start our own. It might be that this process failed to stop this existing `dnsmasq` instance.

Could you check that there is no `dnsmasq` instance running after running `setup_checks.sh`? You can use `pgrep -a dnsmasq`.

chriswales1024, 2019-12-22:
If I stop dnsmasq and mosquitto first then I get no complaints when running start_flash.sh but it does not succeed. After running setup_checks, pgrep -a dnsmasq returns nothing.

If I put a plug in to pairing mode it will sit there flashing for some time but a second after I run start_flash the relay fires and the LED stops flashing, so I surmise that there is contact but not on to the next step.

On my Pi, I first flashed a fresh Rasbian and then cloned tuya-convert and ran install_prereq.sh, and nothing else so only that which is stock will be running. I did not configure any networking on the Pi as I set my IPs in DHCP. I do concede that something must be different though as I have successfully configured other plugs with tuya-convert 2.3.0

countcobolt, 2020-01-03:
Similar behaviour on the  BSD34. It connects, the relay clicks and the led stops flashing. Same as https://github.com/ct-Open-Source/tuya-convert/issues/483 for behaviour

kueblc, 2020-01-03:
I did not notice this previously, but the PSK identity contains an unexpected prefix. This is likely the cause of your issues. Thanks @countcobolt for pointing out the similarity with your issue.

If you are flashing by wire, please make a backup of the stock firmware first. We can use this to debug and potentially develop a new workaround.

kueblc, 2020-01-24:
Let's merge any future discussion in with #483 to keep things tidy

closed 2020-01-24 by kueblc

iGodPhil, 2020-07-26:
Hey! Are there any news about flashing the BSD.. modules. Today i wanted to flash the bsd34 but it doesn't work. But according to the wiki this model is supported?

Maybe it will help the output and log files:

> tuya-convert v2.4.4
> Checking for network interface wlp3s0... Found.
> Checking UDP port 53... Occupied by dnsmasq with PID 1101.
> Port 53 is needed to resolve DNS queries
> Do you wish to terminate dnsmasq? [y/N] y
> Attempting to stop NetworkManager.service
> Checking UDP port 67... Occupied by dnsmasq with PID 1101.
> Port 67 is needed to offer DHCP leases
> Do you wish to terminate dnsmasq? [y/N] y
> Attempting to stop NetworkManager.service
> Checking TCP port 80... Available.
> Checking TCP port 443... Available.
> Checking UDP port 6666... Available.
> Checking UDP port 6667... Available.
> Checking TCP port 1883... Occupied by mosquitto with PID 1436.
> Port 1883 is needed to run MQTT
> Do you wish to terminate mosquitto? [y/N] y
> Attempting to stop mosquitto.service
> Checking TCP port 8886... Available.
> ======================================================
>   Starting AP in a screen..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
>   Starting web server in a screen
>   Starting Mosquitto in a screen
>   Starting PSK frontend in a screen
>   Starting Tuya Discovery in a screen
> 
> ======================================================
> 
> IMPORTANT
> 1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
>    This step is IMPORTANT otherwise the smartconfig may not work!
> 2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
>    Make sure nothing else is plugged into your IoT device while attempting to flash.
> 3. Press ENTER to continue
> 
> 
> ======================================================
> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> Sending wifiPassword          
> Sending token                 00000000
> Sending secret                0101
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ...........
> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Do you want to try flashing another device? [y/N] n
> ======================================================
> Cleaning up...
> Closing AP
> Exiting...

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4978682/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4978684/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4978685/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4978686/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4978687/smarthack-wifi.log)