ct-Open-Source/tuya-convert #461: Device take IP 10.42.42.35

RoSulek1, 2019-12-22:
Bseed (tuya) double wall switch

I try to flash TC, but my device cannot obtain 10.42.42.42 IP adress.
(log scripts_original.zip).

I try to change ip in scripts to10.42.42.35. After that i receve that connection refused.
(log scripts_35.zip)
[scripts_oryginal.zip](https://github.com/ct-Open-Source/tuya-convert/files/3992036/scripts_oryginal.zip)
[scripts_35.zip](https://github.com/ct-Open-Source/tuya-convert/files/3992037/scripts_35.zip)

kueblc, 2019-12-22:
Do not modify the IP address in the script, this is the mechanism by which the intermediate firmware notifies the script that the installation was successful. It is hard coded to be 10.42.42.42.

It looks like this is unfortunately not a supported device, indicated by the unexpected psk prefix. It may be that this device is not using an ESP, or it may be a patched firmware.

If you end up disassembling one of these, post pictures and we can try to identify the module. If it is an ESP82xx and you get a backup of the firmware we can debug the issue further.

RoSulek1, 2019-12-22:
I can set dhcp range beetwen ~42-42 and set static ip for vtrust-flash
(example 10.42.42.20) and try flash. May it sense?

RoSulek1, 2019-12-22:
This is wr4 module based on rtl chip

https://m.alibaba.com/product/62164279383/WR4-TUYA-Wifi-Module-RTL8710BN-Chip.html

Unsupported module

>

kueblc, 2019-12-22:
> I can set dhcp range beetwen ~42-42 and set static ip for vtrust-flash (example 10.42.42.20) and try flash. May it sense?

No, again, do not modify the IP address. If you do, you are only fooling yourself into thinking you have flashed the device when you have not.

> This is wr4 module based on rtl chip https://m.alibaba.com/product/62164279383/WR4-TUYA-Wifi-Module-RTL8710BN-Chip.html Unsupported module

Too bad, at least we have an answer. Better luck next time!

closed 2019-12-22 by kueblc