ct-Open-Source/tuya-convert #462: Flashing custom binary results in reboot loop

ccrisan, 2019-12-23:
I'm trying to flash Blitzwolf SHP6 devices with a custom firmware based on non-OS SDK 2.x with a standard 512k+512k flash layout recommended by Espressif themselves. The procedure goes well up to the point when device reboots with the new firmware: I get the following serial log:

     ets Jan  8 2013,rst cause:1, boot mode:(3,2)

    load 0x40100000, len 2592, room 16 
    tail 0
    chksum 0xf3
    load 0x3ffe8000, len 764, room 8 
    tail 4
    chksum 0x92
    load 0x3ffe82fc, len 676, room 4 
    tail 0
    chksum 0x22
    csum 0x22

    2nd boot version : 1.7(5d6f877)
    SPI Speed : 40MHz
    SPI Mode : DOUT
    SPI Flash Size & Map: 8Mbit(512KB+512KB)
    jump to run user1 @ 1000

    V2
    Mo
    rf_cal[0] !=0x05,is 0xFF

Given that custom binaries are limited to 512k (where does this limitation come from, by the way?), my actual binary is just the first 512k of a full 1MB image.

Now, since `esp_init_data_default.bin` is AFAIK supposed to contain that `rf_cal` value and it goes at `0xFC000` on an `1MB` device such as mine, could this be the cause? What am I missing? What should I do to get my custom binary onto a Tuya-converted device?

kueblc, 2019-12-23:
> Given that custom binaries are limited to 512k (where does this limitation come from, by the way?), my actual binary is just the first 512k of a full 1MB image.

The limit is imposed by how Espressif OTA updates are performed. In OTA configuration, there are two partitions. At any given time, one is running the currently active firmware and downloads the firmware update to the other.

Since we cannot overwrite running code, we're limited to one partition for the first flash. After that, you can use a bootstrap to install larger firmware binaries, provided the binary and bootstrap together are less than the capacity of the flash. Note that the last 12kb are reserved for system params.

> Now, since `esp_init_data_default.bin` is AFAIK supposed to contain that `rf_cal` value and it goes at `0xFC000` on an `1MB` device such as mine, could this be the cause?

Yes this is absolutely the cause. The flash is wiped and it is up to the user application to set any defaults, and these change with SDK version so it is not safe to assume. See [this thread](https://www.esp8266.com/viewtopic.php?f=6&t=17754) for an easy way to add this to your custom program.

ccrisan, 2019-12-23:
@kueblc that did the trick, many thanks! Besides that, it's a good idea to have such a verification and memory initialization in the firmware, in general.

One last (off-topic) question: why is a 3rd device required to be connected? Do we need to generate some traffic or so? If yes, can it be generated using AP-originated (ar)pings or similar?

kueblc, 2019-12-23:
Great, happy to hear your firmware is working well now!

> why is a 3rd device required to be connected?

We need to transmit UDP broadcast packets over the air for the smart config process, the ESP enters a sniffer mode to pick up the wifi credentials. It's been found that, at least in some environments, at least one client must be connected in order for the packet to be broadcasted.

ccrisan, 2019-12-24:
@kueblc thanks for your explanation. All is clear now.

closed 2019-12-24 by ccrisan

ccrisan, 2020-02-23:
@kueblc apologies for reviving an old issue.

I'm trying to automatize the entire process with a Raspberry Pi and I'm hoping to use a second WiFi adapter to connect to "vtrust-flash". Do I need to generate some kind of traffic or anything happening via this secondary wlan? Or is it enough just that another client is connected and has an IP?

kueblc, 2020-02-23:
It has been enough for me to have a client connect, no additional traffic necessary. YMMV. Some have reported that a third device is not necessary, while others have reported better success with some traffic.

I am honestly not too sure what determines whether a broadcast packet will make it to link layer or not, whether it is decided by `hostapd`, your hardware, or otherwise. If you do discover what the minimum conditions are, I would love to hear about it.

ccrisan, 2020-04-23:
@kueblc I have put together a ready-made [Tuya Convert OS](https://github.com/qtoggle/tuya-convert-os) that should make the conversion slightly easier, providing a friendly UI.

There is no secondary Wi-Fi client connected and the conversion seems to succeed every time, although I must admit I only tested it with just a couple of device models.