ct-Open-Source/tuya-convert #464: Can't get an IP address from vtrust-flash

RedGlow, 2019-12-26:
Hi,

I'm trying to use a raspberry pi 4 with latest raspbian (debian buster) in order to flash a device, but I'm stuck before flashing. My phone can see the vtrust-flash SSID, but when it tries to connect it gets no IP address.

Clone and install prereq are successful. When I start the script I get this:

```
pi@raspberrypi:~/tuya-convert/tuya-convert $ ./start_flash.sh
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 1726.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to terminate mosquitto
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
```

So everything seems ok too.

In attachment the logs produced:
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4003453/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4003454/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4003455/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4003456/smarthack-mqtt.log)

dnsmasq seems to be working too regarding DNS resolution:

```
pi@raspberrypi:~/tuya-convert/tuya-convert/scripts $ dig a.tuyaus.com

; <<>> DiG 9.11.5-P4-5.1-Raspbian <<>> a.tuyaus.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 22873
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;a.tuyaus.com.                  IN      A

;; ANSWER SECTION:
a.tuyaus.com.           12      IN      A       52.27.158.217
a.tuyaus.com.           12      IN      A       52.26.84.114
a.tuyaus.com.           12      IN      A       54.148.206.113

;; Query time: 13 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Thu Dec 26 21:01:19 GMT 2019
;; MSG SIZE  rcvd: 89

pi@raspberrypi:~/tuya-convert/tuya-convert/scripts $ dig a.tuyaus.com @10.42.42.1

; <<>> DiG 9.11.5-P4-5.1-Raspbian <<>> a.tuyaus.com @10.42.42.1
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 43892
;; flags: qr aa rd ra ad; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;a.tuyaus.com.                  IN      A

;; ANSWER SECTION:
a.tuyaus.com.           0       IN      A       10.42.42.1

;; Query time: 1 msec
;; SERVER: 10.42.42.1#53(10.42.42.1)
;; WHEN: Thu Dec 26 21:01:26 GMT 2019
;; MSG SIZE  rcvd: 57
```

Here are the relevant lines from the processes started from the script:

```
pi        1974  0.0  0.3   7808  2980 pts/0    S+   20:55   0:00 /bin/bash ./start_flash.sh
root      2169  0.0  0.2   8132  2180 ?        Ss   20:55   0:00 SCREEN -L -Logfile smarthack-wifi.log -S smarthack-wifi -m -d ./setup_ap.sh
root      2171  0.0  0.3   7676  2844 pts/2    Ss+  20:55   0:00 /bin/bash ./setup_ap.sh
root      2201  0.0  0.0      0     0 ?        I    20:55   0:00 [kworker/3:2-ipv6_addrconf]
nobody    2230  0.0  0.2  11076  1912 ?        S    20:55   0:00 dnsmasq --no-resolv --interface=wlan0 --bind-interfaces --listen-address=10.42.42.1 --dhcp-range=10.42.42.10,10.42.42.40,12h --address=/#/10.42.42
root      2235  0.0  0.3   9948  3172 pts/2    S+   20:55   0:00 sudo hostapd /dev/stdin
root      2241  0.0  0.1   7440  1456 pts/2    S+   20:55   0:00 hostapd /dev/stdin
root      2249  0.0  0.2   8132  2076 ?        Ss   20:55   0:00 SCREEN -L -Logfile smarthack-web.log -S smarthack-web -m -d ./fake-registration-server.py
root      2251  1.3  1.7  24764 16232 pts/3    Ss+  20:55   0:07 python3 ./fake-registration-server.py
root      2260  0.0  0.2   8132  2180 ?        Ss   20:55   0:00 SCREEN -L -Logfile smarthack-mqtt.log -S smarthack-mqtt -m -d mosquitto -v
mosquit+  2262  0.0  0.5   8776  4920 pts/4    Ss+  20:55   0:00 mosquitto -v
root      2271  0.0  0.2   8132  2208 ?        Ss   20:55   0:00 SCREEN -L -Logfile smarthack-psk.log -S smarthack-psk -m -d ./psk-frontend.py -v
root      2273  0.0  0.7  15580  7416 pts/5    Ss+  20:55   0:00 python2 ./psk-frontend.py -v
root      2282  0.0  0.2   8132  2084 ?        Ss   20:55   0:00 SCREEN -L -Logfile smarthack-udp.log -S smarthack-udp -m -d ./tuya-discovery.py
root      2284  0.0  1.1  19944 11092 pts/6    Ss+  20:55   0:00 python3 ./tuya-discovery.py
```

I don't really know what else to try right now, since I can't get any clear error from anywhere.

RedGlow, 2019-12-27:
(I just realized it may be a firewall problem, as soon as I can I'll test the hypothesis and update the bug report)

RedGlow, 2019-12-27:
Ok, can confirm it was a firewall problem, closing the bug report.

closed 2019-12-27 by RedGlow

kueblc, 2019-12-27:
Glad to hear you found your issue. Do you mind sharing the details of your resolution? It may concern #424 #450

RedGlow, 2019-12-27:
In my case the problem was caused by firewalld, which by default stopped everything but HTTP/S and SSH.

The solution was just a matter of:
```bash
# /etc/init.d/firewalld stop
```

And everything started working again.