ct-Open-Source/tuya-convert #469: AP not visible

criskolkman, 2019-12-28:
Hello,

Really pulling my hear atm...
I'm trying to start **tuya-convert** and I'm using a Sitecom N300 X2 USB adapter.
The adapter seems to support AP mode:

```
root@kali:~/tuya-convert# iwconfig
lo        no wireless extensions.

wlan35    unassociated  Nickname:"rtl_wifi"
          Mode:Master  Access Point: Not-Associated   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"f2\x0D\xB71X\xA3Z%]\x05\x17X\xE9^\xD4\xAB\xB2\xCD\xC6\x9B\xB4T\x11\x0E\x82tA!=\xDC\x87"
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
```

In the **config.txt** I assigned the following: **WLAN=wlan35**

When starting **tuya-convert** the following entries appear in the logs and the AP it should create will not become visible.

```
==> scripts/smarthack-wifi.log <==
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
Config dir: /tmp/create_ap.wlan35.conf.8Iw8rvFe
PID: 8799
Network Manager found, set wlan35 as unmanaged device... DONE
No Internet sharing
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlan35.conf.8Iw8rvFe/hostapd_ctrl
Configuration file: /tmp/create_ap.wlan35.conf.8Iw8rvFe/hostapd.conf
Line 4: invalid/unknown driver 'nl80211'
1 errors found in configuration file '/tmp/create_ap.wlan35.conf.8Iw8rvFe/hostapd.conf'

Error: Failed to run hostapd, maybe a program is interfering.
If an error like 'n80211: Could not configure driver mode' was thrown
try running the following before starting create_ap:
    nmcli r wifi off
    rfkill unblock wlan

Doing cleanup.. done

==> scripts/smarthack-psk.log <==
Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 97, in main
    proxies = [PskFrontend(gateway, 443, gateway, 80), PskFrontend(gateway, 8886, gateway, 1883)]
  File "./psk-frontend.py", line 48, in __init__
    self.server_sock = listener(listening_host, listening_port)
  File "./psk-frontend.py", line 17, in listener
    sock.bind((host, port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address

==> scripts/smarthack-web.log <==
Could not start server on port 80
[Errno 99] Cannot assign requested address

[... 33 lines cut ...]
192.168.60.100 {'ip': '192.168.60.100', 'gwId': '70770354c44f3393bc9a', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.101 {'ip': '192.168.60.101', 'gwId': '707703542cf432a47fa5', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.113 {'ip': '192.168.60.113', 'gwId': 'bf018809fde6cae3b2o7sg', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
192.168.60.103 {'ip': '192.168.60.103', 'gwId': '33360064600194d6b63d', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.108 {'ip': '192.168.60.108', 'gwId': '33360064840d8e6d90b4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.107 {'ip': '192.168.60.107', 'gwId': '33360064840d8e6dee38', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.106 {'ip': '192.168.60.106', 'gwId': '33360064840d8e6d77f4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.117 {'ip': '192.168.60.117', 'gwId': 'bf3e0f3fcc64691b71xbnq', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
192.168.60.100 {'ip': '192.168.60.100', 'gwId': '70770354c44f3393bc9a', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.101 {'ip': '192.168.60.101', 'gwId': '707703542cf432a47fa5', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.113 {'ip': '192.168.60.113', 'gwId': 'bf018809fde6cae3b2o7sg', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
192.168.60.103 {'ip': '192.168.60.103', 'gwId': '33360064600194d6b63d', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.108 {'ip': '192.168.60.108', 'gwId': '33360064840d8e6d90b4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.107 {'ip': '192.168.60.107', 'gwId': '33360064840d8e6dee38', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.106 {'ip': '192.168.60.106', 'gwId': '33360064840d8e6d77f4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.117 {'ip': '192.168.60.117', 'gwId': 'bf3e0f3fcc64691b71xbnq', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
192.168.60.100 {'ip': '192.168.60.100', 'gwId': '70770354c44f3393bc9a', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.101 {'ip': '192.168.60.101', 'gwId': '707703542cf432a47fa5', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.113 {'ip': '192.168.60.113', 'gwId': 'bf018809fde6cae3b2o7sg', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
192.168.60.103 {'ip': '192.168.60.103', 'gwId': '33360064600194d6b63d', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.108 {'ip': '192.168.60.108', 'gwId': '33360064840d8e6d90b4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.107 {'ip': '192.168.60.107', 'gwId': '33360064840d8e6dee38', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.106 {'ip': '192.168.60.106', 'gwId': '33360064840d8e6d77f4', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keymhedstsqgpyrq', 'version': '3.3'}
192.168.60.117 {'ip': '192.168.60.117', 'gwId': 'bf3e0f3fcc64691b71xbnq', 'active': 2, 'ablilty': 0, 'encrypt': True, 'productKey': 'keyuh8u8nhvwfs9t', 'version': '3.3'}
```

criskolkman, 2019-12-28:
Every time I run **./start_flash**, the wifi adapter gets changed back to **Mode:Managed**.
Don't know if that's normal behavior?

riker65, 2020-01-12:
same for me on ubuntu
AP not starting
trying it with internal wlan adapter and edimax wlan usb stick

criskolkman, 2020-01-12:
> same for me on ubuntu
> AP not starting
> trying it with internal wlan adapter and edimax wlan usb stick

Hello,

I did this on my Raspberry Pi 3 and it works like a charm. 
That the AP doesn't show is probably because of your wlan adapters (I had the same trying with the build in adapter and an external adapter).

kueblc, 2020-01-22:
Looks like you are using the `create-ap` fork which is not up-to-date. Have you tried with the `master` release?

Make sure you are not confusing AP mode with master. You can use `iw list` to find supported modes.

A quick search for your error
```
invalid/unknown driver 'nl80211'
```
suggests that perhaps you might try removing the `driver` line from the config entirely.

As far as `tuya-discovery` revealing the Tuya devices on your network, this is expected and can be useful for debugging. You can safely ignore it until we figure out the `hostapd` issue.

kueblc, 2020-01-25:
Let me know if you need any more assistance. Closing until I hear back from you.

closed 2020-01-25 by kueblc