ct-Open-Source/tuya-convert #470: Smart Plug won't connect to vtrust-flash

jolson316, 2019-12-29:
Device: Gosund smart plug wp3.
RPi 1B
Wifi adpater: TL-WN725N
Firmware: ?

When I run start_flash.sh, vtrush-flash starts and I can connect to it with my phone.  I connect in the normal IP range 10.42.42.13.  When I put the plug in programming mode, it blinks quickly for a few seconds and then stops, so something it happening.  I can't see any evidence of this in the log files.  

The install_prereq script did not have any errors.

How do I diagnose what is happening?

Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 370.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 331.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.

  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
...........Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
Exiting...

mqtt log
1577577513: mosquitto version 1.5.7 starting
1577577513: Using default config.
1577577513: Opening ipv4 listen socket on port 1883.
1577577513: Opening ipv6 listen socket on port 1883.
^C1577577660: mosquitto version 1.5.7 terminating

psk log
new client on port 443 from 10.42.42.13:48060
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no
shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.13:52630
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no
shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.13:46375
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no
shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.13:40617
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no
shared cipher (_ssl.c:727)'))
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 104, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

udp log
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C

web log
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...

wifi log
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 34:e8:94:df:23:27 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED a0:cb:fd:27:bd:90
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED a0:cb:fd:27:bd:90
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

jolson316, 2020-01-02:
So I did a pcap recording and looked through it  I could not find any record of the smart plug responding.  I see the access point sending out UDP messages to a lot of different IP addresses but nothing responds.

The smart plug has been connected to the online server so it likely has new the new firmware.  Is there a way to use the app to get the device to connect to the vtrust-flash network?

jcageman, 2020-01-02:
#472 is my open issue which sounds the same. There i got the suggestion to use the tuyapi project to connect it to v-trust. You could give it a shot, although it didn't work for me (check the linked issue for instructions)

jolson316, 2020-01-12:
So, I got some time to get the tuyapi project installed.  I ran the script as suggested and did not get any errors, but it did not remove the device from my network.  I'm not sure how to debug the script to see what is going on.

kueblc, 2020-01-25:
Are you able to connect the device to your home network? If so, you can use the `tuya-discovery.py` service to identify the device and determine if it is supported.

bartowl, 2020-01-27:
If this is just the smartconfig part, you may want to switch it to slow-blink-mode (usually by another 5-second-long press of the button, or keep it for 10 seconds instead of 5, depending on the device). It should then present its own AccessPoint named SmartLife-xxxx. Once you get there, you can start original TuyaSmart App on your phone, go to add Device, Choose device type, then on Top Right choose other modes and select the Access Point method (second option). You are then asked to confirm the device blinks slowly, just click to go on. Then you will be asked Wifi Credentials - enter here vtrist-flash and its password. Then, you will be asked to connect to the SmartLife-xxxx wifi with your phone. Actually this is the moment to start your device slow blink mode (by long pressing button). Once you connect your phone to the WiFi hosted by Tuya Device, click confirm. It will connect directly to your device (without contacting Tuya cloud) and pass vtrust-flash AP data to it. Having tuya-convert running in the background, your device **has to** connect to it.
This helped me to flash some sockets that did not respond to smartconfig packets being sent.
Hope this helps.

jolson316, 2020-01-29:
I tried using the SmartLife app to force the plug to connect to the vtrust-flash AP.  I got a network error 50500.  Someone else has this [problem](https://github.com/ct-Open-Source/tuya-convert/issues/427) and solved it by switching from Raspbian Buster to Raspbian Stretch.  I'll give that a try.

jolson316, 2020-11-12:
I eventually got the plugs to flash. The TP Link usb wifi adapter I was using has problems with AP mode. I could see the AP from my phone but it never really connected. I tried a different wifi adapter and it worked the first time.

closed 2020-11-12 by jolson316