ct-Open-Source/tuya-convert #471: will the swa11 be compatible?

baldarn, 2019-12-29:
Just to know... :) I bought 2 plugs and I tried this hack but without success.

the swa11 is not listed in the supported devices, so this is a feature request :)

kueblc, 2019-12-29:
There's no way to tell without trying, if you need help deciphering your logs please post them.

closed 2019-12-29 by kueblc

baldarn, 2019-12-30:
it worked!! yesterday I was dumber and I didn't see that it was ok! today I successfully flashed tosmota on it!

the swa11 is supported!! 

:)

kueblc, 2019-12-30:
Great, very happy to hear it worked out, and thank you very much for your support!

thenightfighter, 2020-01-08:
Won't work for me :(.

Any ideas what I could do?

marcob1281, 2020-12-03:
hi all, coming back to this old thread. 
I have the same smart plug, I could flash Tasmota FW, but Energy monitoring is not supported. any hint?