ct-Open-Source/tuya-convert #472: device did not appear with the intermediate firmware

jcageman, 2019-12-29:
I am trying to flash a Woox R5077 (tuya wifi bulb) wifi RGB lightbulb. Today i already did it succesfully with one, but i didn't use that bulb for some time with the tuya app (hence its firmware might have been updated?)

Any suggestions for trying?

I get the following output from running start_flash.sh:
```
./start_flash.sh 
Checking for network interface wlo1... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 2159.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen..
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n

```

Logs
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4008288/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4008289/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4008290/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4008291/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4008292/smarthack-wifi.log)

kueblc, 2019-12-29:
> could not establish sslpsk socket: <class 'ssl.SSLError'> returned a result with an error set

This line in your psk log suggests your OpenSSL is outdated.

jcageman, 2019-12-29:
how to upgrade?

jcageman, 2019-12-29:
> > could not establish sslpsk socket: <class 'ssl.SSLError'> returned a result with an error set
> 
> This line in your psk log suggests your OpenSSL is outdated.

i checked and this is due to my own attempts reading issues on github, one recommended changing psk-frontend.py shebang to python3, afterwards i reverted it back, which is also shown in the error logs. thanks for noticing though! Sorry for the mess in the logs, should have cleared them.

kueblc, 2019-12-29:
Could you please check your OpenSSL version?

Run `openssl version`. I believe it should be at least 1.1.1.

jcageman, 2019-12-29:
openssl version
OpenSSL 1.1.1  11 Sep 2018

kueblc, 2019-12-29:
OK great, we can rule that out. Can you repeat the process while recording the network traffic? Run `tcpdump -i wlo1 -w capture.pcap` and attempt `start_flash` again.

jcageman, 2019-12-29:
I cleaned all my log files and did one more try. Probably better to look at these:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4008448/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4008449/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4008450/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4008451/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4008452/smarthack-wifi.log)

jcageman, 2019-12-29:
had to rename extension, since github doesn't support upload, but its the pcap file as requested
[capture.txt](https://github.com/ct-Open-Source/tuya-convert/files/4008458/capture.txt)

kueblc, 2019-12-29:
This latest attempt, the device never connected. I can see the device MAC address in your earlier wifi log. See if you can get there again.

FYI once the device makes a connection (appears in wifi log) you do *not* need to repeat the pairing process with each attempt. It will automatically reconnect when you run `start_flash` again unless you put it back into pairing mode (resetting the wifi config).

jcageman, 2019-12-29:
It just seems i am not able to connect to these specific bulbs (the one in the first log i posted is probably the bulb that i did manage to convert). They are all exactly the same, except that i didn't use the first bulb with the Tuya app recently. I can still pair with the tuya app itself without a problem while pairing so its something else..

kueblc, 2019-12-29:
OK I see. It is added to the Tuya app now? Can you retrieve the device info?

meingraham, 2019-12-29:
I did see a user post about a week ago that they thought they'd encountered a device with an ESP82xx that would not take the TC flash... because it appeared the the Tuya firmware had been patched ☹️ I hope not...

jcageman, 2019-12-29:
> OK I see. It is added to the Tuya app now? Can you retrieve the device info?

What info do you need to know? i had tya-cli working at some point

jcageman, 2019-12-29:
[ { name: 'GU10 Smart Bulb R5077 5',
    id: '75014088b4e62d6bfd59',
    key: '1ee8539aef2271e4' },

For this specific light i get the above information from tuya-cli

kueblc, 2019-12-29:
@meingraham could you link to the discussion? A copy of the firmware in question would be really valuable.

@jcageman if you have the device added to your network, you can run `tuya-discovery.py` on the same network to pick up the relevant device information.

jcageman, 2019-12-29:
> tuya-discovery.py

Thanks, it says listening, but doesn't give any logs back and i have multiple tuya devices connected at the moment

ViXXoR, 2019-12-29:
I'm having the exact same issue with a new pair of AUKEY SH-PA1s bought off of Amazon.ca within the last couple if weeks.
After seeing this I'm really hoping the firmware isn't patched.

kueblc, 2019-12-29:
OK it looks like this device should be compatible with tuya-convert.

The last 12 characters of the gwId is the MAC address, which on looking up the OUI, is definitely an Espressif device. Version 3.3 is the TLS firmware so it should be making calls through the `psk-frontend` which we should see in the psk log.

It seems like the current roadblock is just getting the device added to the `vtrust-flash` network.

kueblc, 2019-12-29:
We don't have an official chatroom but I assure you I am keeping this thread open.

@ViXXoR can you provide more details as to how your issue is related? The issue we're seeing here is that the device is not pairing. If you have been able to flash other devices but have a device that will not appear in the wifi log, it might be the same issue.

jcageman, 2019-12-29:
just checking, but my` vtrust-flash` network has no password, while i read some guides which say `flashmeifyoucan` should be the  password

ViXXoR, 2019-12-29:
I believe it is related.

I just flashed a couple CE plugs a day or two ago with no issues. With the AUKEY, it flashes the intermediate firmware and never goes past that. I get the same "device did not appear" and I see the SSL cipher errors in smarthack-psk log.

kueblc, 2019-12-29:
In this issue, @jcageman was not able to install the intermediate firmware, so it is not likely related. The SSL cipher errors are irrelevant as the device never connected, so they are from another device (likely the mobile device).

kueblc, 2019-12-29:
@jcageman the password requirement was removed a few versions back to make the process easier.

kueblc, 2019-12-29:
@jcageman I have another idea, since you're able to get the device added to `tuya-cli`, we can issue a command to force the device to switch from your home network to `vtrust-flash`.

kueblc, 2019-12-29:
I'm assuming you already have `tuyapi`, if not do `npm i -g tuyapi`.

Then open a NodeJS repl, `node`.

Paste the following
```
const {MessageParser, CommandType} = require('tuyapi/lib/message-parser')

function switchAP( ip, key ){
	const payload = JSON.stringify({
		ssid: "vtrust-flash",
		passwd: "",
		token: ""
	})
	const parser = new MessageParser({
		key,
		version: 3.3
	})
	const buffer = parser.encode({
		data: payload,
		commandByte: CommandType.AP_CONFIG,
		sequenceN: 0
	})
	const client = new net.Socket
	client.connect( 6668, ip )
	client.on( 'connect', () => {
		client.write( buffer )
		client.destroy()
	})
	client.on( 'error', console.log )
}
```

Then type `switchAP( <device IP>, <device key> )`

This should ask the device to switch from your home network to the `vtrust-flash` network. I believe it is OK to do this even when `vtrust-flash` isn't running, it will connect when it becomes available and then you can skip the pairing process.

ViXXoR, 2019-12-29:
@kueblc I may have made an incorrect assumption in saying that the intermediate firmware was flashed.
My output matches what the OP was showing, so I may in fact be in the same boat.
I'll simply follow this thread and try any solutions that come up and see if they help.

kueblc, 2019-12-29:
@ViXXoR you can check if the intermediate flashed by looking in your web log. If it did, you should see two requests for `upgrade.bin`. The output for a device that fails to complete the process will look the same regardless of the underlying cause, but you can find out more by reading the logs.

jcageman, 2019-12-29:
> I'm assuming you already have `tuyapi`, if not do `npm i -g tuyapi`.
>
> Then open a NodeJS repl, `node`.
>
> Paste the following
>
> ```
> const {MessageParser, CommandType} = [...]

do i have to be in pairing mode for this or just execute after already being paired to tuya app? seems not to be doing anything. 
> switchAP(192.168.1.102, keywrqvhydt3ghp3)
...

kueblc, 2019-12-29:
You need to execute that on a network which the device is already connected to. If it is paired to your home network and set up with `tuya-cli`, use the key from that and IP address of the device on your network.

kueblc, 2019-12-29:
You may also try changing `CommandType.AP_CONFIG` to `CommandType.AP_CONFIG_NEW`, not sure if that also changed with version 3.3.

jcageman, 2019-12-29:
> You need to execute that on a network which the device is already connected to. If it is paired to your home network and set up with `tuya-cli`, use the key from that and IP address of the device on your network.

Did exactly that as far as i know and the script is just hanging without doing anything

I used the discovery script to make sure i used the right information:

192.168.1.102 {'ip': '192.168.1.102', 'gwId': '75014088b4e62d6bfd59', 'active': 2, 'ability': 0, 'mode': 0, 'encrypt': True, 'productKey': 'keywrqvhydt3ghp3', 'version': '3.3'}

kueblc, 2019-12-29:
Also make sure that the key is the one from `tuya-cli`, not the `productKey` from `tuya-discovery`

kueblc, 2019-12-29:
Once you run the `switchAP` function, the change should take effect immediately, but there won't be any output. You can verify that the device has disconnected from your home network though.

jcageman, 2019-12-29:
Oke, seems to be hanging still.

I found the key

[ { name: 'GU10 Smart Bulb R5077 5',
    id: '75014088b4e62d6bfd59',
    key: '3baea4fcf4fd3549' },

Still get the following and the device is still part of the network
```
switchAP(192.168.1.102,3baea4fcf4fd3549)
...
```
Tried commandByte: CommandType.AP_CONFIG_NEW as well

kueblc, 2019-12-29:
> switchAP(192.168.1.102,3baea4fcf4fd3549)

You must quote the arguments, sorry for not mentioning

kueblc, 2019-12-29:
switchAP("192.168.1.102", "3baea4fcf4fd3549")

jcageman, 2019-12-29:
> switchAP("192.168.1.102", "3baea4fcf4fd3549")

my javascript sucks, was already doubting. It executes right away, tried both options again, not errors, but device is still part of the network

kueblc, 2019-12-29:
Are you able to `ping 192.168.1.102`?

jcageman, 2019-12-29:
> Are you able to `ping 192.168.1.102`?

```
ping 192.168.1.102
PING 192.168.1.102 (192.168.1.102) 56(84) bytes of data.
64 bytes from 192.168.1.102: icmp_seq=1 ttl=255 time=11.7 ms
64 bytes from 192.168.1.102: icmp_seq=2 ttl=255 time=39.8 ms
64 bytes from 192.168.1.102: icmp_seq=3 ttl=255 time=6.24 ms
64 bytes from 192.168.1.102: icmp_seq=4 ttl=255 time=86.4 ms
```

kueblc, 2019-12-29:
Hmm I am a loss right now. If we can't establish a connection, we won't be able to tell if the firmware has changed. It looks like it should be able to flash if we can get past the pairing step.

jcageman, 2019-12-29:
ok, i managed to connect the device though! 

const TuyAPI = require('tuyapi');

const device = new TuyAPI({
  id: '75014088b4e62d6bfd59',
  key: '3baea4fcf4fd3549'});

(async () => {
  await device.find();

  await device.connect();

  let status = await device.get();

  console.log(`Current status: ${status}.`);

  await device.set({set: !status});

  status = await device.get();

  console.log(`New status: ${status}.`);

  device.disconnect();
})();

This toggles the light!

kueblc, 2019-12-29:
Excellent, so for sure we have the correct key.

Now I'm wondering if the `AP_CONFIG` command format has changed.

Are you comfortable and able to modify DNS requests on your home network, ie through `dnsmasq` on your router or on a home server? If you can capture `*.tuya(us|eu|cn).com` requests, we can redirect them to our `fake-registration-server`.

meingraham, 2019-12-29:
I can't find the chat. All I know was that when I saw it I ordered some of the GD-30W Diffuser so I'd get the version that I can easily Tuya-Convert... before the stock upgrades to a later Tuya firmware.

I'll keep my eyes & ears open for more buzz... should it come up again. And I will ask folks to download the firmware for analysis.

jcageman, 2019-12-29:
> `dnsmasq`

Never done anything like it, i would like to have a look at the tuyAPI first, maybe we are just slightly off (i have a feeling we are already close)

jcageman, 2019-12-29:
I tried modifying your script a bit and added logging, but its really connecting and writing the data to it. Might really indicate a change in how to connect to these lights

bryeartem, 2019-12-30:
Not to hijack the thread... but I think I was having a similar issue... and seeing this in the web log... any suggestions?
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/site-packages/tornado/web.py", line 1697, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "./fake-registration-server.py", line 224, in post
        self.reply(answer, encrypted)
      File "./fake-registration-server.py", line 85, in reply
        payload = b64encode(AES.new(options.secKey.encode(), AES.MODE_ECB).encrypt(pad(answer))).decode()
      File "/home/brian/.local/lib/python3.7/site-packages/Crypto/Cipher/_mode_ecb.py", line 135, in encrypt
        c_uint8_ptr(plaintext),
      File "/home/brian/.local/lib/python3.7/site-packages/Crypto/Util/_raw_api.py", line 235, in c_uint8_ptr
        raise TypeError("Object type %s cannot be passed to C code" % type(data))
    TypeError: Object type <class 'str'> cannot be passed to C code
[31m[E 191229 21:24:28 web:2246][m 500 POST /gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=03046203807d3a2d771e&t=87&v=1.0&sign=f515eaf936ba6a2a11dea08041d888de (10.42.42.1) 7.59ms

kueblc, 2019-12-30:
@bryeartem not related to this issue, feel free to open a new one with your full logs attached. For a quick fix you might try adding `.encode()` after `answer` on line 85 of `fake-registration-server`.

jcageman, 2019-12-30:
this morning i tried with a light bulb from tuya that showed version 3.2 and they flashed without any problem.  These are just normal light bulbs not spots like the GU10 one i couldn't flash.

ViXXoR, 2019-12-30:
> @kueblc I may have made an incorrect assumption in saying that the intermediate firmware was flashed.
> My output matches what the OP was showing, so I may in fact be in the same boat.
> I'll simply follow this thread and try any solutions that come up and see if they help.

FWIW, and I feel silly having not tried this before posting, after plenty of power-cycling and watching log files, I started thinking it may be a connectivity issue.
I moved some things around and got the plug just a few physical inches from the `vtrust-flash` source. After that, I was able to flash two plugs without issue. I guess the antennas in some devices are significantly weaker than others.

Sorry for adding fuel to the fire without troubleshooting more myself.

RedGlow, 2020-01-06:
I had a similar problem, and the solution was to turn off my firewall (it was stopping the communication between the raspberry with tuya convert and the lamp).

(for @kueblc : yes, I did the same error TWICE because apparently I have the memory of a goldfish xD)

deiger, 2020-01-07:
It seems I have the same issue, with a version 3.3 device:

smarthack-mqtt.log
```
1578428296: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1578428296: Using default config.
1578428296: Opening ipv4 listen socket on port 1883.
1578428296: Opening ipv6 listen socket on port 1883.
^C1578428442: mosquitto version 1.4.10 terminating
```

smarthack-psk.log
```
new client on port 443 from 10.42.42.31:39720
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:661)'))
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 104, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```

smarthack-udp.log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
10.42.42.28 b'{"ip":"10.42.42.28","gwId":"e62b573f2a5e6906","active":0,"ablilty":0,"encrypt":true,"productKey":"a3g9fxaujjgtei27","version":"3.3"}'
^C
```

smarthack-web.log
```
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
```

smarthack-wifi.log
```
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:00:00:00:00:00 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 28:6d:cd:16:48:91
wlan0: AP-STA-CONNECTED 60:00:00:00:00:00
wlan0: AP-STA-DISCONNECTED 28:6d:cd:16:48:91
wlan0: AP-STA-CONNECTED 28:6d:cd:16:48:91
wlan0: AP-STA-DISCONNECTED 28:6d:cd:16:48:91
wlan0: AP-STA-CONNECTED 28:6d:cd:16:48:91
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 28:6d:cd:16:48:91
wlan0: AP-STA-DISCONNECTED 60:00:00:00:00:00
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

kueblc, 2020-01-08:
@deiger, this is a different issue. `tuya-convert` only supports ESP82xx devices and yours is unfortunately a WinnerMicro based device.

deiger, 2020-01-08:
Thanks @kueblc. Could you please tell me where you saw that?
I had a look at the chip itself, and all it says is "TW-02 V2.0" and "RM-SMART M".

kueblc, 2020-01-09:
I noticed the typo `ablilty` in your discovery packet, which tells me its not an ESP. Then I checked the MAC address and saw the manufacturer is Winner.

jjtParadox, 2020-01-10:
I'm running into this same issue with two of the four devices I flashed in the last two days.

https://templates.blakadder.com/kmc-4.html - Successful
https://templates.blakadder.com/euri_lighting_LIS-A1001.html - Successful
https://templates.blakadder.com/lohas-ZN036.html - Both did not appear with intermediate firmware

None of these were connected to the tuya app before attempting flash.

I'm on a raspberry pi 3 B+ running buster (killing `wpa_supplicant` before flashing)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4044112/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4044113/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4044114/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4044115/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4044116/smarthack-wifi.log)

(the ea:fc:74... mac in the wifi log is the randomized mac of my phone)

`dig a3.tuyaus.com` points to 10.42.42.1

I also did a tcpdump of one attempt
[capture.pcap.txt](https://github.com/ct-Open-Source/tuya-convert/files/4044118/capture.pcap.txt)

I can get the LOHAS spotlights into a "flashing twice per second" mode and a "flashing once every 3 seconds" mode. The slow flash appears to be AP mode, which does nothing with tuya-convert.

The fast flash appears to work for a moment, as the device shows up in the wifi logs. However, the flashing stops when the smart config packets are sent, followed by tuya-convert retrying and eventually timing out.

jjtParadox, 2020-01-10:
I just got it working after checking out the development branch and fixing a bug!

Bug: https://github.com/ct-Open-Source/tuya-convert/blob/development/scripts/psk-frontend.py#L66 needs to be `hint=self.hint.decode()` instead of `hint=self.hint`

Both spotlights are now flashed!

blakadder, 2020-01-11:
Had the same issue. Device was connected to Tuya app first.

I've pulled a clean newest dev branch, installed all the prereqs and added the PR #496 fix for psk-frontend.py but still no success in flashing. It just trying to resend smartpackets

all the logs are quiet:
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4049277/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4049278/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4049279/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4049280/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4049281/smarthack-psk.log)

tuya-discovery:
[tuya.log](https://github.com/ct-Open-Source/tuya-convert/files/4049287/tuya.log)

tcpdump:
[capture.zip](https://github.com/ct-Open-Source/tuya-convert/files/4049293/capture.zip)

photonforge, 2020-01-12:
Im having the same problem with Wipro 16a smart plug [https://www.amazon.in/gp/product/B07XPQYVLG/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1](url)

I tried using raspberry pi zero USB serial gadget mode

these are my logs
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4050820/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4050821/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4050822/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4050823/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4050824/smarthack-wifi.log)

what am i doing wrong?

jcsogo, 2020-01-12:
I have same issue with some smart switch modules that before were ESP and now it seems that they have moved to a Beken Corp chip. IThe previous versions were completely white, and these have the connections in green, so I don't know if all the ones like this are not ESP, but you should be aware.

Mine were from [MOES Home](https://www.moeshouse.com/collections/diy-smart-switch/products/mini-wifi-smart-light-switch-diy-breaker-module-smart-life-tuya-app-remote-control-works-with-alexa-echo-google-home-1-2-way)

I got the issues listed in this [thread](https://community.home-assistant.io/t/unkown-tuya-chip/153591)

I am attaching the log files I got when tried to tuya-convert it.
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4051509/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4051510/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4051511/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4051512/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4051513/smarthack-mqtt.log)

kueblc, 2020-01-12:
@photonforge I did not find an ESP82xx based device connecting in your web logs unfortunately, suggesting it might not be possible to use ESP third party firmware on that device.

[28 more replies]