ct-Open-Source/tuya-convert #473: Device did not appear with the intermediate firmware

bryeartem, 2019-12-30:
Trying to flash eco4life single outlet smart plug... 

```
Traceback (most recent call last):
File "/usr/local/lib/python3.7/site-packages/tornado/web.py", line 1697, in _execute
result = method(*self.path_args, **self.path_kwargs)
File "./fake-registration-server.py", line 224, in post
self.reply(answer, encrypted)
File "./fake-registration-server.py", line 85, in reply
payload = b64encode(AES.new(options.secKey.encode(), AES.MODE_ECB).encrypt(pad(answer))).decode()
File "/home/brian/.local/lib/python3.7/site-packages/Crypto/Cipher/_mode_ecb.py", line 135, in encrypt
c_uint8_ptr(plaintext),
File "/home/brian/.local/lib/python3.7/site-packages/Crypto/Util/_raw_api.py", line 235, in c_uint8_ptr
raise TypeError("Object type %s cannot be passed to C code" % type(data))
TypeError: Object type <class 'str'> cannot be passed to C code
�[31m[E 191229 21:24:28 web:2246]�[m� 500 POST /gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=03046203807d3a2d771e&t=87&v=1.0&sign=f515eaf936ba6a2a11dea08041d888de (10.42.42.1) 7.59ms
```
You mentioned adding answer to line 85 of fake-registration-server.py can you show me a larger snippet so I know I have the placement of the () and stuff correct?

Attached are logs... some of which may contain an error from initial improper placement of said ".encode()"

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4008928/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4008929/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4008930/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4008931/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4008932/smarthack-wifi.log)

kueblc, 2019-12-30:
Hi @bryeartem 

Try replacing line 85 of `fake-registration-server.py` with
```
payload = b64encode(AES.new(secKey.encode(), AES.MODE_ECB).encrypt(pad(answer).encode())).decode()
```
Note the `.encode()` after `pad(answer)`

bryeartem, 2020-01-01:
so that took care of that error but still doesn't seem to be working.  Would the new logs help?
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4013725/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4013726/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4013727/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4013728/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4013729/smarthack-wifi.log)

bryeartem, 2020-01-01:
That took care of it!  Worked great, thank you for your time.

closed 2020-01-01 by bryeartem