ct-Open-Source/tuya-convert #475: Unable to flash Monster Illuminessence LED light strip

rpelletierrefocus, 2019-12-30:
I can successfully connect this device to tuya.

I have the most up to date version of tuya-convert and have flashed many other devices without issue. Would appreciate any help to troubleshoot.

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4011254/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4011255/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4011256/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4011257/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4011258/smarthack-udp.log)

I do have this showing up in my smarthack-psk log. 
new client on port 443 from 10.42.42.13:62861
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.13:62862
('could not establish sslpsk socket:', SSLError(1, u'[SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:727)'))
new client on port 443 from 10.42.42.13:62863
('could not establish sslpsk socket:', SSLError(1, u'[SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:727)'))

$ openssl version
OpenSSL 1.1.1d  10 Sep 2019

kueblc, 2019-12-30:
The SSL errors are from your mobile device, it looks like the device never connected to the `vtrust-flash` AP. Can you provide information about the device you are trying to flash?

rpelletierrefocus, 2019-12-31:
Here is the device information from tuya-cli

{
    name: 'RGBW LED Strip',
    id: 'eb178b415375cfb5fei7ur',
    key: '0bd9ab46dc17e9bc'
  },

rpelletierrefocus, 2019-12-31:
I wiped the logs and tried tuya-convert again. Here are the rsulting logs.
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4012603/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4012604/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4012605/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4012606/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4012607/smarthack-psk.log)

kueblc, 2019-12-31:
I can see the device has connected now, but unfortunately it is not a supported device. We can only install ESP82xx firmware and this device does not use an ESP82xx.

closed 2019-12-31 by kueblc

rpelletierrefocus, 2019-12-31:
Bummer, but good to know, thank you. May I ask how can you tell from the above information? Teach this guy to fish a little please.

kueblc, 2019-12-31:
Sure, there are three clues.

The first is the MAC address from your wifi log, [looking up the OUI](https://aruljohn.com/mac/68572D) you can see it is not from Espressif Inc.

The second clue is that the device ID does not end with the MAC address of the device, or a MAC address of any kind. Devices with the ESP SDK use an ID constructed from the product ID followed by the MAC address.

The third clue is a bit more obscure, in your udp log, the discovery packet for this device contains the misspelling "ablilty" which is not present in the Tuya ESP8266 SDK, only in their other SDKs.

rpelletierrefocus, 2020-01-01:
My apologies but one thing I don’t understand is how there is a tasmota template for this device if it is not supported. https://templates.blakadder.com/monster_smart_illuminessence.html

kueblc, 2020-01-01:
The manufacturers are in the process of switching over, you may be able to find models that still use the ESP82xx.