ct-Open-Source/tuya-convert #476: Ubuntu Bionic

tmuehlhoff, 2019-12-30:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2019-12-30:
Hi @tmuehlhoff, are you saying that tuya-convert is not working with a wifi adapter that does not support AP mode? This is expected as AP mode is required to, well, create an AP.

tmuehlhoff, 2019-12-31:
Yeah sure, but what I'm after is an according error output....

nl80211: Could not configure driver mode

Could cause to output one, or ?

Also it could issue an rfkill unblock....

kueblc, 2019-12-31:
Sorry I don't understand what you are asking. If your card does not support AP mode, you will get errors if you try anyway. None of the services can start if that first step fails.

tmuehlhoff, 2019-12-31:
What about an error message if the wifi chip isn't supported ?

kueblc, 2020-01-01:
[We do this already](https://github.com/ct-Open-Source/tuya-convert/blob/dbeab1ba9d8730b2fbf97b48b7115c2b418a592e/scripts/setup_checks.sh#L32). Can you specify what you think it should do differently?

tmuehlhoff, 2020-01-02:
well, as you can see from above, I didn't receive any, only took that info from the logs...

my shell output:
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID vtrust-flash
Sending wifiPassword
Sending token 00000000
Sending secret 0101

tmuehlhoff, 2020-01-02:
obviously my intel card doesn't produce that error message you're matching on... Let me try to reproduce it, I will come back with some more info on why this isn't catched.

tmuehlhoff, 2020-01-06:
Got it: if you just grep for "AP" in iw list it will find the string "AP" even though it's not supported by the adapter:

tomas@thinky:~$ iw list | grep AP
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Driver supports full state transitions for AP/GO clients
tomas@thinky:~$ 

but:

        Supported interface modes:
                 * IBSS
                 * managed
                 * monitor

does not list AP mode.
Sure you can now argue this is maybe an interface fault, but it's an Intel "Ultimate N Wifi Link 5300", not the most recent, but pretty common still ;)

On top of that, if I plug in the 2nd Wifi adapter, "iw list" will print all capabilities for BOTH adapters!

Attaching my iw list output for only the Intel adapter as reference.
BR/To

`Wiphy phy0
	max # scan SSIDs: 20
	max scan IEs length: 195 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * monitor
	Band 1:
		Capabilities: 0x107e
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 4 usec (0x05)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
[... 121 lines cut ...]
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Device supports TX status socket option.
	Device supports HT-IBSS.
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	Device supports per-vif TX power setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Device supports static SMPS
	Device supports dynamic SMPS
	Device supports configuring vdev MAC-addr on create.
`

kueblc, 2020-01-06:
> Got it: if you just grep for "AP" in iw list it will find the string "AP" even though it's not supported by the adapter

How about `iw list | grep -A 10 "Supported interface modes" | grep -e "\* AP$"`?

> On top of that, if I plug in the 2nd Wifi adapter, "iw list" will print all capabilities for BOTH adapters!

Yes, this is a known issue with how `iw list` works. I decided it was better to catch some bad configurations than none at all. If you know of a way to list the capabilities of only the specified interface, please let us know.

tmuehlhoff, 2020-01-06:
Yes, suggestion works well here, thanks for fixing....

The other thing: if the device is RFKILLED... you could run an rfkill unblock <dev> over it ?!

Or do you want me to file another ticket ?

kueblc, 2020-01-08:
You do not need to file another ticket. A wifi device wouldn't normally be `rfkill`ed, unless done explicitly by the user, in which case it would not be the responsibility of `tuya-convert` to change this. Can you explain why a device might be `rfkill`ed without the user knowing?

tmuehlhoff, 2020-01-08:
well, MY device WAS rfkilled, the only thing I did is enable Ethernet in NetworkManager (which implicitly disables Wifi, thus rfkills it).

commit 9fc1eae0 (2020-01-06): Check that AP is supported by grepping under "Supported interface modes" only, fixes #476
PR #490 (2020-01-12): Release v2.4.0
- Added docker support by @Cyber1000 
- Print a warning if running over SSH
- Update tasmota.bin to 8.1.0.2
- Revert identity prefix correction
- Add version information to wifi log for debugging
- Update `sslpsk` for python3
- Removed all python2 dependencies
- Replace unmaintained `pycrypto` with equivalent `pycryptodome` package
- Fix encoding issues with `pycryptodome`, solves #473 #332
- Incorporate `stop_flash` into `start_flash` and refactor
- Add hidden skip option when checking occupied ports
- Improve AP support check, fixes #476
closed 2020-01-12 by kueblc