ct-Open-Source/tuya-convert #478: Try to flash a door sensor TYWE3S but not succesfull help appreciated ;-)

andyprivate, 2019-12-31:
[image]
Hi all

i am a newbie here so sorry if i do something wrong i tried to flash a doorsensor image attached but cannot get it to work your help will be very usefull thanks in advance Andy
Logfiles are also attached
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4012485/smarthack-mqtt.log)

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4012448/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4012449/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4012450/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4012451/smarthack-wifi.log)

kueblc, 2019-12-31:
Door and motion sensors only stay online for a brief period. See #135. Closing as duplicate.

closed 2019-12-31 by kueblc

andyprivate, 2019-12-31:
Thanks to kueblc work now like a charm thanks and an nice new year ;-) thanks Andy

kueblc, 2019-12-31:
Glad to hear it! Happy New Year

kueblc, 2021-11-13:
Hi @muammersayar, I'm afraid I do not have the binaries you are looking for. Best of luck.