ct-Open-Source/tuya-convert #479: Bricked an SP112

pizzaice, 2019-12-31:
Hi and good afternoon,
I just tried flashing a Gosund SP112 and unfortunately seemed to have bricked the device.

I follow the usual steps but was told "Device did not appear with the intermediate firmware"
Subsequently I tried again to flash with no success :-(
Now the plug does not react at all - no LED and nothing on button presses. Can you help?
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4012815/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4012816/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4012817/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4012818/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4012819/smarthack-mqtt.log)

Thanks a lot!
BR Chris

kueblc, 2019-12-31:
It is not bricked, it is stuck on the intermediate firmware. See [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting) for device unresponsive and `vtrust-recovery` is visible.

pizzaice, 2020-01-01:
Thanks a lot for the hint @kueblc and happy new year!
Managed to fix it by first connecting to vtrust-recovery directly with my phone and browsed to http://10.42.42.42/undo 
This resulted in  "rebooting into userspace 1"
and made vtrust-recovery dissappear. I was then able to flash normally! 
Thanks again - just flashed 4 x SP112 and one SP1

closed 2020-01-01 by pizzaice