ct-Open-Source/tuya-convert #481: tasmota template diffuser problem tuya

roberdlv, 2020-01-01:
Hello, I have installed tasmota in my aromatherapy (https://es.aliexpress.com/item/32948538359.html?spm=a2g0s.9042311.0.0.274263c0oT8boY) all very well, but I can't find a correct template for my correct aromatherapy, tried to use this:
https://templates.blakadder.com/essential_oil_400ml.html

But the light switch on HA does not work and neither does the water sensor, when I press the light switch on HA it turns me off and turns on the LED of the aromatherapy state and not the RGB light, when the water sensor runs out of water Water keeps marking OK.

Could you help me solve and make everything work correctly for me? I'm going to put some pictures of the plate that my aromatherapy has.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thank you and sorry for my bad English.

kueblc, 2020-01-01:
Hi @roberdlv, glad tuya-convert was able to help you successfully install third party firmware. For support with your third party firmware, please see the respective project.

closed 2020-01-01 by kueblc

orlandogil2007, 2020-01-17:
Hola Roberdlv que idioma hablas, yo tengo un diifusor con la placa parecida como 95% igual lo diferente es el chip wifi, pero no es de smart life es de ewelink, me gustaría intercambiar ideas.

[image]