ct-Open-Source/tuya-convert #483: BSD34 flashing: sslpsk error

countcobolt, 2020-01-03:
Hi all

I am trying to flash a BSD34 smart socket. I have downloaded the latest tuya master built using git. Working on an RPI 3B. I can connect the socket using the normal smart life app. I did this after I absolutely could not manage to flash it at first (still cant). Decided to dive a bit deeper into it today and found this:

In smarthack-psk.log I have a ton of entries as following

```
new client on port 443 from 10.42.42.20:53000
ID: 0242416f68626d64366147393149465231509241f729c9f0af3aa41e355b7cbeb1ece63da6ff54b74f271af0ef044466e6
PSK: d9488a1b4524ae3e31acd0342e6d0b2eedbb5d55e957a9a51073b95e36ab1c5c
('could not establish sslpsk socket:', SSLError(1, u'[SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:727)'))
new client on port 443 from 10.42.42.20:5371
ID: 0242416f68626d64366147393149465231509241f729c9f0af3aa41e355b7cbeb1ece63da6ff54b74f271af0ef044466e6
PSK: d9488a1b4524ae3e31acd0342e6d0b2eedbb5d55e957a9a51073b95e36ab1c5c
('could not establish sslpsk socket:', SSLError(1, u'[SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:727)'))
```

My smartphone and device are actually connecting to the acces point.  In smarthack-wifi.log you see
```
wlan0: AP-STA-DISCONNECTED ac:57:75:99:f2:9f
wlan0: AP-STA-CONNECTED ac:57:75:99:f2:9f
wlan0: AP-STA-CONNECTED c4:4f:33:bc:10:c6
wlan0: AP-STA-DISCONNECTED ac:57:75:99:f2:9f
wlan0: AP-STA-CONNECTED ac:57:75:99:f2:9f
wlan0: AP-STA-DISCONNECTED ac:57:75:99:f2:9f
wlan0: AP-STA-CONNECTED ac:57:75:99:f2:9f
```
The AC mac is my smartphone, while the C4 is the actual device I am trying to flash.

The mqtt log remains quite virgin
```
1578041479: mosquitto version 1.5.7 starting
1578041479: Using default config.
1578041479: Opening ipv4 listen socket on port 1883.
1578041479: Opening ipv6 listen socket on port 1883.
```

as does the UDP log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```
Smarthack-web.log does not give me more either
`Listening on 10.42.42.1:80`

I have browsed and google for hours now and mostly I find that this might be due to not using the ESP8x chip. Yet funnily enough, when I simply configure them using the smart life app, I see them in my network as ESP_some_ref_number in it. And it does get an IP address. 

The socket looks like this

[image]

Any clues on why the SSL error shows up?
Using openssl 1.1.1d / Python 2.7.16 and 3.7.3 

Kind regards 
Steve

rsbob, 2020-01-03:
Runs on fw 1.1.4 in smarlife app
[image]
[image]
Have the same issue.ssl 1.1.1d and python 2 and 3 .have to flash with soldering. Esp chip is mounted.

jcageman, 2020-01-03:
Could you please check your OpenSSL version?

Run openssl version. It should be at least 1.1.1.

rsbob, 2020-01-03:
1.1.1c

countcobolt, 2020-01-03:
> Runs on fw 1.1.4 in smarlife app
> [image]
> [image]
> Have the same issue.ssl 1.1.1d and python 2 and 3 .have to flash with soldering. Esp chip is mounted.
> For to open the smart plug you have to eat first Southtirolean Speckknödl
> [image]

Just wondering did you just pull really hard on it or what is the trick to open it? Turn? Worried to break them

rsbob, 2020-01-03:
moved a screwdriver into one  hole and then leveraged.nothing breaks

countcobolt, 2020-01-03:
One more stupid question: I got it open but how did you get the PCB out? Stuck like hell in mine?

countcobolt, 2020-01-03:
A few more stupid questions : 
1: did you solder on the bottom or on the dots?
2: did you connect the gnd to reset?
3: which tasmota did you put on it?

Thanks a lot in advance :)

rsbob, 2020-01-03:
2 screws in total.hav no usb serial programmer here.its on the way ... only ioo to ground .the rest on grenn pcb .from right to left.....3.3 gnd......

kueblc, 2020-01-03:
Hi @countcobolt and @rsbob, it looks like you may have discovered a new version of the Tuya firmware. The PSK identity in your psk log does not have the expected prefix, which explains why the calculation is failing.

If you are flashing by wire, could you please make a backup of the stock firmware? We can use this to further debug and possibly develop a new workaround.

countcobolt, 2020-01-03:
@kueblc 
Ok I have it open and will try tomorrow morning. A few stupid questions (I am on Linux or Mac) do you have some tutorials on how to download the existing firmware?

Kind regards

rsbob, 2020-01-03:
https://github.com/espressif/esptool
https://github.com/espressif/esptool/blob/master/README.md

kueblc, 2020-01-03:
Once you have it installed you can use something like:
```
esptool.py read_flash 0 0x100000 firmware-backup.bin
```

rsbob, 2020-01-04:
flashed now ...
[image]
 #[image]
[image]
@kueblc

rsbob, 2020-01-04:
Here the orig.FW
[BSD34--image1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/4021684/BSD34--image1M.zip)

countcobolt, 2020-01-04:
@rsbob  which. pin did you connect the GPIO0 on? I have CW, R and then 00 in one line (or similar (can't really read it properly on them)) Thanks

Steve

rsbob, 2020-01-04:
gpi0 (esp pcb side soldered  to ground ,then give voltage and run commans )Green pcb .   3,3 volt Gnd   rx  tx     on the main platine side . yellow cable to the the blue ones.see foto

countcobolt, 2020-01-04:
@rsbob  hey, I know I need to connect GPIO0 to the ground, but that is the problem: which one is GPIO0 on the PCB? The printing on my PCB is really bad, so cannot read it (literally :) )

rsbob, 2020-01-04:
😂You se   IO on blue board near 3,3 volt and grount lines?but if you can wait wait a hour i add a foto.

countcobolt, 2020-01-04:
[image]
Wondering which of these is gpio0 :)

rsbob, 2020-01-04:
the left from Rx .the one without trace .

countcobolt, 2020-01-04:
@rsbob thanks;, got it :) now got it flashed and working :) @kueblc let me know if you need testing, I have 2 more here I can test with tuya-convert

countcobolt, 2020-01-04:
just one thing: using the template on https://github.com/blakadder/templates/blob/master/_templates/BSD34 the led is not working, toggle etc is working as expected

Farfar, 2020-01-10:
I get the same sslpsk errors in smarthack-psk.log though slightliy different ID:
ID: 0242416f68626d643661473931494652315ee70ad6ee67338a4f486a0750cfc89d52ee6a3be1727dcaa5eef3054e2165f5
PSK: 2b25c3013c2186c23697c1cc752161c039d1672c36b3d7e6df51c4f43b87586a
('could not establish sslpsk socket:', SSLError(1, u'[SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:727)'))

The device is a Deltaco SH-P01E socket with power monitoring.

smarthack-wifi.log with my Huawei phone and smart plug connected:
...
wlx90f652e46df6: AP-ENABLED 
wlx90f652e46df6: AP-STA-CONNECTED 30:45:96:1d:86:56
wlx90f652e46df6: AP-STA-CONNECTED d8:f1:5b:8b:34:59

Would be nice to be able to use tuya-convert since I don't know how to take the plug apart without damaging it.

rsbob, 2020-01-10:
any backup of old smartlife fw   for me😂bsd34

Farfar, 2020-01-13:
If anyone is interested, here's the firmware from my Deltaco SH-P01E plug.
[Deltaco_SH-P01E_20200110_image1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/4052386/Deltaco_SH-P01E_20200110_image1M.zip)
The firmware is the latest up until Jan 10th 2020.

kueblc, 2020-01-22:
Thank you @rsbob and @Farfar for collecting this data. Apologies for the late response as we have had a loss in the family. I will try to work on the issue this week, please feel free to ping me for a follow up.

kueblc, 2020-01-24:
Took some time on this today. Unfortunately these latest firmware builds seem to have changed a number of things making this a non-trivial fix.
- firmware OS SDK version did not change, but the hash and build time did
  - `OS SDK ver: 2.0.0(e8c5810) compiled @ Jan 25 2019 14:26:04`
  - `OS SDK ver: 2.0.0(29f7e05) compiled @ Sep 30 2019 11:19:12`
    - interesting that this build date corresponds with the release of tuya-convert 2.0, coincidence?
- psk begins with `02` rather than `01`
  - guessing this will tell us the psk algorithm version
- disabled most of the serial debugging output
  - unfortunate as this was a useful reverse engineering resource
- did not respond to our `smartconfig` procedure
  - it may use a new mechanism or additional restrictions have been imposed

Thus the cat and mouse game with Tuya continues.

kueblc, 2020-01-24:
If anyone is able to capture network traffic from one of these devices pairing with the cloud, along with the corresponding firmware backup, this would be instrumental in reverse engineering efforts

Tollbringer, 2020-01-29:
If anyone finds a US variant of this I would be happy to get a  couple sacrificial units and give it a go.

AIexBV, 2020-01-31:
@kueblc If you are interested to analyse the traffic, you could probably use the bulb sold in a big online store as "_bakibo smart wlan led_". I have the same issue with it. It shows `could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:720)`.

eku, 2020-01-31:
@AlexBV mine is a Bakibo TB95 with the same problem.

Tollbringer, 2020-01-31:
@AIexBV  It would be great for anyone wishing to contribute, who has the ability to disassemble/backup the original FW before doing anything and then another backup after trying TC.

Logs are good too!! We're going to need to sacrifice (anyone in the community who is willing/able) to sacrifice some modules to play this game of Cat & Mouse

Tollbringer

AIexBV, 2020-02-01:
@Tollbringer  I already tried to open the bulb. But I do not have access to good tools, so unfortunately I damaged the chips. But maybe the log entries will help. I got many lines like this:

`new client on port 443 from 10.42.42.11:63510
ID: 0242416f68626d643661473931494652318f1af4a8c59aa2df416796cf6c3509959021a9fc70c1265ab969a0c21765060a
PSK: cae5783da388990f138996dc22b9b50e3c155cc50fdd80401e8e27db0a2d58c5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:720)`

eku, 2020-02-01:
@kueblc  please find attached a tcpdump of the communication: [bakibo_bulp_pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/4142919/bakibo_bulp_pcap.zip)

aviogit, 2020-02-01:
Hey guys, I think I'm in an even worse situation, when trying `tuya-convert` from within docker I get a wall of these:

> new client on port 443 from 10.42.42.31:37666
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37670
> could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37674
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37676
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37680
> could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37686
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37694
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37730
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37738
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37748
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37756
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37760
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37772
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37784
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37796
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37806
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37810
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!
> new client on port 443 from 10.42.42.31:37820
> could not establish sslpsk socket: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:852)
> don't panic this is probably just your phone!

The device is a [Nooie Led Smart Bulb](https://www.amazon.it/Lampadina-2800k-6000k-Multicolor-intelligenti-Compatibile/dp/B07X3K64HL), the FW is at version 1.0.5, so it should be old enough, but it talks protocol version 3.3 (I think) and [I had some trouble fetching its key](https://github.com/AMoo-Miki/homebridge-tuya-lan/issues/147#issuecomment-580943280).

[image]

I say it talks a 3.3 version protocol because looking at the traffic with it's official app, I see a lot of `3.3` in each packet (see attach) and the already known `0000 0000 55aa` sequence (maybe I'm wrong, it's just a little bit more of 24h that I'm studying this topic :)

[tcpdump-host-and-port-X.txt](https://github.com/ct-Open-Source/tuya-convert/files/4143147/tcpdump-host-and-port-X.txt)

If you need some reverse engineering help, let me know, I'm ready to mostly everything except cracking open the device.

kueblc, 2020-02-01:
Thank you for the data @eku I'll take a look

@aviogit 
> don't panic this is probably just your phone!

aviogit, 2020-02-01:
I've tried both with my phone and my tablet, given the problems I had with the bulb I bet it's the bulb.

kueblc, 2020-02-01:
@aviogit I'm not saying your bulb does not have problems, but that specific error you are reading is unrelated. You can check which device is assigned that IP address.

aviogit, 2020-02-01:
Mmm, running everything outside of docker, I get this:

> new client on port 443 from 10.42.42.10:36974
> argument 2 must be str, not bytes
> new client on port 443 from 10.42.42.10:36975
> argument 2 must be str, not bytes

And you're right, 10.42.42.10 is my tablet:

> #> arp -n
> Address                  HWtype  HWaddress           Flags Mask            Iface
> 10.42.42.10              ether   10:7b:44:bd:f2:b7   C                     wlan1

I confess I'm not understanding the pairing procedure and why another device is needed, so if you have any advice on how to solve this, I'm all ears.

kueblc, 2020-02-01:
> argument 2 must be str, not bytes

You must run `install_prereq.sh` again after updating `tuya-convert`

aviogit, 2020-02-01:
I don't know, the device completes the pairing procedure, but the message is always the same:

> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder

I've re-run `install_prereq.sh`, now I get again the NO_SHARED_CIPHER messages, as it was inside docker. I'll try again.

> Using interface wlan1 with hwaddr 00:02:6f:8d:e7:e3 and ssid "vtrust-flash"
> wlan1: interface state UNINITIALIZED->ENABLED
> wlan1: AP-ENABLED 
> wlan1: AP-STA-CONNECTED 68:57:2d:6d:f1:d7
> wlan1: AP-STA-CONNECTED 10:7b:44:bd:f2:b7
> wlan1: AP-STA-CONNECTED 68:57:2d:6d:f1:d7

> #> arp -n
> Address                  HWtype  HWaddress           Flags Mask            Iface
> 10.42.42.27              ether   68:57:2d:6d:f1:d7   C                     wlan1
> 10.42.42.10              ether   10:7b:44:bd:f2:b7   C                     wlan1

kueblc, 2020-02-01:
> I don't know, the device completes the pairing procedure, but the message is always the same:
> 
> > Device did not appear with the intermediate firmware
> > Check the *.log files in the scripts folder

Yes, the message is always the same on failure. You'll need to check your log files. Since your issue is not related to this one, if you would like further assistance, please open your own issue with your log files attached and I would be happy to continue the discussion with you there.

pixomanie, 2020-02-14:
> @AlexBV mine is a Bakibo TB95 with the same problem.

Bakibo TP22Y the same problem
Version: 
WI-FI Modul: 1.1.4
MCU-Modul 1.1.4

[bakiboT22Y.pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/4205411/bakiboT22Y.pcap.zip)

eku, 2020-02-16:
@kueblc Colin, is there anything else I can contribute to support for the new PSK format beyond a network recording before returning the smart bulbs? Or should I better keep them because it is foreseeable that they can be converted soon?

andreashergert1984, 2020-02-23:
unfortunatly i discoverd another "thing" with the same error: it´s a tuya smart shutter . 
[image]

iracigt, 2020-02-24:
After spending some time digging through the firmware, it's pretty clear Tuya has upped their game here. It appears a unique preshared key is programmed into each device at the factory. Unlike before where the key could be derived from the TLS identity, now the identity is a hash of the device's id and MAC address. For those following along, the new identity derivation is:

'\x02' + 'BAohbmd6aG91IFR1' + sha256(prod_idx + mac_addr)

where prod_idx is the ASCII of the 8 digit product ID, and mac_addr is the MAC address of the device in hex, all lowercase, no punctuation. These parameters are all stored in the flash in a JSON blob at 0xfb000.

kueblc, 2020-02-24:
Great findings @iracigt, thank you for sharing.

@eku I really appreciate the data you have collected, unfortunately it is not possible to say how long it will take to find a workaround, but I am optimistic someone in the community will.

connoleg, 2020-02-28:
@kueblc @iracigt thanks for your help with this. Much appreciated.

brott8, 2020-02-28:
Appreciate your work @kueblc @iracigt .  I just received a tuya wifi smart led bulb and connected to Smart Life.  Then tried to flash the next day and received similar errors to those reported here.
Let me know if there's anything I can do to help with my device.

leifclaesson, 2020-03-04:
Here's the firmware dump from a [Moeshouse Smart Downlight](https://www.aliexpress.com/item/4000296147883.html) with the same issue. New right out of the box, never paired with the tuya app.

[moeshouse_downlight_1m.zip](https://github.com/ct-Open-Source/tuya-convert/files/4287008/moeshouse_downlight_1m.zip)

Tallboy22, 2020-03-06:
Think I'm seeing the same issue with a BSD29 - getting the following error in the smarthack-psk log:

('could not establish sslpsk socket:', SSLError(1, u'[SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:727)'))

Anyone worked through a solution yet?

Woodpeckercz, 2020-03-08:
Is there any way to backup/download the firmware on windows? I got 2x BSD29 and was unable to flash it with tuya convert, decided to open them up and flash them the old fashioned way, was successful on the first one, so I would like to do a firmware backup and share it here before I flash the second one.

Woodpeckercz, 2020-03-08:
Here's the original firmware for BSD29 smart plug.
Hope it helps.

[BSD29_firmware_1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/4303944/BSD29_firmware_1M.zip)

Tallboy22, 2020-03-09:
Woodpeckercz - How did you open them up?  They look pretty impenetrable !

Tallboy22, 2020-03-09:
How did you manage to open the devices up to flash them?  They look pretty impenetrable!

Woodpeckercz, 2020-03-10:
> How did you manage to open the devices up to flash them? They look pretty impenetrable!

Pushed a screwdriver into the grounding pin hole and pushed until the weld gave in, nothing broke, i only had to use a tiny bit of superglue to make sure the cover didn't come off when pulling the plug out of the socket.
I also had to desolder the L N terminals to access the bottom of the pcb for programing the esp8285.
I can post pictures if needed.

Tallboy22, 2020-03-10:
Pictures would be great if you could post some.  Been a while since I've done any fine soldering, so slightly apprehensive about opening it up..

leifclaesson, 2020-03-14:
Wow, buying tuya gear now is _really_ hit and miss. Received four MALITAI Smart WiFi 9W RGBCCT E27 bulbs today. Three of them worked perfectly with tuya convert. The fourth one would not enter fast flashing mode, until I finally toggled the bulb on and off around 10 times... and then, it turned out to be blocked! I'll bet they pulled four bulbs off the shelf, _helpfully_ tested one by pairing it, at which point it got updated.

I solved that one with a soldering iron -- not the easiest tasmota conversion ever, but... yeah. you know.

CHAZICLE, 2020-03-15:
> now the identity is a hash of the device's id and MAC address

@iracigt  I'm not certain yet but it looks like you can use the Smart Life app to pair the device and get the prod_idx. I found it under "Device Information" as the first 8 characters of the "Virtual ID" string and it seems to match the prod_idx I found in the json blob. Can anyone else confirm this?

Also it looks like the prod_idx happens to be identical for the particular set of 6 BSD29s that I bought. I'll take a closer look when I have some more time but is that all we need to flash the firmware?

akaBaib, 2020-03-15:
@CHAZICLE I have checked my BSD34, the Virtual ID is 83203175d8f15bd49015. Version of WI-FI module is 1.1.4. But I have no idea how to check the prod_idx "in the json blob" :(

[231 more replies]