ct-Open-Source/tuya-convert #484: Alternatives for those of us with RTL8710 ("WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware")

TRSx80, 2020-01-03:
First off, thank you @kueblc for picking up the torch on this, and thanks to everyone who contributed in the past, for that is the only way for any of us to Have Nice Things(tm). :smile: 

I have been reading for the last couple days, unfortunately it appears that I have purchased some modules that contain the RTL8710 as I am getting the dreaded error as posted in the title. Therefore I won't bother posting logs, unless you feel they are relevant (so far all the threads I read point to this being a fairly reliable indicator).

At this point, I suppose the only things left to do are:
1. Warn others.
2. Contribute to any development that may be going on with RTL8710.
and/or
3. Put these to use in my home in some other way.

And so, on to each of those in turn:

# 1. Warn others
I would love to do my part, however I could not even seem to figure out exactly the model number of the plugs I purchased from AliExpress, but here is the link: https://www.aliexpress.com/item/4000346047401.html  Any advice here would be appreciated.

# 2. Contribute to any development that may be going on with RTL8710
I read in some post that you were not aware of any stable alternative, is this still the case? I have fair amount of equipment here and I am very willing to sacrifice one or more of my units for the greater good (as so many others have done before us).

# 3. Put these to use in my home in some other way
So they are not a complete loss (and in case I get any more modules like this in future) I think I would like to do something like keeping them on their own subnet, not connected to Internet (ie, "Network Of Things" (NOT) ) and using something like [TuyaApi](https://github.com/codetheweb/tuyapi) with the [tuya-mqtt](https://github.com/TheAgentK/tuya-mqtt) wrapper for openHAB. But that is more a statement than a question, and leaving it here to offer others in my same situation an alternative.

meingraham, 2020-01-03:
A few months ago I saw a blog or a YouTube video (I could not find it again) where someone was able to flash an RTL Wi-Fi chip. It was a PoC of replacing the firmware only. There was no actual firmware for it. So, flashing is possible. But I have not seen any projects for an actual RTL firmware replacement.

TRSx80, 2020-01-03:
Thanks @meingraham for the feedback.

In the meantime, I tried to leave a warning on the item page thru feedback, although it hasn't shown up yet.

kueblc, 2020-01-03:
Regarding 2., here are some potential resources that might help start an alternative firmware project

https://github.com/TuyaInc/tysdk_for_rtl8710bn/
https://www.amebaiot.com/en/ameba-arduino-getting-started-rtl8710/

amoksworld, 2020-01-03:
I received some "no-name" sockets from ebay and got the message (...no esp...), too.
Inspired of https://github.com/ct-Open-Source/tuya-convert/issues/483 I deceided not to give up to soon, because I got some SSL-errors, too.
So I immolated one and had a look inside:

[image]

[image]

Does anyone know if this is an ESP or not ?!?

If so, I can try to extract the Firmware for debugging ...

kueblc, 2020-01-03:
That is a WinnerMicro W600, not an ESP unfortunately

TRSx80, 2020-01-03:
@kueblc,
Do you think it's worth it do disassemble one of mine and confirm what is inside?

kueblc, 2020-01-03:
You might be able to determine the manufacturer without opening it by the MAC address of the device

TRSx80, 2020-01-03:
Good idea. According to [The Wireshark OUI lookup tool](https://www.wireshark.org/tools/oui-lookup.html):
>Results
>28:6D:CD BeijingW Beijing Winner Microelectronics Co.,Ltd.

Tollbringer, 2020-01-29:
TRSx80 ! Seriously? Are you a proud user of the imfamous Coco/Tandy product? It was my 1st PC.

It will be very interesting to see progress on RTL chips! if I am not mistaken - they add the 5ghZ range?

Not sure how that would help unless a lot of data needs to be transmitted/received.

- Keep up the good work all!

Tollbringer!

brumela, 2020-02-27:
How about replacing (resolder) RTL8710 module with something like this [ESP8285](https://www.aliexpress.com/item/32910642969.html?spm=a2g0o.productlist.0.0.64685049Llwkiq&algo_pvid=5d626660-a13a-49f9-b739-d141453a054a&algo_expid=5d626660-a13a-49f9-b739-d141453a054a-3&btsid=0b0a187915828009436535576e294a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) or [similar](https://www.aliexpress.com/item/33060326374.html?spm=a2g0o.productlist.0.0.64685049Llwkiq&algo_pvid=5d626660-a13a-49f9-b739-d141453a054a&algo_expid=5d626660-a13a-49f9-b739-d141453a054a-9&btsid=0b0a187915828009436535576e294a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)?

meingraham, 2020-02-27:
@brumela 

I saw a couple of users post that they were able to replace the RTL based Tuya module with the ESP based Tuya module and that worked. The Tuya modules are pin compatible. The RTL and ESP Wi-Fi SoC are quite different in this regard.

TRSx80, 2020-06-22:
So, I started disassembling my unknown module.  I decided to put that in a separate thread, which you should find linked immediately above this post (feature of GitHub when you link from somewhere else).

Laces, 2021-01-20:
Hi, I also bought a tuya light switch but unfortunately there is no esp8226 inside, but something else. I managed to upload MicroPython with W600 tool but I can't use the WIFI of the chip. Did you guys saw this chip before? I'm out of ideas...
[image]

Godkezza, 2021-02-21:
Has there been any movement on this? I currently have four Teckin (SP27) plugs which I believe to have RTL8710 chips inside (haven't been able to open them yet as I don't want to damage them). Tuya is really playing up for me as I can no longer integrate with IFTT or Home Assistant.

Lahvuun, 2021-02-22:
There isn't any alternative firmware for RTL8710 to my knowledge. An alternative that worked for me was [localtuya](https://github.com/rospogrigio/localtuya) with homeassistant.

Laces, 2021-02-23:
I have ordered and ESP-12 and will try to replace the original chip with the ESP-12.

b0rder, 2021-03-11:
> I have ordered and ESP-12 and will try to replace the original chip with the ESP-12.

I found some Tuya devices with Realtek modules successfully modified by replacing WBR3 Module (RTL8720) to ESP-07. They are pins compatible (at least power and tx/rx).
For example, here: https://github.com/aoe1/WThermostatBeca/blob/master/MH-1823.md
Hope you will success also, so please keep in touch.

Laces, 2021-04-30:
I gave up. I have tried to replace the original chip with an ESP-12, but couldn't upload a firmware, then I used a D1 mini to upload the firmware to the ESP-12 and soldered to the board once again, but still doesn't work. I assume some resistors are needed for the ESP-12 to be able to start which are not present on the board, so it's time to give it up and order a proper board.
[image]
[image]

openshwprojects, 2022-04-22:
Thanks for mention, @nx10 
I am already able to also flash BL602 and RTL8710. 
MQTT and Home Assistant already works on BL602 devices.

https://www.elektroda.com/rtvforum/viewtopic.php?p=19984465#19984465

mstaack, 2022-04-29:
> Regarding the RTL8710 in particular, I'm working on a PlatformIO development platform with Arduino support, with ESP32 compatibility in mind:
> 
> https://github.com/kuba2k2/platformio-libretuya
> 
> I already have some basics and WiFi AP/Station/Client/Server working on RTL8710, using Arduino example sketches directly. In the nearest future, I'll be adding more functionality and libraries (like HTTP(S), I2C, SPI, MQTT, etc).
> 
> The goal is to run ESPHome on this chipset - which will provide a full-fledged Home Assistant integration for these unsupported Tuya boards.
> 
> 
> 
> In the (further) future, I'm planning to also support other chipsets, like BK7231T with which I have a few bulbs.
> 
> 
> 
> I'd happily accept any contribution, like testing the project, bug reports, or even code contribution :smile: 

Sweet!

openshwprojects, 2022-07-24:
@Laces OpenBeken now works on W800 with OTA (updating firmware by WiFi, but for initial flashsing you need wires), OpenBeken will also soon support W600 as well.

jbalius, 2022-09-04:
Can maybe someone help me to identify chip on this noname RGBCW XLD-W70-1.1 ? Chip just says TY5, tuya convert says it is not ESP8266xx.

[image]
[image]

kuba2k2, 2022-09-04:
@jbalius 
[image]

(from https://www.reddit.com/r/tasmota/comments/kpsbwf/ty5_mcu_identification/)

openshwprojects, 2022-09-08:
OpenBeken already supports BK7231T, BK7231N, W800, W801, T34, BL602 and soon W600 and RTL.

We also have an interactive online devices database with device templates that is accessible through OBK javascript web panel. Devices database is held online, so even if you have old binary you will always have all new templates, even with photos:
[image]

vinibali, 2023-01-09:
Hi @kuba2k2 @openshwprojects!
Are you planning to work together or join your forces?
You both are very agile in the non-ESP development.

amirmazmi, 2023-04-17:
Just found this. In case no one has seen this, someone swapped out the RTL with an esp. 

"STOP using Tuya! Transplant your ESP8266 100%" by digiblurDIY on youtube - https://youtu.be/d_HpkIiWC3Y

VWFeature, 2023-06-08:
I am only an egg, but it seems some of these problems w the Realtek RTL8710BN and Tuya in general are solved w LibreTiny:
My understanding is limited; hope this helps someone else!

https://docs.libretiny.eu/docs/status/supported/#tuya-inc
https://docs.libretiny.eu/docs/projects/esphome/
https://docs.libretiny.eu/docs/status/supported/

smiell, 2024-01-17:
Hello, recently I bought wifi socket plug, that using Smart Life app and Tuya cloud. It is look like this:
[image]

I tried to [tuya-convert](https://github.com/ct-Open-Source/tuya-convert) but, It won't find my wifi socket, so I am considering about what is inside... ESPxx chip or RTL. I want flash esphome into.

I also check by mac addres C4:82:E1:28:3B:7A and vendor is Tuya Smart Inc. 
I do not know nothing about this wifi plug....

I bought this form polish marketplace allegro: https://allegro.pl/oferta/gniazdko-wifi-smart-watomierz-timer-licznik-14195644657

In product descryption is nothing that can give me more information about what is inside.... 

Maybe somone know what is inside and it is posibility to flash tasmota or esphome (esphome will be exactly!) into ;)

kuba2k2, 2024-01-18:
That plug is most likely based on a Beken chipset. There are currently two firmware options - [ESPHome](https://esphome.io/components/libretiny) (with [LibreTiny](https://docs.libretiny.eu/docs/projects/esphome/)) or [OpenBeken](https://github.com/openshwprojects/OpenBK7231T_App) (Tasmota-like firmware).

mrfy, 2024-05-11:
> Hello, recently I bought wifi socket plug, that using Smart Life app and Tuya cloud. It is look like this: ![IMG_8243](https://private-user-images.githubusercontent.com/18280428/297530183-0f9bde04-5623-4261-8e5a-cfebb7b1ff26.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0MTM4NDIsIm5iZiI6MTcxNTQxMzU0MiwicGF0aCI6Ii8xODI4MDQyOC8yOTc1MzAxODMtMGY5YmRlMDQtNTYyMy00MjYxLThlNWEtY2ZlYmI3YjFmZjI2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxMVQwNzQ1NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZWVjNmJlNDA3NTZhOWE4NTc4YjJjYmI4ZWE4NjM2YjZmZDE3OWIwNTg5MzEyYjRmODhmMzBmNTZjMGViNGU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NnRrn0mIZFtLbCZnJsee6sl42sxZxloTtWprj7_PqQY)
> 
> I tried to [tuya-convert](https://github.com/ct-Open-Source/tuya-convert) but, It won't find my wifi socket, so I am considering about what is inside... ESPxx chip or RTL. I want flash esphome into.
> 
> I also check by mac addres C4:82:E1:28:3B:7A and vendor is Tuya Smart Inc. I do not know nothing about this wifi plug....
> 
> I bought this form polish marketplace allegro: https://allegro.pl/oferta/gniazdko-wifi-smart-watomierz-timer-licznik-14195644657
> 
> In product descryption is nothing that can give me more information about what is inside....
> 
> Maybe somone know what is inside and it is posibility to flash tasmota or esphome (esphome will be exactly!) into ;)

Hi, did you manage to flash this socket? Is that beken? I have similar. Also Tuya (based on mac address).

smiell, 2024-05-11:
> Hi, did you manage to flash this socket? Is that beken? I have similar. Also Tuya (based on mac address).

In February this year, I successfully flashed 5 sockets of this type to the free Open Beaken software, authored by a Pole. Here I leave you the link to the repository [Github repo](https://github.com/openshwprojects/OpenBK7231T_App)

You will even find videos on their channel on how to change the firmware in similar modules that contain the **BK7231 chip**.

They also have their own flasher prepared for this ;)

mrfy, 2024-05-11:
> > > Hello, recently I bought wifi socket plug, that using Smart Life app and Tuya cloud. It is look like this: ![IMG_8243](https://private-user-images.githubusercontent.com/18280428/297530183-0f9bde04-5623-4261-8e5a-cfebb7b1ff26.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0MTM4NDIsIm5iZiI6MTcxNTQxMzU0MiwicGF0aCI6Ii8xODI4MDQyOC8yOTc1MzAxODMtMGY5YmRlMDQtNTYyMy00MjYxLThlNWEtY2ZlYmI3YjFmZjI2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxMVQwNzQ1NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZWVjNmJlNDA3NTZhOWE4NTc4YjJjYmI4ZWE4NjM2YjZmZDE3OWIwNTg5MzEyYjRmODhmMzBmNTZjMGViNGU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NnRrn0mIZFtLbCZnJsee6sl42sxZxloTtWprj7_PqQY)
> > > I tried to [tuya-convert](https://github.com/ct-Open-Source/tuya-convert) but, It won't find my wifi socket, so I am considering about what is inside... ESPxx chip or RTL. I want flash esphome into.
> > > I also check by mac addres C4:82:E1:28:3B:7A and vendor is Tuya Smart Inc. I do not know nothing about this wifi plug....
> > > I bought this form polish marketplace allegro: https://allegro.pl/oferta/gniazdko-wifi-smart-watomierz-timer-licznik-14195644657
> > > In product descryption is nothing that can give me more information about what is inside....
> > > Maybe somone know what is inside and it is posibility to flash tasmota or esphome (esphome will be exactly!) into ;)
> > 
> > 
> > Hi, did you manage to flash this socket? Is that beken? I have similar. Also Tuya (based on mac address).
> 
> In February this year, I successfully flashed 5 sockets of this type to the free Open Beaken software, authored by a Pole. Here I leave you the link to the repository [Github repo](https://github.com/openshwprojects/OpenBK7231T_App)
> 
> You will even find videos on their channel on how to change the firmware in similar modules that contain the **BK7231 chip**.
> 
> The software provides support for MQTT so In my opinion that is enough for using with home assistant and etc.
> 
> They also have their own flasher prepared for this ;)

Thank you very much for answer :). I have few sockets so I think it is worth to flash them. Are you using simple USB UART converter, something like this? https://allegro.pl/oferta/konwerter-usb-uart-rs232-cp2102-3-3v-5v-kable-15478664604

smiell, 2024-05-19:
Yeas, I used simply USB uart converter ;) That should work fine [USB UART Converter](https://allegro.pl/oferta/konwerter-usb-uart-rs232-cp2102-3-3v-5v-kable-15478664604)