ct-Open-Source/tuya-convert #485: Smart Live - Smart Living let me not connect to vtrust AP

lucki1000, 2020-01-04:
I would to try it on my Smart LED Bulb I've used Kali Linux and downloaded the Smart Live app from Play Store on my Phone and created an Account by Tuya but while I would add my Bulb it ask me with which WiFi I want to connect my lamp and if i select vtrust it let me not continue because it's not password protected, after i've set a password in the AP config file, I get the error that i don't have Internet. Is there a Solution for this or any older apk to make them working?

jcageman, 2020-01-04:
why would you want to use the tuya app to connect to V-trust? I assume you want to flash custom firmware on your LED Bulb? You can use the instructions on this github to do this, which is the way to do it without the smartapp. I would even recommend not connecting to the smart app, since you run the risk of upgrading the firmware of tuya making it non-compatible anymore.

See: https://github.com/ct-Open-Source/tuya-convert/blob/master/README.md#procedure

lucki1000, 2020-01-04:
But I got at this way a loop of:

SmartConfig complete.
Resending SmartConfig Packets
.................

kueblc, 2020-01-04:
Hi @lucki1000, as @jcageman you are not supposed to use the app. Follow the procedure as described in the README. Good luck!

closed 2020-01-04 by kueblc

lucki1000, 2020-01-05:
@kueblc That whats I mean if i use the guide from the GitHub readme i got this loop.

kueblc, 2020-01-05:
@lucki1000 there are instructions for when you open a new issue that you overwrote and you did not follow them. If you have an issue the only way I can help you is if you help me help you.

That means you need to provide all of the information requested, including the logs and information about your platform and device. The problem you described tells me nothing about your issue, there are too many things that could cause the process to stop at smartconfig.