ct-Open-Source/tuya-convert #486: Device did not appear with the intermediate firmware

iqmend, 2020-01-04:
I tryed to flash a motion sensor of tuya but i failed.

i tryed this ... but dont help 
[image]
[image]

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4022721/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4022722/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4022723/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4022724/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4022725/smarthack-wifi.log)

kueblc, 2020-01-05:
ESP power is getting cut before the upgrade completes. Duplicate of #135

kueblc, 2020-01-05:
You can try @Scoobler's suggestion https://github.com/ct-Open-Source/tuya-convert/issues/135#issuecomment-545756091

closed 2020-01-05 by kueblc

iqmend, 2020-01-05:
can not hold it awake. The Motion Sensor work only 1 time every 2min

kueblc, 2020-01-05:
Unfortunately there is nothing we can do in tuya-convert to keep it awake that I'm aware of

Scoobler, 2020-01-05:
Colin is correct, the MCU that detects motion is what powers up the ESP.
The device needs to enter pairing mode to keep power on to the ESP, firmware on the ESP can do this, but that isn’t till after Tuya-convert has loaded it which is too late.

One other thing to try if the device has a pairing button (normally press and hold for 5-8 seconds till the light flashes) hold it to enter pairing mode, this action tells the MCU to power up the ESP, it normally holds power for around 90-120 seconds, or till the ESP responds (this would normally be a message from the original firmware to say ‘all finished’), however when the ESP reboots after loading a new firmware from Tuya-convert it spits out a load of info on the serial line, the MCU doesn’t understand so shuts down the power to the ESP, this would mean however your at step 1 of the convert process, holding the pairing button again would keep power on and may allow Tuya-convert to continue - you may need to do this a couple of times as the different stages of firmware are loaded / rebooted.

iqmend, 2020-01-05:
@Scoobler i tryed exactly yout idea for 20 times. After i read your idea here i tryed again and than after 3-4 times it worked :)

Scoobler, 2020-01-05:
@iqmend 👍

Scoobler, 2020-01-16:
It should be possible, if you do have Tasmota on to keep the device on without soldering. 

If you press and hold the button for around 5-8 seconds this tells the MCU to enter pairing mode, the Low Power Protocol suggests the MCU should keep power on for around 90 seconds, as it enters pairing mode it sends a message to the ESP telling it to enter pairing mode, it then waits for 90 seconds OR till it gets a message to say pairing complete, as Tasmota doesn’t know to send that message it should stay on for 90 seconds. However when you save something in Tasmota it reboots, this will make the ESP spit out a load of info on the serial line, as the MCU is waiting for the completed message, it can get confused and shuts down power sooner than the 90 seconds, simply press and hold to get another 90 seconds.

karlitos, 2020-01-16:
I just did a check. Holding the button of the LSC Motion Sensor starts the LED blinking after some 3-5s. The blinking goes on for aprox 15s and stops. Pushing the button for longer time does not bring any different behavior. I am affraid, that the MCU puts the ESP to sleep chip quite forecibly.