ct-Open-Source/tuya-convert #487: CE Smart Home outlet won't flash

jcbshw, 2020-01-05:
Attempting to flash my CE Smart Home outlet, it always disconnects from vtrust-flash a few seconds after connecting. Using a RPi4 running Raspbian Buster. 

Side note: When I first set these devices up to the cloud, ez mode never worked... assuming it was because of my AP combined 2.4Ghz and 5Ghz SSID.

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4022837/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4022838/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4022839/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4022840/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4022841/smarthack-psk.log)

kueblc, 2020-01-05:
Did you try [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting)? Perhaps "vtrust-flash rejects subsequent connections after the first"?

countcobolt, 2020-01-05:
Hi just wondering if this is similar to #483 ? I had the same behaviour? It had tuya 1.1.4 firmware on it

kueblc, 2020-01-05:
No @countcobolt, you were able to get the device to connect as evidenced by your PSK log. Here, @jcbshw's device isn't staying connected at all.

countcobolt, 2020-01-05:
@kueblc  Okido :)

jcbshw, 2020-01-05:
@kueblc thanks for pointing that out. I attempted on my RPi4 running Arch Linux but had the same issues as well. I'm not sure if its directly related to the issue described in troubleshooting as it was on Raspbian Stretch with hostapd 2.4 which was fixed in 2.6.

I was able to successfully flash using my laptop on Ubuntu 19.10.

closed 2020-01-05 by jcbshw