ct-Open-Source/tuya-convert #488: Switch dont work after flashing

iqmend, 2020-01-05:
Hello Guys, 
i flashed yesterday one device and was happy. 
today i tryed to flash to more switches. The flash progress was ok but after falshing tasmota on it the device dont switch the light on/off. 

Also te rest function dont work (hold 10sec) ... i also can not switch the light manual with my finger. 
When i hold 3 sec the blue light for pairing mode works. But thats all.

[image]
[image]
[image]

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4023434/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4023435/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4023436/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4023437/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4023438/smarthack-wifi.log)

iqmend, 2020-01-05:
Edit: 
Now i changed the config template and the secon switch is working. 

But my next problem is, that i dont have access to my first switch. i can not find it in my router and i also can not open it with my ip anymore -.-

dont understand this switches are the same product and both was succesfully flashed

iqmend, 2020-01-05:
i think i installed a wrong template on the first switch and the switch is now in deep sleep ot something like that. 

I just need to get access to the web interface to change the template than it will work. But how can i get access or restore the template configuration without access to the switch ? 
its not listet in my wifi devices -.-

kueblc, 2020-01-05:
It sounds like you are looking for [Tasmota](https://github.com/arendst/Tasmota) support. We don't develop or provide support for thirdparty firmware, only the means to flash it. Good luck!

closed 2020-01-05 by kueblc

epicurean70, 2020-01-21:
> i think i installed a wrong template on the first switch and the switch is now in deep sleep ot something like that.
>
> I just need to get access to the web interface to change the template than it [...]

Did you get your button and dimmer working with the Teekar? which firmware did you use?