ct-Open-Source/tuya-convert #492: Device did not appear with the intermediate firmware - Moes 3-way switch

mikeg1130, 2020-01-09:
I am having receiving the 'Device did not appear with the intermediate firmware' error while attempting to  flash 2 new Moes 3-way switches. Model No. SS01. A few weeks ago I flashed a nearly identical pair of the switches without issues. These seem to be a different revision of the switch, although the wi-fi modules appear to be identical.

Looking through similar issues, but i did not see anything that was immediately apparent as a solution.

Attached are my logs. I appreciate the assistance.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4040901/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4040902/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4040903/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4040904/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4040905/smarthack-wifi.log)

dr-azbest, 2020-01-18:
I have same issue here. I was able to flash exactly the same wall switch (minitiger 1band) and now Im not. I found 
> wlan0: Could not connect to kernel driver
line in wifi.log and just wonder if processing "rpi-update" command could broke compability ?
Did you @mikeg1130 also proceed "rpi-update" command last time ?

dr-azbest, 2020-01-18:
Reffering to https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting `

wlan0: Could not connect to kernel driver

` is not an issue

dr-azbest, 2020-01-18:
Ok, I think indeed rpi-update could brake compability as i was able to make it work again by this below commands which reverts kernel to stock
`sudo apt update
sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel`

kueblc, 2020-01-25:
Looks like you are experiencing the same issue as #495. We'll track progress over there.

closed 2020-01-25 by kueblc