ct-Open-Source/tuya-convert #495: Cannot flash device Avatto S07 Tuya Universal Smart 2.4G Wifi Ir remote control

jholweg, 2020-01-10:
Flashing doesn't succeed for this device. After opening it up, I can see the device is ESP8266EX. The flashing tool reports that it doesn't receive the device's firmware.
(Note: flashing it by using the pins also doesn't work, probably because the TX/RX pins are connected to some hardware, so using that method is not an option --> hope I will be able to use TuyaConvert!)

Logs are attached.
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4045619/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4045620/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4045621/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4045622/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4045623/smarthack-mqtt.log)

kueblc, 2020-01-12:
From the web log
```
[32m[I 200110 12:18:35 web:2162][m 206 HEAD /files/upgrade.bin (10.42.42.29) 42.55ms
[32m[I 200110 12:18:35 web:2162][m 200 GET /files/upgrade.bin (10.42.42.29) 46.16ms

POST /gw.json?a=s.gw.upgrade.updatestatus&et=1&gwId=56685573d8bfc050889e&t=1578655114&sign=b99faa1c724b1869f7f0ce3b33b27d4c
Host: 10.42.42.1
User-Agent: ESP8266SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 101

payload {"upgradeStatus":4,"upgradeText":7}
```

It looks like the Tuya firmware fetched the tuya-convert firmware but encountered an error

jholweg, 2020-01-14:
@kueblc any idea what these code (4 resp. 7) mean? And more importantly: is there a way around it?

kueblc, 2020-01-14:
`"upgradeStatus":4` means an error occurred, maybe due to the wrong hash or some other problem with the response format. It might indicate a new patched firmware, but it's difficult to say what the exact problem is without a copy of the firmware. You said you don't think you can flash by wire so let's hope someone else runs into the same issue and can obtain a firmware backup.

jholweg, 2020-01-20:
I've tried with another IR remote control device. This one has a Tuya TYWE3S chip onboard:

[image]

Flashing doesn't work here either (same {"upgradeStatus":4,"upgradeText":7} result).

When trying to upload firmware directly (serial connection, booting into firmware upload mode), I get an error (ESP8266 chip stub error) - see the following screenshot:

[image]

Any idea how to get the firmware out, flash it with tuya-convert, or to put new firmware in through the serial connection?

Thanks,

Jaap

jholweg, 2020-01-21:
Actually, I found the problem: I put a voltage regulator between the USB adapter and the ESP8266, since the USB adapter generated 5V instead of 3.3V. That caused the problem. After removing the voltage regulator, it works now (except that I broke the ESP8266 now).
I managed to get the firmware out of it. Is this of any help to see what's the problem?

[tuya_flash_1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/4090996/tuya_flash_1M.zip)

kueblc, 2020-01-22:
Thank you @jholweg, yes this will definitely help debug the issue. I will need to do some testing and follow up with you. Please ping me if I do not follow up soon.

jholweg, 2020-03-24:
Hi @kueblc , any update on this? Thanks...