ct-Open-Source/tuya-convert #497: TypeError: argument 2 must be str, not bytes

eku, 2020-01-11:
I've just upgraded ```tuya-convert``` to ```HEAD``` of branch ``` development```, reinstalled the required Python 3 modules and get the following error:

```
new client on port 443 from 10.42.42.14:6147
Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 107, in main
    p.data_ready_cb(s)
  File "./psk-frontend.py", line 78, in data_ready_cb
    self.new_client(_s)
  File "./psk-frontend.py", line 66, in new_client
    hint=self.hint)
  File "/usr/local/lib/python3.7/site-packages/sslpsk/sslpsk.py", line 104, in wrap_socket
    _ssl_set_psk_server_callback(sock, cb, hint)
  File "/usr/local/lib/python3.7/site-packages/sslpsk/sslpsk.py", line 80, in _ssl_set_psk_server_callback
    _      = _sslpsk.sslpsk_use_psk_identity_hint(_sslobj(sock), hint if hint else "")
TypeError: argument 2 must be str, not bytes
```

System is Fedora 30/x86.

```
# pip3 list
pycrypto         2.6.1   
pycryptodomex    3.9.4   
sslpsk           1.0.0   
```

dseven, 2020-01-21:
Seems that the fix for https://github.com/ct-Open-Source/tuya-convert/issues/500 is incomplete. I applied the following change, based on @flufftronix comment in https://github.com/ct-Open-Source/tuya-convert/issues/424 and it solved the problem for my Aoycocr switch.

```
diff --git a/scripts/psk-frontend.py b/scripts/psk-frontend.py
index 586f616..c69ea77 100755
--- a/scripts/psk-frontend.py
+++ b/scripts/psk-frontend.py
@@ -63,7 +63,7 @@ class PskFrontend():
                                ssl_version=ssl.PROTOCOL_TLSv1_2,
                                ciphers='PSK-AES128-CBC-SHA256',
                                psk=lambda identity: gen_psk(identity, self.hint),
-                               hint=self.hint)
+                               hint=self.hint.decode())

                        s2 = client(self.host, self.port)
                        self.sessions.append((ssl_sock, s2))
```

kueblc, 2020-01-22:
This is symptomatic of not reinstalling `sslpsk`, as the older version required `str`. Updating this would allow you to run the software as written. You may have to uninstall all instances of `sslpsk` first, or as #508 suggests, using `--reinstall`.

dseven, 2020-01-22:
So... https://github.com/ct-Open-Source/tuya-convert/blob/master/install_prereq.sh#L8 installs from a hard-coded github URL. I did this:

```
root@rpi3:/home/pi/tuya-convert# pip install --upgrade git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints
Collecting git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints
  Cloning https://github.com/drbild/sslpsk.git (to use-byte-string-for-identity-hints) to /tmp/pip-Mv0Hd8-build
Installing collected packages: sslpsk
  Found existing installation: sslpsk 1.0.0
    Uninstalling sslpsk-1.0.0:
      Successfully uninstalled sslpsk-1.0.0
  Running setup.py install for sslpsk ... done
Successfully installed sslpsk-1.0.0
root@rpi3:/home/pi/tuya-convert#
```

but the problem persists. Is there a new source for sslpsk?

kueblc, 2020-01-22:
My mistake, yes I meant the `--upgrade` flag. Perhaps you can try:
```
sudo -H python3 -m pip install -I git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints
```

It may be that you have multiple sslpsk installations and `pip` is referencing a different location than `python3`.

dseven, 2020-01-22:
Ahh, right, pip is actually python2.7 in my case - duh! I should have used pip3. That seems to have fixed it. At one point, I did re-run `install_prereq.sh`, but because sslpsk was already installed, it didn't do anything. I wonder if adding `--upgrade` would allow it to work for both new installations and updating existing ones?

kueblc, 2020-01-22:
Great! Indeed, I will add `--upgrade` to the next release to avoid this issue in the future. Thanks for your feedback!

PR #496 (2020-01-12): Fix byte encoded string bug in psk-frontend
Fixes #497
commit 50a03c83 (2020-01-24): Use upgrade flag in pip to force sslpsk reinstall, fix #497, fix #508 thanks @Lovely-Maisonette
closed 2020-01-25 by kueblc