ct-Open-Source/tuya-convert #498: unable to flash door sensor

forty76, 2020-01-12:
Hi. I want flas tasmota to door sensor with TYWE3S. MCU HDSC-HC32L110 (ARM Cortex-M0+, 32 MHz, 32 kB ROM, 4 kB RAM)

Device have magnetic switch + tamper switch, I don't connect to wifi by app (smart life). 
When try flash device stop at backup fimware, after many test I can download fetching bootloader to put +vcc on rst pin of TYWE3S. But don't flash tasmota firmware.

I try flash TYWE3S by wire whit NODEMCUflashplayer but don't recognize device:
   FTDI232
   GPIO-0 to 0
   Rx-Tx connect
   RST and EN connect to Vcc(3.3)
   I also tried with external power

[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/4050939/device-info.txt)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4050940/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4050941/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4050942/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4050943/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4050944/smarthack-wifi.log)
[firmware-7e2d86_bin.txt](https://github.com/ct-Open-Source/tuya-convert/files/4050949/firmware-7e2d86_bin.txt)
[image]
[image]
[image]
[image]

haubke, 2020-01-19:
Door sensors does not work properly woth ota update( only stays on for a few sec. If you flash it by serial you dont need tuya convert. Then just flash what you want. (Tasmota?)

kueblc, 2020-01-22:
Hi @forty76, as @haubke mentions there are some caveats when flashing these low power devices. Please follow #135.

closed 2020-01-22 by kueblc